arendst/Sonoff-MQTT-OTA-Arduino #2: Compiles and uploads but unit non responsive

xbmcnut, 2016-04-12:
Hi there,

Great work. Uploaded this today on 3 x Sonoff with 5 pin header after identifying the right pins. Uploaded fine and unit is joining WiFi (router leasing address) but the switch does not work to pulse the relay. Can you advise how I go about debugging this? I tried R94 of ESPEasy on one and that works just fine so I know the board/s are OK and everything is working.

xbmcnut, 2016-04-12:
Disregard. Used serial monitor to see the unit saying "Smartconfig active for 100 seconds" and then button became responsive. Is that normal behaviour?

arendst, 2016-04-12:
This is normal behaviour for a bad Wifi connection. If a Wifi connection can not be made it is retried 4 (wifiretry) times for 4 (wificounter) seconds. If the timers time out it starts Smartconfig for 100 seconds and if no response is received it restarts sonoff and the sequence is executed again. Did you notice this on all three sonoffs? Did they work after the smartconfig restart sequence?

xbmcnut, 2016-04-12:
Thanks for the prompt reply. Excuse my noob ignorance, but what are you supposed to do when it goes into 'Smartconfig active' mode? I added STA_SSID, STA_PASS and  MQTT_HOST as "10.0.1.8", is that is all that is needed in the sonoff file?

xbmcnut, 2016-04-12:
Found the Android app. Now get this and then shortly after I lose comms to it and the switch no longer works.
`Smartconfig active for 100 seconds
Smartconfig SSID = xbmcnut and Password = xxxxxxxxxx

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Project sonoff (Topic sonoff, Fallback DVES_81B97A, GroupTopic sonoffs) Version 1.0.6 (Boot 31, SDK 1.5.1(e67da894))`

arendst, 2016-04-12:
In Smartconfig mode sonoff "listens" for SSID and Password provided by an android tool as documented in the README.md chapter Usage, bullet two. If you want to interrupt the Smartconfig process keep the button pressed for four seconds and sonoff will restart.
Your updates to sonoff.ino are indeed the only changes needed to adapt to your situation.

xbmcnut, 2016-04-12:
Seems I had a typo in my IP address for MQTT. Instead of 10.0.2.8 I had 10.0.1.8. Could not find a way to fix that other than to reflash. Am getting messages back now via serial. e.g. Status 1 = STATUS = sonoffs, DVES_81B97A, http://sidnas2:80/api/arduino/sonoff.ino.bin, xbmcnut, xxxxxxxx, 10.0.1.200, 0, 5

arendst, 2016-04-12:
You can change MQTT host with the serial command HOST. If you want to use OTA you'll also have to change the hostname in the OTAURL. Glad you solved it on your own.

xbmcnut, 2016-04-12:
Really appreciate your help, thank you. All working now wit MQTT on my Synology NAS. Have come in cold so the learning curve is a bit steep but up to the challenge.

closed 2016-04-12 by xbmcnut