arendst/Sonoff-MQTT-OTA-Arduino #4: Minor bug - status 1 command

PuceBaboon, 2016-04-30:
Theo,

   First, a very big "Thank You!" for such a marvellous package (I'm a big fan!).  I couldn't believe how much functionality you'd managed to get into Sonoff-MQTT-OTA and  have it working with a sensible MQTT command/string structure, too.

   I've been playing with it quite a bit recently, trying to put together a reference/tutorial with a few working examples and I came across only one obvious anomaly.  The "status 1" command produces output on the serial console (whether triggered from the serial console or from MQTT), but doesn't produce any output to MQTT.  Plain "status" and "status 2" work fine in all cases.

   The main difference seems to be that for the "status 1" command, the output far exceeds the size assigned to svalue[].  With fully qualified hostnames, the output from "status 1" in my case is 140 characters long.  It looks as though the issue is that PubSubClient just quietly returns "false" to the publish request, presumably because the size of the data payload is just too big.  I tried tweaking the size assignment, but without success.  However, if I remove the longest string (sysCfg.otaUrl) from the "status 1" output, it works in both cases (serial monitor and MQTT) as expected.  I ended up just creating a couple of other status options (3 & 4) for MQTT and OTA status, respectively:-

`if (!strcmp(type,"STATUS")) {
      sprintf_P(svalue, PSTR("%s, %s, %s, %d, %d"),
        VERSION, sysCfg.mqtt_topic, sysCfg.mqtt_subtopic, sysCfg.power, sysCfg.timezone);
      if ((data_len > 0) && (payload == 1)) {
        sprintf_P(svalue, PSTR("NET: %s, "MQTT_CLIENT_ID", %s, %s, %d, %d"),
          sysCfg.mqtt_grptopic, ESP.getChipId(), sysCfg.sta_ssid, sysCfg.sta_pwd, heartbeat, sysCfg.saveFlag);
          // sysCfg.mqtt_grptopic, ESP.getChipId(), sysCfg.sta_ssid, sysCfg.sta_pwd, sysCfg.mqtt_host, heartbeat, sysCfg.saveFlag);
      }
      if ((data_len > 0) && (payload == 2)) {
        sprintf_P(svalue, PSTR("FMWR: Version %s, Boot %d, SDK %s"), VERSION, ESP.getBootVersion(), ESP.getSdkVersion());
      }
      if ((data_len > 0) && (payload == 3)) {
        sprintf_P(svalue, PSTR("MQTT: %s, MQTT-MAX-PACKET: %d, MQTT-KEEPALIVE: %d"), sysCfg.mqtt_host, MQTT_MAX_PACKET_SIZE, MQTT_KEEPALIVE); 
      }
      if ((data_len > 0) && (payload == 4)) {
        sprintf_P(svalue, PSTR("OTA: %s"), sysCfg.otaUrl); 
      }
    }
    else if....`

   (I included the MQTT_MAX_PACKET_SIZE mainly because I thought that might be the problem, when the tweak of svalue[SIZE] didn't work, but it is being set to 1024).

arendst, 2016-04-30:
Thanks for the kudos! I'm glad you like it.

I do not observe the problem you raise. I have changed my otaurl in a way that status 1 returns a string of 147 characters and it is displayed by MQTT just fine. I have another project where the syntax string is 216 characters and it is also displayed fine by MQTT.

Any way, you did the right thing by providing your own solution and let us know.

arendst, 2016-07-28:
John,

using your fixes to apply MQTT_MAX_PACKET_SIZE 1024 and MQTT_KEEPALIVE 120 into sonoff.ino just doesn't work! It seems PubSubClient is still using the values from the .h file. I removed the changes from sonoff.ino and reapllied it to PubSubClient.h and it works as designed.

I decided that a packet size of 256 is large enough to send the syntax string via MQTT. 

Your original complaint is most probably also caused by the fact that MQTT_MAX_PACKET_SIZE is still 128 which will drop all MQTT messages consisting of both topic AND message larger than 128 bytes.

I've implemented massive buffer overflow checks in the latest version 1.0.15.

closed 2016-08-21 by arendst