arendst/Sonoff-MQTT-OTA-Arduino #5: Enhancement request HTTP server

hjgode, 2016-05-03:
Hello (again)

I realy like the solution but would like to have a simple HTTP server too. There may be circumstances, where no MQTT broker is available.

I am not sure if the software design allows a simple http server too. A http web server should enable one to check the status (on/off), change the status and possibly start a OTA update.

Unfortunately I am not that familar with ESP8266 and the client and server libs usable.

MANY thanks

Josef

arendst, 2016-08-14:
Josef,

there now is a web server available during configuration of wifi and mqtt parameters. Your request regarding controlling relay and ota are still not implemented as they would need the web server running continuously which I do not prefer.

hjgode, 2016-08-14:
Hello

thank you for the information.

arendst, 2016-08-21:
Since version 1.0.23 an optional webserver is available using station mode allowing control of the relay and among other options starting an OTA upgrade.

See README.md for additional information.

kenshin8, 2016-08-29:
Hi, when I access the web interface, it just failed Intermittently, it shows ERR_EMPTY_RESPONSE, reloading the page can fix it most of the time, tried different hardware, different browsers and still generating the same problem, restarting the device sometime help a little but it will fail again shortly.

Other than keep pressing reload page, all the functions works - Toggle relay, Configure WiFi and MQTT, OTA upgrade, Command line, Restart

Thanks for providing this awesome script.
Ken

arendst, 2016-08-29:
Ken,

the latest version (1.0.26) contains a lot of web server enhancements. Give it a try and let me know if you still have the same problems. If so please document when the error occurs preferably while you have set ```seriallog 3``` to log as much serial info as possible.

Rgrds, Theo.

kenshin8, 2016-08-30:
Hi Theo,

Just tried the latest version 1.0.26, the problem still exist, have syslog level 4 on and some of the logs are below.

It looks like a connection problem b/w the device and the wifi network, when I browse around the web interface, when the page drop out and get reloaded, the log will only show http: handle xxxxx once more, then I keep browsing different page then the wifi connection dropped out completely.

This sonoff device is assigned by a reserved ip, I have 2 more other sonoff devices on the network using your version 1.0.14 without drop out problem, also 3 others nodemcu-12e board on the network using the same network config without drop out issue too.

Hope this can help to diagnose the problem. Thanks

Regards,
Ken

Logs: read from bottom to top

Wifi: Checking connection...
Wifi: Connected
Wifi: Checking connection...
HTTP: Handle Wifi config
Wifi: Connected
Wifi: Checking connection...
Wifi: Connected
Wifi: Checking connection...
Wifi: Connected
Wifi: Checking connection...
Wifi: Connected
Wifi: Checking connection...
HTTP: Handle config
HTTP: Handle MQTT config
Config: Saved configuration to flash at FA and count 883
HTTP: Handle config
HTTP: Logging Seriallog 3, Syslog 4, Host 192.168.0.80, Port 514
HTTP: Parameter save
HTTP: Handle Log config
HTTP: Handle Log config
HTTP: Handle config
HTTP: Handle root

kenshin8, 2016-08-30:
Forgot to mention, it works perfectly fine when it connect via FTDI, no drop out issues at all

Ken

arendst, 2016-08-31:
Ken,

From the log I see nothing strange.

I tried using two different sonoffs with DHCP IP address with and without FTDI connection with seriallog 4 and syslog 4 (and weblog 4 which is soon to be released) and IDE debugging to ALL but the webpages never show any problems.

I can not explain the relation between FTDI and your web problem other than a power issue but then I would expect the opposite of what you experience (Fail with FTDI power but OK with sonoff power).

 Are you sure there are no two devices with the same hostname? It might be a router problem too.

Sorry I can't help.

Rgrds, Theo.

kenshin8, 2016-08-31:
Hi Theo,

I was aware of the conflict in hostname, mqtt client id that can lead to a drop out, but this is not the case, I tried to turn off all other devices on the network and this is still happening, how can i get all your previous versions so I can test it one by one to see which version I start to have issue?

Regards,
Ken

kenshin8, 2016-08-31:
I can pretty much confirm the problem is from the wifi connection, as I can not ping the ip most of the time, I will try to track down the code and see where goes wrong.

Pinging 192.168.0.104 with 32 bytes of data:
Request timed out.
Request timed out.
Reply from 192.168.0.104: bytes=32 time=4ms TTL=128
Request timed out.

And the log just keep showing this when its idle.
Wifi: Connected
Wifi: Checking connection...
Wifi: Connected
Wifi: Checking connection...
Wifi: Connected
Wifi: Checking connection...
Wifi: Connected
Wifi: Checking connection...

Ken

arendst, 2016-08-31:
The log messages are expected. Every 20 seconds it checks the wifi connection. If it's OK it returns Connected. This is just a message not an error. Setting logging to 3 makes them disappear.

kenshin8, 2016-08-31:
So do you have more ideas what can cause the drop out? I have 3 other nodemcu using <ESP8266WiFi.h> but doesn't have this problem. Any hint will be appreciate. 

Thanks
Ken

kenshin8, 2016-08-31:
using ESP8266WiFi.h i mean

arendst, 2016-08-31:
Does it only happen on this device with version 1.0.26?
What happens with version 1.0.14 installed on this device?
Is the device near the router?

kenshin8, 2016-08-31:
It only happens on 2 devices with 1.0.26, I can notice the response time when controlling the light through mqtt on 1.0.14 device is much faster than the other 2 with 1.0.26. All the devices are sitting in the same room and only 5 meters away from router. 

The 2 devices with 1.0.26 is sonoff 240v and the other 2 is sonoff 12v, don't know if this is going to make the different.

arendst, 2016-08-31:
You could try to recompile 1.0.26 without the webserver code by changing line 45 in sonoff.ino from 
```
#define USE_WEBSERVER
```
to
```
//#define USE_WEBSERVER 
```
The webserver is a nice to have anyway. If you use MQTT there is no need for it. That's why I was reluctant to implement a webserver.

I only have 240V sonoffs and they do not behave like yours.

arendst, 2016-08-31:
Also when all is running fine there is no need to have logging set to 3 or 4.

kenshin8, 2016-08-31:
Unfortunately the same problem still exists, it is so strange that the device works perfectly fine after it flashes through ftdi, until it is unplugged and reconnect to 240v and all the problems start to happen, the device take about 30secs to connect to network and losing its ping reply. 

I then plug it back in to fdti and everything works again, connect to wifi within seconds, perfect ping response time.

I think we can rule out the problem from your code, may be I have a faulty hardware, and 2 in  a row.

Thanks for spending time on my issue, Theo.

kenshin8, 2016-08-31:
This is my sonoff 240v board info anyway.

Version	1.0.26
Core version	2_3_0
SDK version	1.5.3(aec24ac9)
Flash write count	41
Boot count	1 
 
ESP Chip id	8502290
Flash Chip id	1327343
Flash size	1024 kB
Sketch size	317 kB
Free sketch space	620 kB
Free memory	35 kB

arendst, 2016-08-31:
This is mine:

Version	1.0.27
Core version	2_3_0
SDK version	1.5.3(aec24ac9)
Uptime	0 Hours
Flash write count	138
Boot count	12
Reset reason	Software/System restart
 
Hostname	sonoff1
IP address	192.168.2.213
Gateway	192.168.2.254
MAC address	5C:CF:7F:0C:A5:5C
 
ESP Chip id	828764
Flash Chip id	1327343
Flash size	1024 kB
Sketch flash size	1024 kB
Sketch size	318 kB
Free sketch space	620 kB
Free memory	35 kB

closed 2016-08-31 by arendst

kenshin8, 2016-09-03:
Hi Theo,

Just to give you an update on my issue, tried everything to isolate devices on the network that might causing the drop out, at the end, it was the wifi compatible issue with ubiquiti unifi access point and sonoff 240v, it works perfectly fine when it connect to other wifi router on the same network, so it means nothing on the network is causing the conflict, I was looking at IP, hostname, etc....I have no idea why the sonoff 240v doesn't work on a newer AC access point, but the sonoff 12v works fine. This is just an update for you so this might help in your future development.

Regards,
Ken

dawsmac, 2017-04-18:
Hey Ken, 

I noticed you are having the same issue as me,  I have now added in the below lines giving the unit a static ip and uploaded the code.. have rebooted about 20 time in different location around the house and successful every time. not sure if this is clean, by all means correct me :) 

FYI i have the same AP  ubiquiti unifi access point.... 

hope this helps you out ? 
  
    **IPAddress ip(192, 168, *, *);**
    **IPAddress gateway(192, 168, *, *);**
    **IPAddress subnet(255, 255, 255, *);**

***********************************************************************************

    btn_timer.attach(0.05, button);
    mqttClient.set_callback(callback);
    WiFi.mode(WIFI_STA);
    WiFi.begin(WIFI_SSID, WIFI_PASS);
    **WiFi.config(ip, gateway, subnet);**
    ArduinoOTA.onStart([]() {

kenshin8, 2017-04-19:
Thanks for letting me know, I have my router reservedthe ip for each sonoff mac address, so similar to static ip but it didn't work out, my solution is to connect those sonoff to my other router and it works fine since, so I didn't bother to find out what happen.