arendst/Sonoff-MQTT-OTA-Arduino #7: No response when Mosquitto service is stopped

RobKuipers, 2016-05-23:
Hi Arendst,

first of all a big THANK YOU for your work.
I tried ESP Easy also, but cannot live with the instabilty which undermines the WAF...

I noticed a small problem which might (or not) apply to your applications of Sonoff too: when the MQTT service cannot be reached, the software remains in a loop too try and (re-)connect. During that time it is unresponsive to the push-button. 
In my case this mains IT-interventions are neccesary to get the household running again. The other members of the family seem to see this as an issue :-)

I think things would work even better, if manual operation remains possible, even when MTQQ cannot be reached. 

How do you feel about this?

Kind regards,
Rob

arendst, 2016-05-24:
Rob, I agree. I'll look into it. Everything for a high WAF!

arendst, 2016-05-24:
Implemented better MQTT reconnect functionality to prevent button lock-out. As the MQTT connect process can take some seconds, depending on the MQTT server malfunction type, a small button lock-out may still be observed.
Please give it a try.

RobKuipers, 2016-05-24:
Hi Theo,

Wow, that was quick!
So I tried and be just as prompt: tested. Works like a charm. 
Like you said, some delays when WiFi or MQTT server are out, but it's hard to think of an even better solution.

Using this combined with Mosquitto, Node-red and Domoticz. 
Seems as reliable as my Zwave switches. Only (a lot) cheaper :-)

Included my node-red configuration for inspiration (or feedback).

Thanks a lot,
Rob
[Node-Red Sonoff to Domoticz.txt](https://github.com/arendst/Sonoff-MQTT-OTA-Arduino/files/280187/Node-Red.Sonoff.to.Domoticz.txt)

arendst, 2016-05-25:
Rob, nice integration with Domoticz. I use another approach with both bash and lua scripts available from my mqtt-domoticz repository with almost the same result.

closed 2016-05-25 by arendst