arendst/Sonoff-MQTT-OTA-Arduino #8: Intermittent Relay Off-On

deswong, 2016-05-27:
Hi,

Big thumbs up for the effort to make this available to the everyday tinkerer :) This has allowed me to start to delve into HA and actually make stuff work!

Question though, I have an issue which is noticed when a light is on, intermittently the relay will click off and then back on. It is very quick, and I don't know what could be causing this.

On my HA (OpenHAB) I have taken a look at logs and there is no indication of a change in MQTT status, and in Mosquitto logs it does not show a change in MQTT status for the SonOff, but the relay does click off and back on. 

I don't have a consistent timeframe either that I can put this down to.

Any hints where I can start looking? Could it be the SonOff is actually rebooting and then restoring its last known state? Or would this take longer for it to do so and not a click-click that is less than 1/2 second?

Thanks again for the code.

Des.

arendst, 2016-05-27:
Hi Des.

A pity you experience this phenomenon. It looks like it's indeed a reboot of sonoff. You can see the same result when you execute the restart 1 command.
To be sure you could monitor MQTT for the hourly HEARTBEAT message send by sonoff at around 2:30 in the hour. This is a counter which will be reset every reboot.
To further investigate you may want to monitor the serial interface for possible reboot information.
Newer versions of the code also support UDP syslog which may provide some info too.

A wifi problem may eventually lead to a reboot as sonoff will try to reconnect, if it fails it starts smartconfig which in turn will reboot after 60 seconds (latest version).

Happy hunting ;-)

deswong, 2016-06-23:
Hi,

It does look like it is rebooting as the counter is being reset. I've updated the code on there as well to try and hunt down the issue.

Thanks for your help!

closed 2016-06-23 by deswong