arendst/Sonoff-MQTT-OTA-Arduino #9: Are you plane support ITEAD SonOff with 433 mHz ?

Lstt2005, 2016-06-07:
Good day! Thank's for you work. A you plan support firmware with integtrated 433 mHz module https://www.itead.cc/smart-home/sonoff-rf.html  ?

arendst, 2016-06-07:
Hi, 

I do not have a sonoff RF version.
While looking at the schematics of the Sonoff RF version I would expect that the RF version works perfectly fine with my code as all the RF hardware does is injecting the button function. I may have missed someting though.

Rgrds, Theo.

Lstt2005, 2016-06-07:
Ok, thank you!

fbeek, 2016-06-09:
Hi,

i am using the code with the RF variant. A far as i can see it works like charm without any changes. Except the button function (only on/off is working because you cant control the E/FW pin directly)

Regards Flo

closed 2016-07-24 by arendst