arendst/Sonoff-MQTT-OTA-Arduino #10: WIFI connected, but no serial or mqtt response

fbeek, 2016-06-09:
Hi,

i have a problem with my Sonoff RF devices.
I compiled your software and flashed it, the led blinks, the button works, the rf remote works and WIFI is connected and i can ping the device.

The device is connected to the mqtt server and i could control it. But after a restart of the MQTT Server there comes no response anymore. So i connected the device to a serialport and powered it via a lab psu. In the Serial Monitor i see the start message from the ESP but nothing from your software (and yes i changed the baudrate between the tests but there is nothing :-) ). if i type a command via serial i get a response message via mqtt (but nothing on the serial line) until i restart the mqtt server. Then there comes nothing to.

The USB connection should be ok because i can see the boot message from the esp until "ld"

So i have two problems:

1.) No Serial response on commands
2.) No MQTT reconnect and only sporadic function

I hope you can help me.

Thank you for your great work !!!

Regards

Flo

arendst, 2016-06-09:
It seems a lot is functioning fine ;-)

Try the **seriallog 4** command as documented in the readme.md. This should show a lot more information.

Rgrds, Theo.

fbeek, 2016-06-10:
Hi,

yesterday i tried your advise an compiled the software with the serial log level set to 4 and flashed the device.

Here the result:

- It is connected to the Wifi AND to the MQTT server (I can pin the device an see the connection in the mosquitto log)
- On the serial line i only get the boot message from the ESP but nothing from the program.
(I changed the empty Serial.println() to Serial.println("Start"), even this message is not showing up in Monitor)
- If i send a command over the serial line i get the echo from the program but nothing more
- Nothing is showing up on the MQTT site, no heartbeat, no command response or status. Nothin.

I tried the Blink example and a serial port test program, the work like expected.

Do i have to wipe the flash before the program upload?

Regards, Flo.

fbeek, 2016-06-10:
Ahhh.

Sorry, this issue can be closed :sweat_smile: there was a typo in my config....
Thanks for your awsome work.

Regards, Flo

closed 2016-06-10 by fbeek