arendst/Sonoff-MQTT-OTA-Arduino #11: Have you considered a different pubsubclient library

KmanOz, 2016-06-15:
Hi

Great work. I can see a lot of thought has gone into this. Have you considered using lmroy/pubsubclient? I ask because it expands on the original work that Nick O'Leary did and adds a whole bunch of classes, in particular the ability to send QOS 1 & 2 messages, keepalive settings in the sketch etc etc, Would really round off your work quite nicely.

Otherwise great job.

arendst, 2016-06-30:
I've tried AsyncMqtt and had the code working but only with workarounds to the then available version.

I'll look into your suggested pubsubclient in the near future.

eazuka, 2016-10-11:
I think @KmanOz is spot on, please look into it. great work

ionciubotaru, 2016-11-13:
https://github.com/Imroy/pubsubclient - it's a good library, you can perform OTA over MQTT and send QOS 1 messages.

TotallyInformation, 2016-11-19:
It would also be really helpful if the telemetry output used a Last Will & Testement with a timeout related to the telemetry period.

In that way, if the Sonoff device hasn't responded after the telemetry period has passed, the telemetry message could switch to "Offline" (if using messageformat 1) or something similar otherwise.

I use this extensively with other devices and it is great for being able to alert when devices go offline.

arendst, 2016-11-23:
LWT message updated to optional JSON format. To sync LWT message with telemetry messages you'll have to change define MQTT_KEEPALIVE in PubSubClient to the same value as TELEPERIOD.

closed 2017-02-11 by KmanOz