arendst/Sonoff-MQTT-OTA-Arduino #14: Button press behavior

hermannw, 2016-07-10:
I have two Sonoff units that I flashed with this software.
For some reason there is a different behavior of the button press when the units are connected directly to the FTDI converter. In this case one short button press does exactly what the description mentioned:  It toggles the relay on/off and sends an MQTT message.
This behavior changes when the units are powered from mains. In this case one short button press does nothing except for the LED blinking twice. No relays, no MQTT message. Two short presses on the other hand switch the relay and send stats/LIGHT on/off.

Any idea why this happens?

Would you care to elaborate on the differences between these three points:
**a short press** toggles the relay. This will blink the LED twice and sends a MQTT status message like **stat/sonoff/POWER on**

**a short press** toggles the relay either by sending a MQTT message like cmnd/sonoff/light 2 or directly. This will blink the LED twice and sends a MQTT status message like **stat/sonoff/LIGHT on**

**two short presses** toggles the relay. This will blink the LED twice and sends a MQTT status message like **stat/sonoff/POWER on**

arendst, 2016-07-11:
There should be no difference between mains powered or FTDI powered other than that FTDI powered may be not be enough power to switch the relay.

I suppose you use the latest version 1.0.13. 

The correct description for one short press is the second point; I'll fix the text to remove the first point.

The difference between one or two short presses is that with version 1.0.13 one short press allows for sending a MQTT message to a server which in turn should send a MQTT message back to switch the relay. This behaviour will occur when the BUTTONTOPIC is not set to 0. In that case a MQTT message with the topic set with BUTTONTOPIC will be send. The server is expected to respond with the correct sonoff MQTT switch topic. I use it with domoticz and my mqtt-domoticz script.
If BUTTONTOPIC is set to 0 the behaviour is the same as with two short pressed; the relay is switched immediatly and a status message is send.

If BUTTONTOPIC is not set to 0 the behaviour you experience may occur as the server may not recognize the command and will not send the switch command to sonoff.

hermannw, 2016-07-17:
Thanks for the answer. 

My mistake was, that I was listening to messages with "stat/#". Then I realized that the button messages start with "cmnd/#". :-0

Just out of curiosity: In the readme you mention several times a file "user_config.h". Somehow I could not find this file... Do you mean config.ino?

Thanks for this great piece of software and for making it available to us!!

Hermann

arendst, 2016-07-18:
Hi Hermann,

I'm glad you fixed it. The reference to "user_config.h" is a result of copy/paste from my plain C implementation. It should be "sonoff.ino" for the Arduino version. Thanks for attending me.

closed 2016-07-18 by arendst