arendst/Sonoff-MQTT-OTA-Arduino #15: Sonoff crashing/looping with Exception (28) error.

Leeev, 2016-07-18:
Hi Theo,

I've just installed your code to a new Sonoff module. It compiles and uploads fine, but doesn't connect to the wifi. All the parameters appear to be correct, but the esp8266 appears to crash and the restart in a permanent loop. The output from the serial console is -
```
 ets Jan  8 2013,rst cause:1, boot mode:(3,2)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld
)

Exception (28):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x001bd9b1 depc=0x00000000

ctx: cont 
sp: 3ffeff30 end: 3fff0270 offset: 01a0
```
Then some more stack data then it starts again.

The Sonoff hardware seems to be OK as I can install the ESPeasy firmware and that seems to work OK, as does the SonoffWebserver firmware. The power supply seems to be ample as I can remotely use the relay with SonoffWebserver. Both of these firmwares connect to the wifi correctly too. Sadly neither have the functionality of your MQTT/OTA firmware.

All this ESP hardware/software is new to me as I'm a long time AVR/Arduino user, and googling hasn't produced anything useful, so I thought I'd cut to the source to see if there's anything fundamentally wrong.

arendst, 2016-07-18:
What worries me is the boot mode:(3,2). Mine says boot mode:(3,1) when I restart:
```

 ets Jan  8 2013,rst cause:1, boot mode:(3,1)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld
þ
App: Project sonoff (Topic sonoff3, Fallback DVES_0CAB6F, GroupTopic sonoffs) Version 1.0.13

```
Are you sure to select Flash Size: "1M (64K SPIFFS)" as sonoff only has 1Mb flash?

Leeev, 2016-07-18:
Hi Theo,

Yes, it's definitely on 1MB/64K. I've just tried it again to be double sure.
I've tried various other settings and it didn't work on those.

Oddly, after I rebooted this time I got a different error -
```
 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld
)

Exception (28):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x001bd9b1 depc=0x00000000

ctx: cont 
sp: 3ffeff30 end: 3fff0270 offset: 01a0
```

The only difference I can see is that I think I have a later version of the Sonoff as I have 5 holes where the USB/TTL adapter is plugged in. This new extra hole is apparently GPIO1. Not sure if that makes any difference in the firmware or not?

arendst, 2016-07-18:
The extra hole should not make a difference.

What you could do is erase the flash completly using the Arduino-esp8266 esptool.exe tool. Mine is located in folder ```C:\Data2\Arduino\arduino-1.6.9-esp8266-2.3.0\portable\packages\esp8266\tools\esptool\0.4.9```

Steps to take:
0) Preparation: I have a FTDI connection with only ground, Tx and Rx to COM8 on my PC.
1) Power off then on sonoff while pressing the button to put it in upload mode.
2) The erase command for sonoff connected to COM8 is ```esptool -cp COM8 -bz 1M -ce -vv```
3) Power off then on sonoff while pressing the button to put it in upload mode.
4) Compile and upload via IDE and COM8 sonoff code to sonoff.
5) Power off then on sonoff to execute sonoff code just uploaded.

If all's well sonoff should now work, at least that's what mine did after the excercise just mentionend.

Leeev, 2016-07-18:
OK, I dropped back to Win10 (normally Ubuntu), installed a new Arduino IDE, added the ESP boards, ran the esptool in program mode, which seemed to work fine, restarted the ESP in program mode again and compiled/downloaded the firmware again. Restarted the ESP again and it's still doing the same thing boot mode (3,7).
```
c:\Users\Leeev\Desktop\arduino-1.6.9\portable\packages\esp8266\tools\esptool\0.4.9>esptool -cp COM3  -bz 1M -ce -vv
esptool v0.4.9 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
        setting port from COM1 to COM3
setting flash size from 512K to 1M
        espcomm_erase_flash
        setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
        flush start
        setting serial port timeouts to 1 ms
        setting serial port timeouts to 1000 ms
        flush complete
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
        read 0, requested 1
trying to connect
        flush start
        setting serial port timeouts to 1 ms
        setting serial port timeouts to 1000 ms
        flush complete
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
        setting serial port timeouts to 1000 ms
        setting serial port timeouts to 1000 ms
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
        espcomm_send_command: receiving 2 bytes of data
```
Sketch compiles and uploads OK
```
Sketch uses 281,321 bytes (29%) of program storage space. Maximum is 958,448 bytes.
Global variables use 36,872 bytes (45%) of dynamic memory, leaving 45,048 bytes for local variables. Maximum is 81,920 bytes.
Uploading 285472 bytes from C:\Users\Leeev\AppData\Local\Temp\build8d192b5721aa8863e713a0a9478e87d6.tmp/sonoff.ino.bin to flash at 0x00000000
................................................................................ [ 28% ]
................................................................................ [ 57% ]
................................................................................ [ 86% ]
.......................................                                          [ 100% ]
```
Now, I'm stuck... :(

Leeev, 2016-07-18:
OK, so I had an idea - to flash the OTA compiled bin into the Sonoff using the esptool.
It works and the Sonoff doesn't crash/reboot like crazy.
However, I think it may still be crashing, just slowly, as I'm getting the relay clicking every minute or so and this output appears on the serial console -
```
 ets Jan  8 2013,rst cause:1, boot mode:(3,3)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x0f
csum 0x0f
~ld
```
Another Sonoff I have is also rebooting, but oddly no clicking from the relay, just the LED flashing, even on mains power  -
```
 ets Jan  8 2013,rst cause:1, boot mode:(3,0)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x0f
csum 0x0f
~ld
```
I'm not sure what this means. Is the build environment wrong? Or has the Sonoff hardware changed?
It's odd that other firmware works fine on it though.

PuceBaboon, 2016-07-18:
Theo,

   It might be worth noting that Pete Scargill (
http://tech.scargill.net/esp-sdk-and-rboot-woes) is seeing some strange
behaviour with the latest 1.5.4 and 1.5.4-1 SDK with his application.

   I've also just received a comment (yesterday) on my site from a user
complaining of the same "exception(28)" errors.  See the comments section
at:-

https://esp8266hints.wordpress.com/2016/04/04/no-phone-sonoff-stuff/

   The submitter,, "Mark", also seems to have just built a clean ESP
environment from the latest available packages.

                                      -John-

Leeev, 2016-07-19:
Hi John/Theo,
I've got the 1.6.9 Arduino IDE installed (on Ubuntu/Win10), I used the BoardMaker to install the ESP dev tools, the version is 2.3.0. This agrees with Theo's readme file. 
The SDK under it seems to be 1.5.3_16_04_18 rather confusingly. And if I'm reading this right isn't affected by Pete's issue, althought he calls it 1.53.

As per the post on your site John - 
I changed the CFG_HOLDER value and recompiled, it changed the error code, but still crashed/rebooted. With nothing left to lose I also reverted to ESP 2.2.0 and tried it again, still no joy.

I've also tried Pete Scargill's older pre-compiled roms. They download to the Sonoff OK, and seem to  kinda work -
```
ESP Starting...
GPIO4 and 5 are outputs.
Current web programming pin: 2
GPIO13 is a LED indicator.
Software version 1.5.01
Use the {debug} command for more information.
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 1344, room 16 
tail 0
chksum 0x9c
load 0x3ffe8000, len 660, room 8 
tail 12
chksum 0xb8
csum 0xb8

rBoot v1.3.0 - richardaburton@gmail.com
Flash Size:   4 Mbit
Flash Mode:   QIO
Flash Speed:  40 MHz
rBoot Option: Big flash

Booting rom 0.
{lG

ESP Starting...
GPIO4 and 5 are outputs.
Current web programming pin: 2
GPIO13 is a LED indicator.
Software version 1.5.01
Use the {debug} command for more information.

STATION mode
Web page control enabled
Waiting for Access Point beyond
... x16
Waiting for Access Point beyond
Switching to AP: beyond

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 1344, room 16 
tail 0
chksum 0x9c
load 0x3ffe8000, len 660, room 8 
tail 12
chksum 0xb8
csum 0xb8

rBoot v1.3.0 - richardaburton@gmail.com
Flash Size:   4 Mbit
Flash Mode:   QIO
Flash Speed:  40 MHz
rBoot Option: Big flash

Booting rom 0.
sdǃr$ğ

ESP Starting...
GPIO4 and 5 are outputs.
Current web programming pin: 2
GPIO13 is a LED indicator.
Software version 1.5.01
Use the {debug} command for more information.

STATION mode
Web page control enabled
Waiting for Access Point beyond
etc
```
But I don't seem to have serial control, so I can't setup the parameters for wifi/mqtt/etc.

It's a good job I didn't buy 5 of these Sonoffs to use with MQTT on my Pi... Oh, wait... I did... :(

PuceBaboon, 2016-07-19:
Leeev,

   Interesting.  Something, somewhere has changed.  If you have the top
open on one of your units, can you check the part number on the EEPROM chip
(the one on the top of the board, next to the switch and LED).  On my
models its a Winbond 25Q80DVSIG (8Mb/1MB).

   Is the version number printed on the board (beside the relay) still
"V1.0"?

                   -John-

arendst, 2016-07-20:
I noticed too that rBoot calls out 4Mbit which is a 512kByte eeprom. 

Mark please try to compile with 512K (64K SPIFFS) and see if that works.

Leeev, 2016-07-20:
Hi Theo,

Tried 512K/64K it didn't seem to make a difference compiling uploading your code.

Hi John,
 
I think we have a new hardware revision... :(
The flash chip I have is a Winbond 25Q80BVSIG. From what I can make out from the data sheets it looks pretty similar at 1MB.
There is no "V1.0" silk screen on the PCB. Next to output terminal block it does say "Sonoff TH".
However, it is definitely NOT the Sonoff TH (10/16) as featured on ITeads website, it was ordered as the "Sonoff - WiFi Wireless Smart Switch For MQTT COAP Smart Home" direct from ITead.

Leeev, 2016-07-20:
Theo/John,

I thought some photos may help -
https://drive.google.com/open?id=0ByprFMkriUdEX1kxbE9BcVV6Rzg

KmanOz, 2016-07-20:
Hi

I'd like to chime in. I have the exact same Sonoff's and can confirm I have had them working with Theos firmware. I'm using my own now but definitely were working fine. You say you have 4 of them, are all of them doing the same thing?

Leeev, 2016-07-20:
Hi Kman,
I've got five and the first two that I've soldered pin headers into are exhibiting this behaviour.  
Just to be clear, yours is exactly same as my pictures with the "Sonoff TH" silk screen AND the "25Q80BVSIG" flash memory?

KmanOz, 2016-07-20:
Exactly the same. I only got mine 4 or 5 weeks ago.

Leeev, 2016-07-20:
Excellent! That rules that out then. 
What OS and what version of the Arduino IDE/ESP tools are you using?

KmanOz, 2016-07-20:
I'll check now but just a thought. Do you have any special characters in your SSID or Password?

KmanOz, 2016-07-20:
Windows 10, Arduino 1.6.9 in portable mode, and ESP Boards v 2.2.0

Leeev, 2016-07-20:
Ah, I've not tried that combination, I'll have at go with that tonight.
As for the SSID/Password, no it's all upper/lowercase letters and numbers.
Just a thought though, I've put IP addresses into the strings for the SYS_LOG host, MQTT host and OTA_URL, as they don't have DNS entries. Could that be an issue?

KmanOz, 2016-07-20:
I used IP addresses too

Leeev, 2016-07-20:
Right then. KmanOz said he was using his own firmware now, and quick search on GitHub revealed Sonoff-HomeAssistant. In about three minutes, I've downloaded it, updated the params, realised I needed the Imroy pubsubclient and flashed it to the Sonoff.
```
ESP ChipID: 1BD9B1
Connecting to BigEnough Wifi . . . . . . . . . . WiFi FAILED!
----------------------------------------------------------------
mqtt broker connection . . . . . . . . . . LOST
 ets Jan  8 2013,rst cause:1, boot mode:(3,7)
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
------------------  Sonoff Powerpoint v1.0p  -------------------
ESP ChipID: 1BD9B1
Connecting to BigEnough Wifi . . . . . . . . . . DONE
IP Address is: 10.0.1.211
Connecting to 10.0.1.1 Broker . . DONE
----------------------------  Logs  ----------------------------
mqtt broker connection . . . . . . . . . . OK
Relay . . . . . . . . . . . . . . . . . . ON
Relay . . . . . . . . . . . . . . . . . . OFF
Relay . . . . . . . . . . . . . . . . . . ON
Relay . . . . . . . . . . . . . . . . . . OFF
mqtt broker connection . . . . . . . . . . OK
Relay . . . . . . . . . . . . . . . . . . ON
mqtt broker connection . . . . . . . . . . OK
Relay . . . . . . . . . . . . . . . . . . OFF
mqtt broker connection . . . . . . . . . . OK
etc
```
So one reboot, presumably due to a failed wifi connection, and it's been up for over two hours now.
It's powered by the USB adapter, it's connected to my MQTT server and I have a digi meter (set to continuity) plugged into the relay, that I can make beep via MQTT. I have it beeping every five mins... 

What this says about the hardware or the libraries used by Sonoff-MQTT-OTA-Arduino I'm not sure...

KmanOz, 2016-07-20:
I'v been playing for ESP8266's for a while now. I'v noticed that some on initial boot fail to join WiFi successfully and more annoyingly it's sporadic. I can't even begin to understand why because the same firmware in another module with the same circuit works every time. It's some sort of timing issue internal to the chip or possibly flaky silicon. I just don't know.

As for software I suggested to Theo to use lmroy's library a while ago mainly because it supports QOS1 and has a a handful of extra class's that make it more useful. This firmware together with the lmroy library would be killer combination.

arendst, 2016-07-21:
I'm glad with all feedback given above. I'm sorry that my code doesn't work for you Leeev. I might take a look at Imroy's library but I stayed away from it as it is based on an older version of knolleary's pubsubclient. The current version of pubsubclient worked fine with me.

Leeev, 2016-07-24:
Hi Theo,
These things happen. It's a bit of odd one though. I tried to use Kman's code with Knolleary's library to see if that was the issue, but there appear to changes to the parameters so it won't compile, and it's bit beyond me to correct it. Kman's code has been running on the Sonoff's for a few days now, so they seem OK, but I've love to know what's changed and where.
KmanOz,
If you're still here, I'd love to take you up on your offer of the code that supports the new GPIO14 header pin - I'd like to attach a light sensor to monitor when the pump that's attached to the first Sonoff cuts out. Taking the header lead out of the side of the case between the LED & the switch means it won't go near the 240V mains tracks (which I've also insulated - just to be sure).

arendst, 2016-07-28:
Leeev,

I introduced additional buffer overflow checks which may have led to your experience. Version 1.0.15 now available might work this time on your hardware. Feel free to check it out.

Leeev, 2016-08-02:
Hi Theo,
I've just tried your latest release 1.0.17, something has changed -
```ets Jan  8 2013,rst cause:1, boot mode:(3,0)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld
Exception (28):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x001bd9b1 depc=0x00000000

ctx: cont 
sp: 3ffeff50 end: 3fff0290 offset: 01a0```

But now the relay changes on/off as the Sonoff reboots. I left it going for a few mins, and tried power cycling it several times, but the same thing happens each time. :(

arendst, 2016-08-02:
Leeev,

thanks for trying the latest code. Pity it still doesn't work. 

What you could do is compile the code again but this time change in file ```user_config.h``` parameter 
```CFG_HOLDER``` to ```0x20160731``` and ```SERIAL_LOG_LEVEL``` to ```LOG_LEVEL_DEBUG_MORE```.
This way all default parameters are rewritten and on the serial interface more information should be shown before sonoff goes dead again.

BTW: you said using the pre-compiled version sonoff seemed to work but rebooted every minute. This is expected as the default SSID and Password are wrong. Sonoff starts SmartConfig and waits for a minute to receive valid information. If not it reboots and the sequence starts again. While this is happening you can still enter serial commands like ```seriallog 4``` followed by command ```ssid``` and your wifi ssid and command ```password``` with your wifi password. It may reboot in the meantime but at least you must be able to see info on the serial interface as long as ```seriallog``` is 4. Using the Arduino IDE be sure to set baudrate to 115200 and Both NL & CR.

Leeev, 2016-08-02:
OK... once more with feeling... :)

With the new CFG_HOLDER and SERIAL_LOG_LEVEL -
It's not clicking the relay anymore, but still looping on restart...
```
00:00:00 Config: Loaded configuration from flash at FA and count 1
00:00:00 Wifi: Connecting to BigWorld as rainpump

Exception (28):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x001bd9b1 depc=0x00000000

ctx: cont 
sp: 3ffeff40 end: 3fff0280 offset: 01a0

>>>stack>>>
3fff00e0:  3fff159c 3fff0308 00000001 00000001  
3fff00f0:  4021ee72 3fff159c 000000fc 4022b6c8  
3fff0100:  00000018 001bd9b1 00000000 3fff0190  
3fff0110:  3fff01a0 40106ef0 3fff159c 3fff01d0  
3fff0120:  4023a841 00000080 3fff01d0 401004d8  
3fff0130:  0000000c 3fff031c 3fff0310 402088cc  
3fff0140:  4022b928 3ffeef3a 3ffe8463 40208c51  
3fff0150:  3fff01a0 3fff0190 0000000c 7fffffff  
3fff0160:  3ffef17c 3fff07b4 00000000 3ffeedd4  
3fff0170:  3ffef17c 3ffeeee9 3ffef00c 40208c88  
3fff0180:  3fff01a0 3fff0190 0000000c 4022ee84  
3fff0190:  4022cb66 3fff07b4 00000000 3ffe853d  
3fff01a0:  3ffef02a 001bd9b1 00000001 402023fc  
3fff01b0:  3ffef17c 3ffeeee9 3ffef00c 40203edc  
3fff01c0:  3ffef00a 3ffeeee9 feefeffe feefeffe  
3fff01d0:  3a707041 6f725020 7463656a 6e6f7320  
3fff01e0:  2066666f 706f5428 73206369 666f6e6f  
3fff01f0:  46202c66 626c6c61 206b6361 6e696172  
3fff0200:  706d7570 7247202c 5470756f 6369706f  
3fff0210:  feefef20 feefeffe feefeffe feefeffe  
3fff0220:  feefeffe feefeffe feefeffe feefeffe  
3fff0230:  feefeffe feefeffe feefeffe feefeffe  
3fff0240:  feefeffe feefeffe feefeffe feefeffe  
3fff0250:  feefeffe feefeffe feefeffe 3ffef258  
3fff0260:  3fffdad0 00000000 3ffef250 40208ac8  
3fff0270:  feefeffe feefeffe 3ffef260 40100718  
<<<stack<<<
```
Using the pre-compile binary, you're right, setting the NL&CR lets me enter the SSID/password, which it does accept and display on screen, but still fails to connect to the network. However, it then drops into SmartConfig, which should pop up a wifi access point I can connect to, but doesn't. :(
```
00:00:00 Config: Loaded configuration from flash at FA and count 5
00:00:00 Wifi: Connecting to BigWorld as esp-1bd9b1-sonoff
00:00:00 App: Project sonoff (Topic sonoff, Fallback DVES_1BD9B1, GroupTopic sonoffs) Version 1.0.17
00:00:01 Wifi: Checking connection...
00:00:01 Wifi: STATION_IDLE
00:00:02 Wifi: Checking connection...
00:00:02 Wifi: STATION_IDLE
00:00:04 Wifi: Checking connection...
00:00:04 Wifi: STATION_IDLE
00:00:05 Wifi: Checking connection...
00:00:05 Wifi: STATION_IDLE
00:00:06 Wifi: Checking connection...
00:00:06 Wifi: STATION_IDLE
00:00:07 Wifi: Checking connection...
00:00:07 Wifi: STATION_IDLE
00:00:08 Wifi: Checking connection...
00:00:08 Wifi: STATION_IDLE
00:00:09 Wifi: Checking connection...
00:00:09 Wifi: STATION_IDLE
00:00:10 Wifi: Checking connection...
00:00:10 Wifi: STATION_IDLE
00:00:10 RTC: sntp 0, Thu Jan 01 00:00:00 1970

00:00:01 Wifi: Checking connection...
00:00:01 Wifi: STATION_IDLE
00:00:02 Wifi: Checking connection...
00:00:02 Wifi: STATION_CONNECT_FAIL
00:00:02 Smartconfig: Started and active for 1 minute
00:00:10 RTC: sntp 0, Thu Jan 01 00:00:00 1970
```
Althought bizarrely, it did something different on one of the earlier loops and managed to find the network time...
```
00:00:06 Wifi: Checking connection...
00:00:06 Wifi: Connected
00:00:06 MQTT: Attempting connection...
00:00:06 MQTT: CONNECT FAILED, rc -2. Retry in 10 seconds
00:00:10 RTC: sntp 1470173149, Tue Aug 02 21:25:49 2016

21:25:56 MQTT: Attempting connection...
21:25:56 MQTT: CONNECT FAILED, rc -2. Retry in 10 seconds
21:26:05 Wifi: Checking connection...
21:26:05 Wifi: Connected
21:26:07 MQTT: Attempting connection...
21:26:07 MQTT: CONNECT FAILED, rc -2. Retry in 10 seconds
```

Leeev, 2016-08-02:
Hmmm... I've been playing a bit more, and typed help, then retried the SSID/password, and then typed Password on its own to see it would show the password, it connected shortly after this...
```
00:00:05 Wifi: Checking connection...
00:00:05 Wifi: STATION_IDLE
00:00:06 Wifi: Checking connection...
00:00:06 Wifi: Connected
00:00:06 MQTT: Attempting connection...
00:00:06 MQTT: CONNECT FAILED, rc 5. Retry in 10 seconds
00:00:10 RTC: sntp 1470176060, Tue Aug 02 22:14:20 2016

22:14:27 MQTT: Attempting connection...
22:14:27 MQTT: CONNECT FAILED, rc 5. Retry in 10 seconds
22:14:36 Wifi: Checking connection...
22:14:36 Wifi: Connected
22:14:38 MQTT: Attempting connection...
22:14:38 MQTT: CONNECT FAILED, rc 5. Retry in 10 seconds
22:14:49 MQTT: Attempting connection...
22:14:49 MQTT: CONNECT FAILED, rc 5. Retry in 10 seconds
```
I've set the MQTT server address, but I guess the username/password isn't correct, hence not connecting. But I see no parameters to set them from the console.

arendst, 2016-08-02:
Way to go!
There is no user ID or password needed for MQTT to work. (At least not for mosquitto as far as i'm aware).

All available commands Will be shown when you type syntax.

Try '''status 0''' for All parameters. See readme.md for More info.

Leeev, 2016-08-02:
Interesting... Status 0 gives -
```
22:39:10 status 0
22:39:10 MQTT: Receive topic cmnd/sonoff/status, data 0
22:39:10 MQTT: DataCb Topic sonoff, Group 0, Type STATUS, data 0 (0)
22:39:10 MQTT: sonoff/STATUS = 1.0.17, sonoff, 0, LIGHT, 0, 1
22:39:10 MQTT: sonoff/STATUS = PRM: sonoffs, DVES_1BD9B1, http://domus1:80/api/arduino/sonoff.ino.bin, 10.0.0.4, 0, 13
22:39:10 MQTT: sonoff/STATUS = FWR: Version 1.0.17, Boot 31, SDK 1.5.3(aec24ac9)
22:39:10 MQTT: sonoff/STATUS = LOG: Seriallog 4, Syslog 0, LogHost domus1, SSId BigWorld, Password ****************
22:39:10 MQTT: sonoff/STATUS = MEM: Sketch size 286080, Free 675840 (Heap 42208), Spiffs start 962560, Flash size 1048576 (1048576)
22:39:10 MQTT: sonoff/STATUS = NET: Hostname esp-1bd9b1-sonoff, IP 10.0.0.211, Gateway 10.0.0.1, Subnetmask 255.255.255.0
22:39:10 MQTT: sonoff/STATUS = MQT: Host 10.0.0.4, MAX_PACKET_SIZE 256, KEEPALIVE 120
```
But still the MQTT: CONNECT FAILED message

I've added extra debugging to the MQTT server, and it's not seeing anything, so I guess it's not connecting somehow. In the User_config.h there are MQTT_USER and MQTT_PASS parameters defined. Are these not used?

PuceBaboon, 2016-08-03:
Leeev,

   Can you actually ping the Sonoff from your machine where the MQTT broker
is running?

                -John-

arendst, 2016-08-03:
Leeev,

Both MQTT_USER and MQTT_PASS are being used during sign on to the mqtt server but they shouldn't be a showstopper as any userid or password should work.

Please try the above suggestion from John. 
If there's a firewall on your mqtt server configure it for access by sonoff.

About your comment regarding SmartConfig: It will NOT show up as an Access Point. You should use the ESP8266 SmartConfig App on your mobile connected to your Wifi as described in the README.md. It's just an alternative for the serial commands ```ssid``` and ```password```.

Regarding your clicking relay I was able to analyze the stack trace and reproducing it. I found that it is caused by the way you configure the MQTT_CLIENT parameter. My code expects a regular expression %06X as it is being replaced by the last 6 digits of the MAC address to keep all sonoffs unique in your MQTT environment. You changed it to "rainpump" which leads to the Exception (28). If you change it back to DVES_%06X you'll see that the exception is gone!

I'm working on a fix and expect to release it as version 1.0.19 any time soon.

Leeev, 2016-08-03:
John/Theo,
Yes, pinging the Sonoff from the RasPi works fine (if a little slow) - 
```
rpi@emonpi:~ $ sudo ping 10.0.0.211
PING 10.0.0.211 (10.0.0.211) 56(84) bytes of data.
64 bytes from 10.0.0.211: icmp_seq=1 ttl=128 time=865 ms
64 bytes from 10.0.0.211: icmp_seq=2 ttl=128 time=1.62 ms
64 bytes from 10.0.0.211: icmp_seq=3 ttl=128 time=1.76 ms
64 bytes from 10.0.0.211: icmp_seq=4 ttl=128 time=1.65 ms
64 bytes from 10.0.0.211: icmp_seq=5 ttl=128 time=2.21 ms
64 bytes from 10.0.0.211: icmp_seq=6 ttl=128 time=7.87 ms
64 bytes from 10.0.0.211: icmp_seq=7 ttl=128 time=4.62 ms
64 bytes from 10.0.0.211: icmp_seq=8 ttl=128 time=1027 ms
64 bytes from 10.0.0.211: icmp_seq=9 ttl=128 time=28.0 ms
```
SmartConfig app noted for later use. I think I confused the Access Point method of setup with another Sonoff firmware.

Using KmanOz's firmware, the MQTT on the RasPi sees/connects/works fine with the Sonoff, so I'm pretty sure it's not a server side issue.

PuceBaboon, 2016-08-03:
Leeev,

   Okay, that's one less thing to worry about (I've seen ARP issues with
some routers) and ESPs.

                                  -John-

Leeev, 2016-08-03:
Theo,
Further to your comment about any username/password for mqtt should work, in my case this is incorrect. I took KmanOz's working firmware and changed the mqtt username/password and compiled /downloaded it again, it didn't connect to the MQTT server. I corrected the u/p and it worked again.
I'm using the EmonSD (from the Open Energy Monitor .Org project) OS image for the Raspberry Pi.
This features the Mosquitto MQTT server. NodeRED (on the image) also seems to need the u/p.
Which MQTT server are you using?

arendst, 2016-08-04:
I'm using mosquito too without any issues regarding userid/password.

Anyway, the latest version 1.0.19 should run fine with your configuration (rainpump). In the source you can change both MQTT userid and password to your likings.

Just give it a go.

Leeev, 2016-08-05:
Theo,

That's fixed it!!! :+1:   
Compiles/downloads, connects to wifi, connnects to MQTT, and can be controlled!!
Many thanks for persistence. 

Looking at the EmonSD RasPi images install of Mosquitto, it seems to enable security and need username/password for devices to connect to MQTT, hence the issues. This would seem to be a good idea to stop rogue devices talking/listening to your MQTT server.

Many thanks!

closed 2016-08-06 by arendst