arendst/Sonoff-MQTT-OTA-Arduino #16: SmartConfig

Leo-C, 2016-07-26:
Hello,
all work properly (communication with MQTT broker is OK, and command are executed by sonoff and stats emitted), but I cannot use SmartConfig app (communication fail, but LED on sonoff is blinking correctly).

SmartConfig is useful to change WiFi settings (by documentation, changing  via MQTT commands SSID and pwd in 2 steps is more difficult because I must retain during config 1 old setting on WiFi network before change both).

How can I resolve this issue?

Many thanks in advance

Leonardo

P.S. as suggestion, also solutions like https://github.com/tzapu/WiFiManager can be interesting for my issue

arendst, 2016-07-26:
For debugging SmartConfig try the following:

- connecting sonoff serially and choose command ```seriallog 4```. This will output as much data as possible.
- Press the button three time quickly which should start smartconfig. Note the APP multipress count up to 3.
- Start SmartConfig App on phone and Confirm
- Verify serial data for status changes which should reflect new SSID and Password.

Do not forget to change seriallogging back to 0 with command ```seriallog 0```

Let me know if you see any anomalies.

Leo-C, 2016-07-26:
Nothing happens, but print at start (and some time after putting sonoff in SmartConfig mode):
```
 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld
   ¦
```
(sonoff is powered now by a serial-USB converter at 3.3 V;
command sent over by serial is only 'seriallog 4', isn't it? I've try also with MQTT command 'cmnd/sonoff/seriallog 4', but with no results)

Many thanks in advance

Leonardo

arendst, 2016-07-27:
A serial-USB converter as power source is not the ideal debugging tool as it cannot provide enough power for sonoff.

Although tricky (as some are afraid that mains power could get to your PC) I use the sonoff power supply for debugging by connecting only Gnd, Rx and Tx to the serial-USB converter. Following the below steps works for me (no mains power problems yet):

1. Disconnect sonoff from mains.
2. Open the sonoff case and connect ONLY three wires between Gnd, Rx and Tx to the serial-USB converter. NO 3V3 CONNECTION!!!
3. Close the sonoff case or move it to a location where you are sure not to touch any mains parts.
4. Connect the serial-USB converter to your PC, set baudrate to 115200 baud and power on sonoff
5. Watch some gibberish data on your screen
6. Type ```seriallog 4``` which should respond with something like:
```
seriallog 4
MQTT: sonoff/SERIALLOG = 4
Config: Saved configuration to flash at F9 and count 30
```
You can now debug SmartConfig issues.

In your case you might have set the baudrate to 74880 which explains the message you received but as sonoff will change to 115200 baud as soon as my code kicks in you may see no data at all.

Leo-C, 2016-07-27:
Hi,
I've connected also power supply (I've imagined that USB-serial converter can't feed enough sonoff, but I've heard relay switch ...), but result is unchanged.

I confirm also that serial port is set to 115200 baud (some words are gibberish, but many other seem well-written), but there was no effect after command 'seriallog 4' in terminal window (neither echo, if not set in terminal program) or via MQTT (I've subscribed to all messages with mosquitto client)

Maybe some wrong configuration during upload of firmware?

Thanks again

Leonardo

arendst, 2016-07-27:
Leonardo,
what version of my code are you using? The latest is 1.0.14

Leo-C, 2016-07-27:
The latest one
(I started working on this project about a week ago, and some day ago I've noticed you have put user config in a separated .h file, then I 've re-downloaded last master branch and I restart to build firmware)

arendst, 2016-07-27:
OK. Backtracking:
You are able to communicate with sonoff via MQTT.
When you perform MQTT command ```cmnd/sonoff/seriallog 3``` nothing happens on the serial interface. Mine shows:
```
MQTT: Receive topic cmnd/sonoff/seriallog, data 3
MQTT: DataCb Topic sonoff, Group 0, Type SERIALLOG, data 3 (3)
MQTT: sonoff/SERIALLOG = 3
Config: Saved configuration to flash at F9 and count 4
```
When you perform MQTT command ```cmnd/sonoff/power on``` still nothing happens on the serial interface. Mine shows:
```
MQTT: Receive topic cmnd/sonoff/power, data on
MQTT: DataCb Topic sonoff, Group 0, Type POWER, data on (ON)
MQTT: sonoff/POWER = On
Config: Saved configuration to flash at FA and count 5
```

Try MQTT command ```cmnd/sonoff/reset 2``` which should respond with:
```
MQTT: Receive topic cmnd/sonoff/reset, data 2
MQTT: DataCb Topic sonoff, Group 0, Type RESET, data 2 (2)
MQTT: sonoff/RESET = Erase, Reset and Restarting
Config: Erasing 186 flash sectors

 ets Jan  8 2013,rst cause:1, boot mode:(3,5)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld
á
Config: Loaded configuration from flash at FA and count -1
Config: Use default configuration
```

Try command ```cmnd/sonoff/restart 1``` which should respond with:
```

 ets Jan  8 2013,rst cause:1, boot mode:(3,5)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld
á
```

Try command ```seriallog 3``` on the IDE serial command line which should respond with:
```
seriallog 3
MQTT: sonoff/SERIALLOG = 3
Config: Saved configuration to flash at F9 and count 2
```

What version of Arduino IDE are you using? I have 1.6.9
What version of esp8266 are you using? I have 2.3.0
Have you set the IDE serial communication to Both NL & CR?

Leo-C, 2016-07-27:
First of all, versions of SW.
I use Arduino IDE 1.6.9 with esp8266 2.3.0; for serial communication I use putty or Arduino IDE (now I've set NL & CR).

Now setting log via MQTT works (before I've tried only via serial, but withouth correct carriage returns).
Going back to your previous indications, setting seriallog to level 4 and attempting to activate SmartConfig returns:
```
seriallog 4
MQTT: sonoff/SERIALLOG = 4
Config: Saved configuration to flash at FA and count 13
RTC: sntp 0, Thu Jan 01 00:00:00 1970

Wifi: Check connection
RTC: sntp 0, Thu Jan 01 00:00:00 1970

RTC: sntp 0, Thu Jan 01 00:00:00 1970

Wifi: Check connection
APP: Multipress 1
APP: Multipress 2
MQTT: Receive topic cmnd/sonoff/light, data 2
MQTT: DataCb Topic sonoff, Group 0, Type LIGHT, data 2 (2)
MQTT: sonoff/LIGHT = On
Config: Saved configuration to flash at FA and count 15
RTC: sntp 0, Thu Jan 01 00:00:00 1970

APP: Multipress 1
APP: Multipress 2
APP: Multipress 3
MQTT: Receive topic cmnd/sonoff/smartconfig, data 1
MQTT: DataCb Topic sonoff, Group 0, Type SMARTCONFIG, data 1 (1)
MQTT: sonoff/SMARTCONFIG = Smartconfig started
Smartconfig: Started and active for 1 minute
RTC: sntp 0, Thu Jan 01 00:00:00 1970

Smartconfig: SSID sonoff_iot and Password sonoff_iot
RTC: sntp 0, Thu Jan 01 00:00:00 1970

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld

Config: Loaded configuration from flash at F9 and count 16
Wifi: Connecting to sonoff_iot as esp-814c3e-sonoff
App: Project sonoff (Topic sonoff, Fallback DVES_814C3E, GroupTopic sonoffs) Version 1.0.14
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
RTC: sntp 0, Thu Jan 01 00:00:00 1970

Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Smartconfig: Started and active for 1 minute
RTC: sntp 0, Thu Jan 01 00:00:00 1970

Smartconfig: SSID sonoff_iot and Password sonoff_iot

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld
…
Config: Loaded configuration from flash at F9 and count 16
Wifi: Connecting to sonoff_iot as esp-814c3e-sonoff
App: Project sonoff (Topic sonoff, Fallback DVES_814C3E, GroupTopic sonoffs) Version 1.0.14
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
RTC: sntp 0, Thu Jan 01 00:00:00 1970

Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Smartconfig: Started and active for 1 minute
RTC: sntp 0, Thu Jan 01 00:00:00 1970

RTC: sntp 0, Thu Jan 01 00:00:00 1970

RTC: sntp 0, Thu Jan 01 00:00:00 1970
```

Then , what's gone wrong during SmartConfig, in your opinion?

Thanks for your patience!

Leonardo

arendst, 2016-07-27:
Glad you fixed the serial issues.

So from the SmartConfig info it seems your Wifi SSID is sonoff_iot and it's password is sonoff_iot too as that is what the sonoff receives from your SmartConfig app.

These should be the same values as the ones you use when correctly connected to MQTT. If not then your SmartConfig App is sending the wrong Wifi information.

What SmartConfig App are you using? I've only tested the one I mention in the README.md (https://play.google.com/store/apps/details?id=com.cmmakerclub.iot.esptouch).

That one works fine with me.

Leo-C, 2016-07-27:
Yes, SSID and password are OK.

I use a Raspberry as AP with a DNS to resolve host names specified on sonoff (actually disconnected from internet).
MQTT server is running on it , then using `mosquitto_pub` from Raspberry I don't use any server (localhost is default).

Finally, App is the one you've linked in README.md

Leonardo

arendst, 2016-07-27:
What do you do to make it communicatie with the MQTT server again?

Leo-C, 2016-07-28:
I'm not sure to have well-understood your question.

Sonoff communicate with MQTT broker using name written in firmware (resolved by a DNS running on Raspberry);
my commands are sent from Raspberry itself to localhost

Leonardo

arendst, 2016-07-28:
To be able to communicate with sonoff it needs a correct Wifi SSID, Wifi password and MQTT host. From the above SmartConfig log I can see that it didn't receive a correct Wifi SSID and/or Wifi password as it fails to connect with Wifi (```Wifi: STATION_IDLE```).

As you say Sonoff communicate with MQTT broker how do you make sonoff communicate with MQTT broker again while it fails Wifi communication?

If you type ```ssid``` and ```password``` on the serial interface what are the responses? You can do this even while SmartConfig is active.

NB. The message ```Wifi: Check connection``` means that sonoff is verifying the wifi connection and if it fails it will retry to connect or start SmartConfig after ```WIFI_RETRY``` (=16 in wifi.ino) attempts.

I expect you have changed the following parameters in ```user_config.h``` to match your wifi and MQTT environment:
```
STA_SSID    "sonoff_iot"
STA_PASS    "sonoff_iot"
MQTT_HOST   "name or ip address of your MQTT broker host"
```
These parameters can also be changed via the serial interface with ```ssid```, ```password``` and ```mqtthost```.

Can you please send the output of commands ```status 0``` until ```status 6```?

Leo-C, 2016-07-28:
Hello, SSID and Password are correct:
```
ssid
MQTT: Receive topic cmnd/sonoff/ssid, data 
MQTT: DataCb Topic sonoff, Group 0, Type SSID, data  ()
MQTT: sonoff/SSID = sonoff_iot
RTC: sntp 0, Thu Jan 01 00:00:00 1970

RTC: sntp 0, Thu Jan 01 00:00:00 1970

password
MQTT: Receive topic cmnd/sonoff/password, data 
MQTT: DataCb Topic sonoff, Group 0, Type PASSWORD, data  ()
MQTT: sonoff/PASSWORD = sonoff_iot
Wifi: Check connection
```
and also MQTT_HOST has to be correct (elsewhere It cannot receive messaged from MQTT broker)

Maybe sometime it lost connection because low power emitted from Wifi dongle connected to Raspberry? (but is few cm far away ...)

Furthermore, each time I launch an MQTT command, it seems that sonoff switch immediately (also if MQTT store messages)

Regarding status messages:
```
status 1
MQTT: Receive topic cmnd/sonoff/status, data 1
MQTT: DataCb Topic sonoff, Group 0, Type STATUS, data 1 (1)
MQTT: sonoff/STATUS = PRM: sonoffs, DVES_814C3E, http://otasrv.local:80/api/arduino/sonoff.ino.bin, mqtt_broker.local, 0, 33

status 2
MQTT: Receive topic cmnd/sonoff/status, data 2
MQTT: DataCb Topic sonoff, Group 0, Type STATUS, data 2 (2)
MQTT: sonoff/STATUS = FWR: Version 1.0.15, Boot 4, SDK 1.5.3(aec24ac9)

status 3
MQTT: Receive topic cmnd/sonoff/status, data 3
MQTT: DataCb Topic sonoff, Group 0, Type STATUS, data 3 (3)
MQTT: sonoff/STATUS = LOG: Seriallog 4, Syslog 0, LogHost syslog.local, SSId sonoff_iot, Password sonoff_iot

status 4
MQTT: Receive topic cmnd/sonoff/status, data 4
MQTT: DataCb Topic sonoff, Group 0, Type STATUS, data 4 (4)
MQTT: sonoff/STATUS = MEM: Sketch size 285920, Free 675840 (Heap 42752), Spiffs start 962560, Flash size 1048576 (1048576)

status 5
MQTT: Receive topic cmnd/sonoff/status, data 5
MQTT: DataCb Topic sonoff, Group 0, Type STATUS, data 5 (5)
MQTT: sonoff/STATUS = NET: Hostname esp-814c3e-sonoff, IP 192.168.255.210, Gateway 192.168.255.254, Subnetmask 255.255.255.128
Wifi: Check connection

status 6
MQTT: Receive topic cmnd/sonoff/status, data 6
MQTT: DataCb Topic sonoff, Group 0, Type STATUS, data 6 (6)
MQTT: sonoff/STATUS = MQT: Host mqtt_broker.local, MAX_PACKET_SIZE 1024, KEEPALIVE 120
RTC: sntp 0, Thu Jan 01 00:00:00 1970
```

Leonardo

P.S. I've recompiled version 1.0.15

Leo-C, 2016-07-30:
Oops! I've forgotten status 0.
I resend all, after reloading firmware with MQTT_MAX_PACKET_SIZE set to 256 (as suggested for version 1.0.15)

```
status 0
MQTT: Receive topic cmnd/sonoff/status, data 0
MQTT: DataCb Topic sonoff, Group 0, Type STATUS, data 0 (0)
MQTT: sonoff/STATUS = 1.0.15, sonoff, 0, LIGHT, 0, 1
MQTT: sonoff/STATUS = PRM: sonoffs, DVES_814C3E, http://otasrv.local:80/api/arduino/sonoff.ino.bin, mqtt_broker.local, 0, 67
MQTT: sonoff/STATUS = FWR: Version 1.0.15, Boot 4, SDK 1.5.3(aec24ac9)
MQTT: sonoff/STATUS = LOG: Seriallog 4, Syslog 0, LogHost syslog.local, SSId sonoff_iot, Password sonoff_iot
MQTT: sonoff/STATUS = MEM: Sketch size 285952, Free 675840 (Heap 43520), Spiffs start 962560, Flash size 1048576 (1048576)
MQTT: sonoff/STATUS = NET: Hostname esp-814c3e-sonoff, IP 192.168.255.210, Gateway 192.168.255.254, Subnetmask 255.255.255.128
MQTT: sonoff/STATUS = MQT: Host mqtt_broker.local, MAX_PACKET_SIZE 256, KEEPALIVE 120

status 1
MQTT: Receive topic cmnd/sonoff/status, data 1
MQTT: DataCb Topic sonoff, Group 0, Type STATUS, data 1 (1)
MQTT: sonoff/STATUS = PRM: sonoffs, DVES_814C3E, http://otasrv.local:80/api/arduino/sonoff.ino.bin, mqtt_broker.local, 0, 67

status 2
MQTT: Receive topic cmnd/sonoff/status, data 2
MQTT: DataCb Topic sonoff, Group 0, Type STATUS, data 2 (2)
MQTT: sonoff/STATUS = FWR: Version 1.0.15, Boot 4, SDK 1.5.3(aec24ac9)

status 3
MQTT: Receive topic cmnd/sonoff/status, data 3
MQTT: DataCb Topic sonoff, Group 0, Type STATUS, data 3 (3)
MQTT: sonoff/STATUS = LOG: Seriallog 4, Syslog 0, LogHost syslog.local, SSId sonoff_iot, Password sonoff_iot

status 4
MQTT: Receive topic cmnd/sonoff/status, data 4
MQTT: DataCb Topic sonoff, Group 0, Type STATUS, data 4 (4)
MQTT: sonoff/STATUS = MEM: Sketch size 285952, Free 675840 (Heap 43520), Spiffs start 962560, Flash size 1048576 (1048576)

status 5
MQTT: Receive topic cmnd/sonoff/status, data 5
MQTT: DataCb Topic sonoff, Group 0, Type STATUS, data 5 (5)
MQTT: sonoff/STATUS = NET: Hostname esp-814c3e-sonoff, IP 192.168.255.210, Gateway 192.168.255.254, Subnetmask 255.255.255.128

status 6
MQTT: Receive topic cmnd/sonoff/status, data 6
MQTT: DataCb Topic sonoff, Group 0, Type STATUS, data 6 (6)
MQTT: sonoff/STATUS = MQT: Host mqtt_broker.local, MAX_PACKET_SIZE 256, KEEPALIVE 120

status 7
MQTT: Receive topic cmnd/sonoff/status, data 7
MQTT: DataCb Topic sonoff, Group 0, Type STATUS, data 7 (7)
MQTT: sonoff/STATUS = 1.0.15, sonoff, 0, LIGHT, 0, 1
```

Further detail that can be useful: during SmartConfig, LED stop blinking, then remain on for few seconds and finally it switch off and after some time ESPConfig SmartConfig fails.

Leonardo

arendst, 2016-07-30:
Leonardo,

SmartConfig doesn't fail. From your log a few days back you can see that SmartConfig is started and shortly afterward it sonoff receives both SSID and password (sonoff_iot). That's all that SmartConfig can and will do. Both SSID and password are saved and sonoff reboots using these parameters to receive IP address information from the Wifi Access Point.

Receiving valid IP address information seems to fail and after 16 attempts SmartConfig is started again.

Perhaps have a look in your DHCP network setup.

Leo-C, 2016-07-31:
Hi Theo,
I believe that SmartConfig fails because, after some time spent attempting to configure device, a popup inform me that _ESPtouch fails_.

I can see only first screen, and I can set only _current_ WiFi password (_sonoff_iot_), because SSID is preset (and unchangeable) to SSID of Wifi network connected to smartphone.

I need change both WiFi SSID and password (in only one step, not in two steps like MQTT commands), but app cannot help me because it fails.

However, DHCP seems work well because all names configured in sonoff firmware are correctly resolved
(also DHCP server _dnsmasq_ run on raspberry)

Maybe the problem is due to time setting? (sonoff cannot connect to internet, then it cannot refresh time with NTP service)

Leonardo

Leo-C, 2016-07-31:
OK, I'll test a simple SmartConfig program on sonoff to test better SmartConfig app.

But I cannot understand a behavior of sonoff: after power on, using a WiFi network with default SSID / password set into sonoff firmware, log is:

```
Config: Loaded configuration from flash at FA and count 67
Wifi: Connecting to sonoff_iot as esp-814c3e-sonoff
App: Project sonoff (Topic sonoff, Fallback DVES_814C3E, GroupTopic sonoffs) Version 1.0.15
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
RTC: sntp 0, Thu Jan 01 00:00:00 1970

Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
Wifi: STATION_IDLE
MQTT: Attempting connection
MQTT: Connect failed, rc -2. Retry in 10 seconds
Wifi: Check connection
Wifi: STATION_IDLE
Wifi: Check connection
RTC: sntp 0, Thu Jan 01 00:00:00 1970

RTC: sntp 0, Thu Jan 01 00:00:00 1970

MQTT: Attempting connection
MQTT: Connected
MQTT: sonoff/NAME = Sonoff switch
MQTT: sonoff/VERSION = 1.0.15
MQTT: sonoff/FALLBACKTOPIC = DVES_814C3E
Wifi: Check connection
RTC: sntp 0, Thu Jan 01 00:00:00 1970

RTC: sntp 0, Thu Jan 01 00:00:00 1970

Wifi: Check connection
RTC: sntp 0, Thu Jan 01 00:00:00 1970

RTC: sntp 0, Thu Jan 01 00:00:00 1970

Wifi: Check connection
RTC: sntp 0, Thu Jan 01 00:00:00 1970

RTC: sntp 0, Thu Jan 01 00:00:00 1970

Wifi: Check connection
RTC: sntp 0, Thu Jan 01 00:00:00 1970

RTC: sntp 0, Thu Jan 01 00:00:00 1970

Wifi: Check connection
RTC: sntp 0, Thu Jan 01 00:00:00 1970

RTC: sntp 0, Thu Jan 01 00:00:00 1970

Wifi: Check connection
RTC: sntp 0, Thu Jan 01 00:00:00 1970

RTC: sntp 0, Thu Jan 01 00:00:00 1970

Wifi: Check connection
RTC: sntp 0, Thu Jan 01 00:00:00 1970

RTC: sntp 0, Thu Jan 01 00:00:00 1970

Wifi: Check connection
RTC: sntp 0, Thu Jan 01 00:00:00 1970

RTC: sntp 0, Thu Jan 01 00:00:00 1970

Wifi: Check connection
RTC: sntp 0, Thu Jan 01 00:00:00 1970
```
... and so on.

Why sonoff enter into section that log `Wifi: Check connection` without any action from me (after a while, connection is stable, IP address is offered by DHCP server and ping to sonoff is ever ok) 

Thanks a lot

Leonardo

arendst, 2016-07-31:
Leonardo, 
sonoff enters section ```Wifi: Check connection``` every 20 seconds to check if a wifi connection is still present. IT'S A DEBUG MESSAGE. Changing seriallog back to 0 gets rid of the message. For version 1.0.17 I've changed the message to ```Wifi: Checking connection...``` and if a connection is still present another message is returned: ```Wifi: Connected```.

From your latest log it seems connecting to your MQTT server also is a challenge.
Connecting to a NTP server seems to fail as well because you have no access to the internet and probably did not configure your raspberry pi as a NTP server.

I guess your network is just a bit slow ;-)

I'm glad you finally are able to communicate with sonoff.

Good luck making the best of your environment.

Theo.

closed 2016-09-05 by arendst