arendst/Sonoff-MQTT-OTA-Arduino #18: MQTT Timing

NeuralSpaz, 2016-08-05:
I have been playing around with this firmware for a week or so now.

I've noticed that if I publish 5 times cmnd/sonoff/? to fast it will cause a reboot.
eg:
cmnd/sonoff/power
cmnd/sonoff/light
cmnd/sonoff/ledstate
cmnd/sonoff/grouptopic
cmnd/sonoff/timezone
....etc

75 millisecond delay between pubs is sufficient to not cause reboot

(actual project name and prefix are different from default but you get the idea).

from serial debug:
Wifi: Check connection
[hostByName] Host: 192.168.2.21 is a IP!
:rn 28
:c0 1, 28
DebugMsg MQTT: Receive topic lights/bedsidelamp/topic, data 
MQTT: Receive topic lights/bedsidelamp/topic, data 
[hostByName] Host: 192.168.2.21 is a IP!
DebugMsg MQTT: DataCb Topic bedsidelamp, Group 0, Type TOPIC, data  ()
MQTT: DataCb Topic bedsidelamp, Group 0, Type TOPIC, data  ()
[hostByName] Host: 192.168.2.21 is a IP!
:wr
:sent 41
:ww
DebugMsg MQTT: bedsidelamp/TOPIC = bedsidelamp
MQTT: bedsidelamp/TOPIC = bedsidelamp
[hostByName] Host: 192.168.2.21 is a IP!
:rn 31
:c0 1, 31
DebugMsg MQTT: Receive topic lights/bedsidelamp/ledstate, data 
MQTT: Receive topic lights/bedsidelamp/ledstate, data 
[hostByName] Host: 192.168.2.21 is a IP!
DebugMsg MQTT: DataCb Topic bedsidelamp, Group 0, Type LEDSTATE, data  ()
MQTT: DataCb Topic bedsidelamp, Group 0, Type LEDSTATE, data  ()
[hostByName] Host: 192.168.2.21 is a IP!
:wr
:sent 35
:rn 33
:ww
DebugMsg MQTT: bedsidelamp/LEDSTATE = On
MQTT: bedsidelamp/LEDSTATE = On
[hostByName] Host: 192.168.2.21 is a IP!
:c0 1, 33
DebugMsg MQTT: Receive topic lights/bedsidelamp/grouptopic, data 
MQTT: Receive topic lights/bedsidelamp/grouptopic, data 
[hostByName] Host: 192.168.2.21 is a IP!
DebugMsg MQTT: DataCb Topic bedsidelamp, Group 0, Type GROUPTOPIC, data  ()
MQTT: DataCb Topic bedsidelamp, Group 0, Type GROUPTOPIC, data  ()
[hostByName] Host: 192.168.2.21 is a IP!
:wr
:sent 47
:rn 31
:ww
DebugMsg MQTT: bedsidelamp/GROUPTOPIC = bedsidelamps
MQTT: bedsidelamp/GROUPTOPIC = bedsidelamps
[hostByName] Host: 192.168.2.21 is a IP!
:c0 1, 31
DebugMsg MQTT: Receive topic lights/bedsidelamp/timezone, data 
MQTT: Receive topic lights/bedsidelamp/timezone, data 
[hostByName] Host: 192.168.2.21 is a IP!
DebugMsg MQTT: DataCb Topic bedsidelamp, Group 0, Type TIMEZONE, data  ()
MQTT: DataCb Topic bedsidelamp, Group 0, Type TIMEZONE, data  ()
[hostByName] Host: 192.168.2.21 is a IP!
:wr
:sent 35
:rn 28
:ww
DebugMsg MQTT: bedsidelamp/TIMEZONE = 10
MQTT: bedsidelamp/TIMEZONE = 10
[hostByName] Host: 192.168.2.21 is a IP!
:c0 1, 28
DebugMsg MQTT: Receive topic lights/bedsidelamp/power, data 
MQTT: Receive topic lights/bedsidelamp/power, data 
[hostByName] Host: 192.168.2.21 is a IP!
DebugMsg MQTT: DataCb Topic bedsidelamp, Group 0, Type POWER, data  ()
MQTT: DataCb Topic bedsidelamp, Group 0, Type POWER, data  ()
[hostByName] Host: 192.168.2.21 is a IP!
:wr
:sent 32
:ww
DebugMsg MQTT: bedsidelamp/POWER = On
MQTT: bedsidelamp/POWER = On
[hostByName] Host: 192.168.2.21 is a IP!
Fatal exception 3(LoadStoreErrorCause):
epc1=0x402021e2, epc2=0x00000000, epc3=0x00000000, excvaddr=0x4020be5a, depc=0x00000000

Exception (3):
epc1=0x402021e2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4020be5a depc=0x00000000

ctx: cont 
sp: 3fff1660 end: 3fff2c5c offset: 01a0

>>>stack>>>
....
Soft WDT reset
ctx: cont 
sp: 3fff1290 end: 3fff2c5c offset: 01b0
....
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld

arendst, 2016-08-05:
I looked into it and it seems that lack of buffer space in memory caused by too many requests in a short time results in the exception(3). 

To work around the problem I introduced a semaphore allowing up to three requests being serviced and discarding others until requests can be serviced again.

The fix is available now in Version 1.0.20.

As you also noticed the below bash script with a delay of 70 mSec still runs fine without losing responses:

```
#!/bin/sh
SLEEP=0.07
mosquitto_pub -t cmnd/sonoff/power -n
sleep $SLEEP
mosquitto_pub -t cmnd/sonoff/light -n
sleep $SLEEP
mosquitto_pub -t cmnd/sonoff/ledstate -n
sleep $SLEEP
mosquitto_pub -t cmnd/sonoff/grouptopic -n
sleep $SLEEP
mosquitto_pub -t cmnd/sonoff/timezone -n
sleep $SLEEP
mosquitto_pub -t cmnd/sonoff/power -n
sleep $SLEEP
mosquitto_pub -t cmnd/sonoff/light -n
sleep $SLEEP
mosquitto_pub -t cmnd/sonoff/ledstate -n
sleep $SLEEP
mosquitto_pub -t cmnd/sonoff/grouptopic -n
sleep $SLEEP
mosquitto_pub -t cmnd/sonoff/timezone -n
```

A smaller delay will result in losing responses but no exceptions. 

In addition I think it's not a big problem as all parameter information can also be retrieved using command ```status 0```.

NeuralSpaz, 2016-08-05:
Thankyou. disregarding messages is much preferred than a exception(3) reboot.

Agreed status N  is a better way to get the config data, it only showed up in one of my api test which sub'ed to all the topics and pub'ed to the them all to verify my config.

Tested 1.0.20. no longer reboots no matter how much I abuse it. Closing.

closed 2016-08-05 by NeuralSpaz