arendst/Sonoff-MQTT-OTA-Arduino #19: Relay state after power failure

Orgjvr, 2016-08-24:
Hi,

I am not sure if I am missing something, but after a power failure the sonoff will always switch the relay on, whether it was on or off before the power failure. 

I have played with the following two settings in the config, but it did not change the behavior:
#define APP_POWER              0
and
#define APP_POWER              1

Is this by design?

Thanks,

Org

arendst, 2016-08-25:
I don't see this behaviour; when sonoff is off and I remove power from it sonoff stays off when power is restored. The same is true in the case that sonoff is powered on and power is lost. After power is restored sonoff also powers on.

The define APP_POWER is only used during initial power on and when selecting the default configuration with command ```reset 1```. Once sonoff is up and running the current power state will be used.

For debugging purposes you could connect a serial cable and enter command ```seriallog 3``` to see if sonoff starts up selecting the default configuration all the time which could mean a problem with your selected flash size or a real flash problem.

Orgjvr, 2016-08-25:
Hi Theo,

It turned out that while I were doing initial testing, I send a MQTT message with the retained flag set to TRUE. Recently all my messages had the flag set to FALSE, thus the retained message caused the sonoff to switch ON after every reboot.

Thanks for the GREAT software.

Org

closed 2016-08-25 by Orgjvr