arendst/Sonoff-MQTT-OTA-Arduino #21: upload error

Andrijano, 2016-09-01:
config:30: error: redefinition of 'void CFG_Default()'

arendst, 2016-09-02:
?

arendst, 2016-09-02:
You seem to be using a version older than 1.0.16.
I recommend downloading the latest being 1.0.30 and retry.

petermoffat, 2016-09-27:
Version 1.0.30 of what?

Andrijano, 2016-10-31:
The issue is not solved, config:30: error: redefinition of 'void
CFG_Default()'.I m using 1.6.12 with recent 2.0.6 version.

arendst, 2016-10-31:
Please copy the complete error message (all lines) provided by the IDE. Also copy the first 40 lines of your file ```config```.

To my knowledge there is no file called ```config``` in my repository so I have no idea where this message is coming from.

Andrijano, 2016-11-01:
extern "C" {
#include "spi_flash.h"
}

extern "C" uint32_t _SPIFFS_end;
#define CFG_LOCATION (((uint32_t)&_SPIFFS_end - 0x40200000) /
SPI_FLASH_SEC_SIZE) - 2  // End of SPIFFS area

SYSCFG myCfg;

void CFG_Save()
{
  char log[80];

  if (memcmp(&myCfg, &sysCfg, sizeof(SYSCFG))) {
    noInterrupts();
    if (sysCfg.saveFlag == 0) {  // Handle default and rollover
      spi_flash_erase_sector(CFG_LOCATION + (sysCfg.saveFlag & 1));
      spi_flash_write((CFG_LOCATION + (sysCfg.saveFlag & 1)) *
SPI_FLASH_SEC_SIZE, (uint32 *)&sysCfg, sizeof(SYSCFG));
    }
    sysCfg.saveFlag++;
    spi_flash_erase_sector(CFG_LOCATION + (sysCfg.saveFlag & 1));
    spi_flash_write((CFG_LOCATION + (sysCfg.saveFlag & 1)) *
SPI_FLASH_SEC_SIZE, (uint32 *)&sysCfg, sizeof(SYSCFG));
    interrupts();
    sprintf_P(log, PSTR("Config: Saved configuration to flash at %X and
count %d"), CFG_LOCATION + (sysCfg.saveFlag & 1), sysCfg.saveFlag);
    addLog(LOG_LEVEL_DEBUG, log);
    myCfg = sysCfg;
  }
}

void CFG_Default()
{
  addLog(LOG_LEVEL_INFO, "Config: Use default configuration");
  memset(&sysCfg, 0x00, sizeof(SYSCFG));
  memset(&myCfg, 0x00, sizeof(SYSCFG));
  sysCfg.cfg_holder = CFG_HOLDER;
  sysCfg.saveFlag = 0;
  sysCfg.version = VERSION;
  sysCfg.seriallog_level = SERIAL_LOG_LEVEL;
  sysCfg.syslog_level = SYS_LOG_LEVEL;
  strcpy(sysCfg.syslog_host, SYS_LOG_HOST);
  strcpy(sysCfg.sta_ssid, STA_SSID);
  strcpy(sysCfg.sta_pwd, STA_PASS);
  strcpy(sysCfg.otaUrl, OTA_URL);
  strcpy(sysCfg.mqtt_host, MQTT_HOST);
  strcpy(sysCfg.mqtt_grptopic, MQTT_GRPTOPIC);
  strcpy(sysCfg.mqtt_topic, MQTT_TOPIC);
  strcpy(sysCfg.mqtt_topic2, "0");
  strcpy(sysCfg.mqtt_subtopic, MQTT_SUBTOPIC);
  sysCfg.timezone = APP_TIMEZONE;
  sysCfg.power = APP_POWER;
  sysCfg.ledstate = APP_LEDSTATE;
  CFG_Save();
}

void CFG_Load()
{
  char log[80];

  noInterrupts();
  spi_flash_read((CFG_LOCATION) * SPI_FLASH_SEC_SIZE, (uint32 *)&sysCfg,
sizeof(SYSCFG));
  spi_flash_read((CFG_LOCATION + 1) * SPI_FLASH_SEC_SIZE, (uint32 *)&myCfg,
sizeof(SYSCFG));
  interrupts();
  if (sysCfg.saveFlag < myCfg.saveFlag) sysCfg = myCfg;
  sprintf_P(log, PSTR("Config: Loaded configuration from flash at %X and
[... 171 lines cut ...]
C:\Users\Andrijano\Desktop\kontrol\arduino-1.6.12\portable\sketchbook\sonoff\wifi.ino:
In function 'void WIFI_Connect(char*)':

wifi:77: error: redefinition of 'void WIFI_Connect(char*)'

 void WIFI_Connect(char *Hostname)

      ^

support:389: error: 'void WIFI_Connect(char*)' previously defined here

 void WIFI_Connect(char *Hostname)

      ^

exit status 1
redefinition of 'void CFG_Default()'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Many thanks in advance

Andrijano

arendst, 2016-11-01:
Andrijano, you have a basic version update problem: If you use a new version you'll have to delete ALL files from your sonoff directory and copy the new ones from Github.

In your case there still are files called config.ino and wifi.ino which have been deleted since version 1.0.16

Andrijano, 2016-11-01:
Many thanks, now it seems ok.

Best Regards

Andrijano

closed 2016-11-01 by arendst