arendst/Sonoff-MQTT-OTA-Arduino #23: Wifi wkaku

connectedthings, 2016-09-02:
Theo, great project tx. Have tried it on a a sonoff and a generic homebrew module.

I notice from the pics of your Wifi wkaku hack that the PSU has the 3.3v pin soldered. I also use similar modules and they have pads for a 1117 LDO regulator but usually no output caps, am assuming the caps on your pic are for 3.3v smoothing? Would e great if you could share your PCB design. Am curious about your ESP vertical mount as well, cool approach.

Tx, Sven

PS  Am not very impressed with smartconfig but have played with WPS which works great, any thoughts why you did not go that route?

arendst, 2016-09-02:
I stopped using the itead 5V PSU after one of the mains resistors blew out probably caused by a dirty PCA. I always used a seperate 3V3 regulator.

I now use the HiLink 5V PSU with a seperate 3V3 regulator. On both 5V and 3V3 I re-use capacitors de-soldered from the original KAKU PCA. Currently 6 wkaku units are happily being used.

The ESP vertical mount needs a resistor on GPIO15 not shown in the schematics as it is mounted from the top row down to the PCA.

The wkaku files are now available on Github.

I don't use WPS as you would need physical access to your router starting the WPS process. I did implement WifiManager next to SmartManager allowing to configure Wifi from your phone using the inbuilt AccessPoint and web server.

Rgrds, Theo

connectedthings, 2016-09-03:
Thanks for that Theo. I use the supplies from Electrodragon http://www.electrodragon.com/product/ac-85-265v-dc-5v-power-module/#prettyPhoto and have had good results 10 plus units going strong.

Have also used an external 3.3v supply up to now, am going to try mounting the 1117 and see how things go.

I see your vertical mount does not use a daughter board but ESP mounts directly. Are you using right angle 2mm pitch headers? I am finding these difficult to find.

Tx, Sven

arendst, 2016-09-03:
I solder the wires I need directly to the ESP-12 module

arendst, 2016-09-05:
See Wiki for more information

closed 2016-09-05 by arendst