arendst/Sonoff-MQTT-OTA-Arduino #24: OTA UPLOAD failed

nigelb135, 2016-09-03:
Hi Theo - thanks for sharing your great work.

I have 10 sonoff's running with v1.20 and have upgraded one to v1.30 using the serial transfer. However, the any upgrade OTA gives the following error...

3 September 2016 at 4:22:11 PM AESTa9885435.617ab8
stat/sonoff/UPLOAD : msg.payload : string [74]
Upgrade 1.0.30 from http://192.168.159.183:8081/api/arduino/sonoff.ino.bin
3 September 2016 at 4:22:16 PM AESTa9885435.617ab8
stat/sonoff/UPGRADE : msg.payload : string [20]
Failed To less space

Can you tell what I have done wrong? Here is full status information for the unit

1.0.30, sonoff, 0, LIGHT, 0, 99, 1

PRM: GroupTopic sonoffs, OtaUrl http://192.168.159.183:8081/api/arduino/sonoff.ino.bin, Uptime 1 Hr, Bootcount 3, SaveCount 8

FWR: Version 1.0.30, Boot 31, SDK 1.5.3(aec24ac9)

LOG: Seriallog 2, Weblog 2, Syslog 0, LogHost 192.168.159.195, SSId Barrow Place 600, Password xxxxx, TelePeriod 300

MEM: Sketch size 321kB, Free 104kB (Heap 39kB), Spiffs start 428kB (64kB), Flash size 1024kB (512kB)

NET: Host sonoff-0421, IP 192.168.159.129, Gateway 192.168.159.1, Subnetmask 255.255.255.0, Mac 5C:CF:7F:1C:A1:A5, Webserver 0

MQT: Host 192.168.159.195, Port 1883, Client DVES_%06X (DVES_1CA1A5), User bphauser, Password xxxx, MAX_PACKET_SIZE 256, KEEPALIVE 120

TIM: UTC Sat Sep 03 07:32:52 2016, Local Sat Sep 03 17:32:52 2016, Start DST Sun Oct 02 02:00:00 2016, End DST Sun Apr 03 02:00:00 2016

arendst, 2016-09-03:
Nigel,

from the status information (MEM) I can tell that your compiled image was made for a flash size of 512 and 64k spiffs. That leaves not enough space for OTA or web upload.

The only solution now is recompile for sonoff (Flash size: "1M (64K SPIFFS) and upload using serial interface.

Rgrds, Theo.

nigelb135, 2016-09-03:
Thanks Theo - I suspected as much, but am new to the arduino/esp world.

closed 2016-09-05 by arendst

matsekberg, 2018-01-12:
I flashed ESP8285 / 1M (64k SPIFFS) and get the same error :(

matsekberg, 2018-01-12:
Okay. Thanx a lot for this great software!