arendst/Sonoff-MQTT-OTA-Arduino #25: Does not turn on after flash

Nayar, 2016-09-03:
So I managed to flash it. Says upload complete. But when i put the sonoff in AC current, no LED lights up nor anything happens when i press the button.

Where can i start debugging? Have i bricked my sonoff?

arendst, 2016-09-03:
Well to keep you on your toes it might be bricked ;-)

First take a look at the open and closed issues here. I'll suggest connecting the serial cable and make sure you compile with seriallog set to  3 or Just execute command seriallog 3

Happy hunting

Nayar, 2016-09-04:
Back to life :)

Read few comments here: https://github.com/arendst/Sonoff-MQTT-OTA-Arduino/issues/16

Flashed it again.Just disconnected the 3.3v cable from the USB TTL then put in AC. The LED lit up and got stuffs in the logs. 

I guess it was some issues in the flashing process. Everything had to be flushed out from what i read somewhere else. Anyways. Thanks :+1:

closed 2016-09-04 by Nayar