arendst/Sonoff-MQTT-OTA-Arduino #26: ElectroDragon relay reversed

patbeirne, 2016-09-06:
Theo,

I'm sorry, it appears that I gave you the incorrect information about the ElectroDragon relay config.

The easier relay to connect is the one associated with GPIO13 (relay) and GPIO2 (button). I verified this GPIO13/GPIO2 wiring on Rev 1 and Rev 1.1 boards.  

So could you please change this code 
```
#elif MODULE == ELECTRO_DRAGON              // programming header 5V/3V/gnd/
  #define APP_NAME             "ElectroDragon module"
  #define LED_PIN              16           // GPIO 16 = Led (0 = Off, 1 = On)
  #define LED_INVERTED         0            // 0 = (1 = On, 0 = Off), 1 = (0 = On, 1 = Off)
  #define REL_PIN              12           // GPIO 12 = Red Led and Relay 2 (0 = Off, 1 = On)
  #define KEY_PIN              0            // GPIO 00 = Button 2
  #define REL1_PIN             13           // GPIO 13 = Red Led and Relay 1 (0 = Off, 1 = On)
  #define KEY1_PIN             2            // GPIO 02 = Button 1
```
into this new code
```
#elif MODULE == ELECTRO_DRAGON              // programming header 5V/3V/gnd/
  #define APP_NAME             "ElectroDragon module"
  #define LED_PIN              16           // GPIO 16 = Led (0 = Off, 1 = On)
  #define LED_INVERTED         0            // 0 = (1 = On, 0 = Off), 1 = (0 = On, 1 = Off)
  #define REL_PIN              13           // GPIO 13 = Red Led and Relay 1 (0 = Off, 1 = On)
  #define KEY_PIN              2            // GPIO 02 = Button 1
  #define REL1_PIN             12           // GPIO 12 = Red Led and Relay 2 (0 = Off, 1 = On)
  #define KEY1_PIN             0            // GPIO 00 = Button 2
```
Thanks

NOTE: 

- There is a JPEG of Rev 1.1 on the ED website that seems to have incorrect information; I should have checked with the hardware; my apologies.
- I have not checked this wiring on the newest boards that use the 5A relays.
- The inverted LED works perfectly

arendst, 2016-09-07:
Thanks Pat,

fixed in 1.0.31 where this information has also moved to user_config.h. 

This way anyone can select which relay he/she wants to use.

closed 2016-09-07 by arendst

patbeirne, 2016-09-07:
That looks great. I'm going offline for a month; see you in October.