arendst/Sonoff-MQTT-OTA-Arduino #28: Upgrade from Sonoff-MQTT-OTA?

ecsfang, 2016-09-15:
Hi Theo!

First - thanks for your great work and inspiration! :)

Secondly - I have some Sonoff units flashed with Sonoff-MQTT-OTA, and I wonder if it is possible to OTA upgrade these to Sonoff-MQTT-OTA-Arduino? I did a simple test, but it just seemed to keep the older version.
Or do I have to disconnect them and upgrade using the serial cable?

Best regards,
Thomas

arendst, 2016-09-15:
Hi Thomas,

I was afraid someone would be asking this question someday ;-)

I've looked into it early after I started using the Arduino version and came to the conclusion that it wouldn't work. The non-Arduino version will store it's OTA image at 0x1000 where the Arduino version needs it at 0x0000. I then decided to stop investigating.

So the only way to start using the Arduino version is to reflash using the serial interface. Sorry.

Rgrds, Theo.

ecsfang, 2016-09-15:
Ok, that explains it ... ;)
No problem - just have to bring the screwdriver along ...

Cheers!

closed 2016-09-15 by ecsfang