arendst/Sonoff-MQTT-OTA-Arduino #29: Puzzling corruption

davidmpye, 2016-09-28:
Hi!

I've got a strange corruption issue, which I haven't yet figured.

Using a Sonoff to control a fish tank light via Node-Red/MQTT, which works very nicely.  Tried this with firmware as of a month or so ago, and as of about a week ago.

The failure is that after a while (maybe a few days) things fail.  The light suddenly develop a mind of its own, turning on/off intermittently (more on than off).  It no longer connects to the MQTT server, and on connecting to the serial console, all you see is the watchdog timer being triggered and it repeatedly rebooting - the software itself never seems to initialise properly and it never connects to the MQTT server.

arendst, 2016-09-29:
David,

I see my earlier reply got lost somehow...

I haven't seen your exact same problem although I did notice reboot errors due to unability to reconnect to the wifi router as it seemed to busy or out of reach. This can be viewed using the web interface and the wifi config page where you can scan for wifi access points and it's connectivity. My solution was connection the device to another wifi router. I did not need to reprogram though.

As you can only solve it by reprogramming it might be a power problem corrupting the flash of the sonoff although you already tried a second unit...

In your case you might enable more logging now using command ```seriallog 3```  to see what's going wrong at the time of the error. The reboots might be caused by wifi connection errors which trigger wificonfig which in turn will reboot sonoff every minute.

I have sonoffs running for month now and never needed to reprogram any to get them to life again.

I assume you are using the sonoff with inbuilt power supply.

Is there a stack trace when sonoff reboots?

Rgrds, Theo.

davidmpye, 2016-10-03:
Hi,

Thanks for replying!

I wonder if it is a corruption issue of sorts, but not entirely sure.

Another interesting thing relating to the corruption is that once corrupted, the relay is triggered as soon as the Sonoff is powered on.  No sign of it connecting to my MQTT server.

I will try increasing debug level and see what I find.  Someone has pointed out at my local hackspace, that the sonoff doesn't have the shielding can over the ESP8266 that the ESP12s etc have, and this one is near a pump for my fish tank, so I wonder if the corruption is triggered by EMF in some way.

If I figure it out, or get more info, I'll post back :-)

David

davidmpye, 2016-10-03:
Aha, got something....

Here's a reboot cycle from the current one - which seems to be a bit chattier.

Interestingly, the relay state seems to be random - sometimes on or off between reboot cycles.

My take on it is that it seems to be going into smartconfig mode, but not entirely sure why - the wifi is available, and the password hasn't changed etc, and other devices in same location are on it.

00:00:00 Wifi: Connecting to Fishface as fishtank_sonoff-7805
00:00:00 APP: Project sonoff (Topic fishtank_sonoff, Fallback DVES_1C1E7D, GroupTopic sonoffs) Version 1.0.30
00:00:07 Smartconfig: Started and active for 1 minute
00:01:08 APP: Restarting
 ets Jan  8 2013,rst cause:1, boot mode:(3,0)
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld

00:00:00 Wifi: Connecting to Fishface as fishtank_sonoff-7805
00:00:00 APP: Project sonoff (Topic fishtank_sonoff, Fallback DVES_1C1E7D, GroupTopic sonoffs) Version 1.0.30
00:00:07 Wifimanager: Started and active for 1 minute for initial request
00:00:08 HTTP: Webserver active on fishtank_sonoff-7805 with IP address 192.168.4.1
00:01:08 APP: Restarting

davidmpye, 2016-10-03:
(Sorry, as an additional:  

My wifi network is Fishface, my IP range is 192.168.1.X

Curiously the smartconfig has picked 192.168.4.1 - presumably it's set up its' on AP?

David

arendst, 2016-10-03:
David,

the reboot sequence is a typical case of wifi connection problem. I don't know what's the cause though..

The ip address 192.168.4.1 is indeed the default AP address used by smartconfig. If smartconfig is active you could try to connect and see your available wifi networks and their signal strength. This might give an hint.

davidmpye, 2016-10-03:
Also interesting that one time round its' reboot cycle, it picks Smartconfig, the next time round it goes for WifiManager.

The code implies that it is using config variables to decide?  It's certainly free of button presses.

Relay also goes on and off with each reboot cycle too...

arendst, 2016-10-03:
For version 1.0.30 this is 'as designed'. The latest versions don't do it that way as they allow to configure a fixed selection with command wificonfig.

Does the relay really switch or does it just quickly goes off/on. If so then another 'as designed' as the sonoff puts the relay controll pin always in off state after a rebooot. This has been solved with the sonoff dual from itead.

davidmpye, 2016-10-03:
The relay stays on for maybe ~1minute, then a brief off while rebooting, then goes back to on...

arendst, 2016-10-03:
So as designed when using an esp8266 for switching relays. If the relay is off you won't hear a switch with every reboot.

It all comes down to a bad wifi connection...

davidmpye, 2016-10-03:
Hmm... I'll keep playing then.  

I still think there's something else going on, as the sonoffs seem to work for a while then fail.  This one seems to be a bit chattier via serial - the others wouldn't talk at all.

Also this one appears to now work (intermittently at least), whereas the other one wouldn't do anything without reprogramming.

Will keep playing - thanks for your help!

davidmpye, 2016-10-03:
This info from here  https://github.com/esp8266/Arduino/issues/2186

may be relevant...

<snip>
PavelDorofeev commented on 1 Sep
I have the same problem on ESP-12 Lolin (Arduino 1.6.8). Reset ESP-12 , WiFI.begin() and WiFi.status() != WL_CONNECTED waits forever. Only repower the WIFI router help me.

This solves the problem fine.

WiFi.persistent(false);
WiFi.mode(WIFI_OFF); 
WiFi.mode(WIFI_STA);
WiFi.begin(ssid, password);
<snip>

Doing some further tests.   If it is this, it's some kind of strange workaround of a WiFi library issue, by the sound of it!

David

arendst, 2016-10-04:
Looks promissing. If you compile yourself you could implement it in the support.ino file. 

I'll have a look when I'm back from holiday...

davidmpye, 2016-10-05:
^^ didn't fix it for me.  Still hunting!

Happy holiday!

arendst, 2016-10-06:
David,

fixed issue 2186 in sonoff version 2.0.2. I also made some other wifi changes which might solve your reconnection problem.

davidmpye, 2016-10-29:
Hi!

So far (!) this seems to be holding up well!  Thanks!
Cheers

David

closed 2016-10-29 by davidmpye