arendst/Sonoff-MQTT-OTA-Arduino #32: Assistance with platformio compile

nigelb135, 2016-10-04:
I fully understand that this is possibly an unreasonable request, but here goes...

I am trying to compile with platformio (on atom) against the espressif generic ESP8266-01 1M board and receive this error:

/Users/nigelball/Downloads/Sonoff-MQTT-OTA-Arduino-master/sonoff/src/sonoff.ino:204:25: error: variable or field 'WIFI_wps_status_cb' declared 
void
strlcpy(sysCfg.hostname, WIFI_HOSTNAME, sizeof(sysCfg.hostname));
^
/Users/nigelball/Downloads/Sonoff-MQTT-OTA-Arduino-master/sonoff/src/sonoff.ino:204:25: error: 'wps_cb_status' was not declared in this scope

Any help would be appreciated - I'm a newbie at Arduino and a total newb at platformio!.

arendst, 2016-10-04:
That's why I still use Arduino IDE.

PuceBaboon, 2016-10-04:
Nigel,

   Create an include file in the src directory named "support.h" containing
this:-

-----------------------------
/*
 * $Id: support.h,v 1.1 2016/09/19 10:08:17 anoncvs Exp $
 */
#ifdef __cplusplus
extern "C" {
#endif

#ifndef __SUPPORT_H__
#define __SUPPORT_H__

#include "user_interface.h"

/* Function prototypes. */
void WIFI_wps_status_cb(wps_cb_status status);

#endif // ifndef __SUPPORT_H__
#ifdef __cplusplus
}
#endif
-----------------------------

   Then add an include line somewhere near the top of src/sonoff.ino to
reference it.  You should then get a clean compile.

                 -John-

nigelb135, 2016-10-04:
That worked! Thank you John.

closed 2016-10-04 by nigelb135