arendst/Sonoff-MQTT-OTA-Arduino #33: OTA upload error

ecsfang, 2016-10-05:
Hi Theo!

When doing an upgrade (payload = 1), I keep getting the following response:

stat/sonoff6/UPGRADE : msg.payload : string [68]
Upgrade 1.0.33 from http://api.[_...my server..._].se/arduino/sonoff.ino.bin

stat/sonoff6/UPGRADE : msg.payload : string [29]
**Failed Server not Report Size**

If I instead upload using the web-server, then the upgrade goes ok.
I have tried to search for the origin of this error message, but I have failed so far :(
Any ideas?

Cheers,
Thomas

arendst, 2016-10-05:
Thomas,

the webserver upload uses a different mechanism then the ota upload so I'm glad you were able to upload anyhow.

The error message is new to me but for some reason the ota download from your webserver does not receive the correct file size. What version of esp8266-arduino are you using? I currently use 2.3.0

ecsfang, 2016-10-05:
Hi, 
Yes, I'm able to upgrade, but it would be more convenient using the command. 
I'm also using 2.3.0 for the esp-arduino.
Maybe i could add some logging and debug the upgrade. 
Cheers!

closed 2016-11-16 by arendst