arendst/Sonoff-MQTT-OTA-Arduino #34: Correct version of platform.txt?

ecsfang, 2016-10-06:
Hi Theo,

Looking through the files, I saw that the '2.3.0' version of platform.txt in your git have the wrong version number:

version=2.2.0

Is this intentional or a copy&paste error?

BR, Thomas

arendst, 2016-10-06:
The file is a copy from the original 2.3.0 set where it also states 2.2.0. I just made some amendments to it so that's why it still says version 2.2.0. I wanted to keep as close as possible to the original file ;-)

ecsfang, 2016-10-06:
Ok, I just saw that on my machine (not my Sonoff dev), it had 2.3.0 installed, and there the version number was correct ... maybe it was updated somewhere without stepping the actual version number ...
Anyhow, you seem to have control of it ... ;)
Cheers!

closed 2016-10-06 by ecsfang