arendst/Sonoff-MQTT-OTA-Arduino #38: Itead Sonoff Pow

ThomDietrich, 2016-10-09:
I've just preordered a few [Sonoff Pow](https://www.itead.cc/wiki/Sonoff_Pow). Schematics are [available here](https://www.itead.cc/wiki/images/5/52/Sonoff_POW_Schematic.pdf). Looks good to be compatible. Are there any plans to integrate the new telemetry power consumption? I would be able to provide first data (as soon as they arrive) if needed.

arendst, 2016-10-09:
I'll have to investigate but would like to support the Sonoff Pow too.

arendst, 2016-10-09:
Just ordered one pow...

ThomDietrich, 2016-10-09:
That's the spirit :D

eazuka, 2016-10-11:
just ordered 2 and more planned. i'm will be following this issue to know progress as i want this support of Sonoff Pow too. i apprciate your efforts @arendst

mdilger, 2016-10-12:
I have also just ordered a couple of units and look forward to support of this module.  Cheers, Mike.

arendst, 2016-10-18:
Experimental support has been made available in version 2.0.5. I still need to test it with a real Pow as I haven't received mine yet.

ThomDietrich, 2016-10-18:
Already a big step! Thanks man ;)
I'll be testing this as soon as I receive mine.

eazuka, 2016-10-18:
good to know @arendst . what measurement details can be gotton from the Sonoff Pow support? apart from real power (watt) and kilowatt-hour (kWh) ? can things like voltage, frequency, reactive power be gotton as well?

arendst, 2016-10-18:
With my implementation it has Voltage (V), Current (A), Current power (W), Period power (Wh) and daily power (kWh). I do not think frequency detection is possible. CosPhi may be possible though.

eazuka, 2016-10-18:
i owe you a beer man. good work

arendst, 2016-10-19:
My Pow has been shipped today.

ThomDietrich, 2016-10-19:
I didn't receive mine yet. Happy Testing ;)

mdilger, 2016-10-19:
My units just shipped yesterday also.

ThomDietrich, 2016-10-23:
Hey @arendst, mine still didn't arrive :-/ I already have one question. I looked through the source code and I didn't find changes in regard to when consumption data is send. Is it only send periodically?
I want to propose to additionally send data whenever power consumption changes by more than x Watts or when crossing certain ranges.
The described behavior is implemented by other power consumption measurement switches (e.g. [Homematic HM-ES-PMSw1-Pl](http://www.elv.de/homematic-funk-schaltaktor-1fach-mit-leistungsmessung-zwischenstecker.html)) and quote useful when observing for example a washing machine and it's phases.
Thanks for your efforts! PLease let me/us know how we can repay you ;)

arendst, 2016-10-24:
Thresholds for Power, Voltage (mine is unstable) and Current have been implemented in version 2.0.6 just released. See wiki for command explanation and MQTT results.
Tested with external oscillator (spare Wemos D1 mini with my oscillation code emulating HLW8012).

Still waiting for my Pow to arrive...

eazuka, 2016-10-26:
@arendst has your Pow arrived yet?  :)

arendst, 2016-10-26:
Nope. Expecting it this week.

Do not forget to take a look at my wiki regarding firmware loading the Sonoff Pow. DANGEROUS

ThomDietrich, 2016-10-26:
This happened to me too once :sweat: If no potential separation hardware is available, your easiest option is to use a Laptop on battery only. Apart from that, an end user should never connect AC while having the casing opened and USB connected to begin with...

eazuka, 2016-10-26:
@arendst The Pow supports setting countdown, single and repeat timing tasks schedules to turn on/off at a specified time. and while discussing with one of the engineers at itead via email as regards some questions on Pow, he made me understand that the tasks are supported at the device/firmware level not just at the EweLink app meaning if there is a disconnection between the device and the EweLink app or itead server, the specified tasks/command will still run at the when the set time or thresholds are reached and Maximum of 8 timing tasks is allowed. 

My question is, does this firmware support such or is it possible to implement such features of turnning on/off the Pow device based on thresholds and or time?

arendst, 2016-10-26:
No this firmware does not provide time related actions. That's the task of your Home Automation manager. If you have ten sonoffs you do not want to access them repeatedly to change time or threshold values locally. You want a Home Automation tool to do that (well at least that's what I want).

I currently use Domoticz but know others use OpenHAB, Node-red or FHEM. In those tools you configure Scenes that control devices like sonoff and others at certain moments in time (Daybreak, Morning, Dusk, TimeToBed etc.). All these tools run locally on your NAS or dedicated server as it should.

My firmware can provide thresholds to trigger your Home Automation tool.

eazuka, 2016-10-26:
well are quite right but same said for itead, the EweLink app and their cloud server could be considered the home automation tool but even so, they still made the command storable in the device directly so that in a situation where the home automation tools are down for whatever reason, the task are still run.

In my case, i will be using 100s of Pow for various purposes and mostly deployed remotely so an automation tool is certainly going to be used to monitor, give commands and update but in a situation where there is a disconnection with the automation tool which most certainly will happens, i want to be the set task are still run as at when due before. This is mostly needed because of the romote nature of some of my use case.

Is it possible i discuss with you in private? pls say hi to my email eugeneazuka@gmail.com when i reply.

ThomDietrich, 2016-10-26:
Hello @eazuka, I have to fully agree with arendst. You want your home automation solution to be the deciding entity and the many different sensing and acting modules to be nodes controlled by it. I actually find great peace in the fact, that all (sadly almost all) smart home devices in my openHAB setup act exactly as dictated by the centralized openHAB logic.

You said you are thinking about deploying " 100s of Pow and other Sonoffs for various purposes". Planing to configure all of them individually to execute tasks at certain times is one more level of chaos you really do not want...!
You should think about your actual system structure. If you fear network reliability problems a probably better solution is to deploy one automation server per site, controlling dozens of modules at once. Yet another server could even act as a master system above all of these. (Node-RED or openHAB are excellent choices for that)

eazuka, 2016-10-26:
@ThomDietrich i think you didn't get me correctly, to be brief, my setup will have a local automation controller per site with a master cloud server that controls these other site located sub controllers.
and i never said i will manually configure them individually rather configure them through the automation tool.

nunofgs, 2016-11-05:
I received a Sonoff POW today but I can't seem to flash it. I believe the button may not be connected to GPI0, so it never enters programming mode.

Has anyone flashed it successfully?

Toshibass, 2016-11-05:
I can confirm that pressing the button prior to powering the sonoff pow does enable programming mode and it flashes no problem ... I'm using some software I have for other sonoff,s I can flash using the arduino ide and control the relay via Mqtt etc but my problem .... I lack the experience to understand how to implement / try this experimental software to test obtaining kw, power, voltage etc, I just need a idiots guide to explain we're / how to load this set of experimental files like we're do I put them, do I need all of them etc, please advise.

arendst, 2016-11-05:
I'm glad the button still works as expected for Toshibass. Mine is still in the mail so I can't tell.

See the wiki (link in readme and at the top of this screen) for implementing my code.

Toshibass, 2016-11-05:
Thanks for your response, I had seen and read the wiki but didn't see the righthand pane showing how to implement the code, 'doh' ...  i'll give it a go thanks.

ThomDietrich, 2016-11-05:
I received my two Sonoff Pows yesterday, didn't find the time to flash them yet though. Testing on Tuesday probably.

MiKa1975, 2016-11-05:
Sonos POW flashed yesterday without problems

Toshibass, 2016-11-05:
hi MiKa1975 are you saying you flashed your sonoff pow with this "experimental software" and its fully working like you can toggle the relay and get current power, voltage etc 

Currently I flashed the software it connects to mqtt in node-red no problem but 
1.) pressing the button on the unit toggles the led but not the relay, 
2,) I can do ... stat/sonoff/STATUS and I get 2.0.7, 1, sonoff, 0, POWER, 0, 1, 1, 1, 1, 0  
3,) I can do ... stat/sonoff/POWER and I get OFF or ON if I have toggled "led" (not the relay) with the button.
4,) but that's about it no action cmd's like cmnd/sonoff/Power toggle or cmnd/sonoff/Status 8 work I just get ...

stat/sonoff/SYNTAX : msg.payload : string [153]
Status, SaveData, SaveSate, Upgrade, Otaurl, Restart, Reset, WifiConfig, Seriallog, Weblog, Syslog, LogHost, LogPort, SSId, Password, Hostname, Webserver
05/11/2016, 14:14:31f7d1a981.907048

stat/sonoff/SYNTAX : msg.payload : string [172]
MqttHost, MqttPort, MqttUser, MqttPassword, MqttClient, Topic, ButtonTopic, ButtonRetain, MqttUnits, MessageFormat, GroupTopic, Timezone, Light, Power, Ledstate, TelePeriod
05/11/2016, 14:14:31f7d1a981.907048

stat/sonoff/SYNTAX : msg.payload : string [83]
DomoticzInTopic, DomoticzOutTopic, DomoticzIdx, DomoticzKeyIdx, DomoticzUpdateTimer 

every time! any ideas what I am doing wrong?.

one other point .. in sonoff.ino there are lines ...
//#define SONOFF                        1            // Sonoff, Sonoff SV, Sonoff D etcl
#define SONOFF_POW                 9            // Sonoff Pow
//#define ELECTRO_DRAGON     10           // Electro Dragon

Is it correct to put the // as above ?  I tried it both ways doesn't appear to make any difference

arendst, 2016-11-05:
There is NO NEED to change anything in sonoff.ino All you have to change is select the correct hardware in user_config.h

So in your case change the first define (MODULE) from SONOFF in SONOFF_POW and recompile and upload.

patrickruijters, 2016-11-05:
Hi, 
Short introduction from myself: Patrick , Living in the southern parts of the Netherlands, I like different kinds of home automation/electronics, using Domoticz. 
I received a couple of Sonoff POWs today. I'm trying to upload with Arduino IDE (ver.1.6.5), but i get the 'fatal error' : "sonoff.ino:121:57: fatal error: ESP8266HTTPClient.h: No such file or directory". I'm a littlebit familiar with the IDE-software and very little with programming.
Could you please push me in the right direction? 

ps.
It would be great if you contributed on the domoticz-forum with your Sonoff firmware... ;-).

arendst, 2016-11-05:
Patrick. Have a look at the wiki at the top of this page. It contains steps regarding IDE levels (I use 1.6.12), esparduino level (2.3.0) which solves your missing HTTPclient.

For all others: LOOK AT THE WIKI FIRST please.

Toshibass, 2016-11-05:
Thanks for your post arendst that got me a little further, I also found unlike other sonoff's the relay will only energise when the sonof pow is powered by mains 230v so current situation is if I switch on the relay with the button I get the red and blue led and every so often I receive via Mqtt values for power current etc so that side of things appears to work, however still I cannot cmnd the relay via Mqtt or request any Sonoff Pow additions, any adeas what I could check next ?
I still get the same response as in 4,) in my previous post.

arendst, 2016-11-05:
Toshibass, can you show me your mqqt command line you use to send something to your sonoff? What should work is a topic like ```cmnd/sonoff/Power``` with a message like ```toggle``` or topic  ```cmnd/sonoff/Status``` with a message like ```8```.

Toshibass, 2016-11-06:
Oh now I get it , I am so dumb, but in all fairness it's not clear on the wiki that you need to split the cmnd to topic and message,  but I guess In reality I should have realised that anyway, ok so problem appears to be solved everything I have tried tonight works (didn't have time to check everything tonight but my guess is it will be fine) thank you so much for your help looks like a great piece of software.

roadfox, 2016-11-06:
Had the same issue on my first tries, maybe it would be better unterstandeble for beginners to add a second row in the wiki with Title "Message".
Theo you yould also consider to make the wiki public editable, i'm sure there are people here who would like to contribute a little to your great project by adding some wiki content: https://help.github.com/articles/changing-access-permissions-for-wikis/

Toshibass, 2016-11-06:
OK after some more testing this morning all the cmnd's and stuff work fine BUT as soon as I connect a load to the sonoff_pow I am getting resets, the reset frequency depends on the type of load, last night I was trying it with my soldering iron the relay blipped occasionally perhaps every 2or3 minutes I though this was perhaps a bad connection or something but this morning I tried a led lamp, thats blipping like once a second, I think this is a software issue because if I flash my basic sonoff.ino which just controls the relay with no fancy stuff or power readings whatever load I connect the sonoff_pow is rock solid so for sure its nothing to do with power surges etc.

Incidentally using your software I am not using the web server, ota or domoticz so I don't think I can check the logs as per issue #51 and for sure I cannot attach FTDI solution considering the warnings NOT CONNECT AC POWER WHILE SERIAL CONNECTION IS BEING USED ON A SONOFF_POW so I think it would be best to wait for your sonoff_pow to turn up so you can test it yourself.

jumpjack, 2016-11-07:
Any newbie tutorial available?
I don't know anything about MQTT, but I know arduino programming and web programming.
Should I just upload POWER.INO downloaded from [here ](https://github.com/arendst/Experimental/tree/master/sonoff-InitialStatus/sonoff) Then what?

Any idea if data logger by the Pow can be shared/exported using standard firmware and app?
Are data stored on flash, or do they disappear at shutdown?

Toshibass, 2016-11-08:
Hi jumpjack it's been said before, follow the tutorial on the wiki, see panel on right hand side of the page it explains everything, just don't trip up like me when it come to Mqtt commands some you need to split the command so for say, toggle the relay if it's a sonoff_pow then it's.  Topic  cmnd/sonoff_pow/power   Payload  toggle

jumpjack, 2016-11-08:
Can you provide a link? Github is a mess.

jumpjack, 2016-11-08:
BTW, this "tutorial" is useless: https://github.com/arendst/Sonoff-MQTT-OTA-Arduino/wiki/Prerequisite
To many things are supposed to be already known, It's not suitable for beginners.

arendst, 2016-11-08:
Walk before you leap. Google MQTT, Mosquitto, node-red, Investigative, Patience, Study.

jumpjack, 2016-11-08:
Or look for a beginner  tutorial .
(probably I'll write it myself)

jumpjack, 2016-11-08:
I can't go further, that "Prerequisite" page is a real mess!
https://jumpjack.wordpress.com/2016/11/08/arduino-esp8266-sonoff-e-mqtt/

Is anybody able to help me finishing the step-by-step tutorial?

patbeirne, 2016-11-09:
I have written a wiki page called "Sonoff & MQTT for Newbies". You can find it in the wiki "Pages" list, but it's not yet in the wiki table-of-contents, pending Theo's approval.

I'm interested in critical comments. Should it be split into two pages? It this page even required...? since most of this info can be found elsewhere on the net.

arendst, 2016-11-09:
Pat, thanks for your work. I made it available via the menu on the right. Feel free to add/update. Looks fine with me although as you suggested perhaps it needs to be split between the sonoff part (top) and the more general part (bottom).

jumpjack, 2016-11-09:
Has "solved" label been added to the issue because the FW now reads right values?

arendst, 2016-11-09:
Correct. It also means that the Pow now is supported.

You hijacked this issue for Newbie tutorial. I suggest to open a new issue if you're still having trouble reading the wiki or have a look at the today released Newbie info in it..

ThomDietrich, 2016-11-09:
@arendst I did not yet come around to do testing measurements. Seems like this is not needed anymore, does wahat you said just now mean that you validated the accuracy of the readings and the function of the threshold function?

arendst, 2016-11-09:
The threshold is still challenging. I have to look into this

closed 2016-11-09 by ThomDietrich

TotallyInformation, 2016-11-24:
@patbeirne re the newbie article. Great start. I'd suggest putting some pointers to Node-RED in there as it is a really helpful starting point for people new to MQTT and home automation. http://nodered.org/docs/ is a good place to start along with the Google Group: https://groups.google.com/forum/#!forum/node-red

wpeterw, 2017-02-08:
I'm using your firmware on all my Sonoffs (6, 4 pow). I can't get the powerconsumption in Home-assistant. 

Does the Sonoff publish the power consumption periodically ? Whay topic do I subscribe to in Homeassistant to get this value ?

arendst, 2017-02-08:
This issue is closed and this version is EOL. I suggest you start using the latest version available from https://github.com/arendst/Sonoff-Tasmota

Periodic data is send enabling teleperiod with a value of 300 to receive updates every 5 minutes.

wpeterw, 2017-02-08:
I'm using the version you linked to. 

As the Sonoff sends the data every 5 minutes it should be enough to add the correct state topic like in the example above ?

smadds, 2017-02-09:
Do you have anything to monitor everything that's going in your mqtt
broker? That could give a clue.

If you are using Node-Red then the MQTT UI Console
<https://flows.nodered.org/flow/1cc12bc47df5b6b9fa839e47c99dd03c> is a
useful flow to have. It's certainly helped me to "see the unseen" and debug
some issues.
ᐧ

------------------------------
Simon Maddox | CEO | BCMX Ltd

*t:* UK: 01737-887657     USA: 1-631-998-5086
*mob:* +44-7785-504340
*e:* simon.maddox@bcmx.com     *w:* http://bcmx.com