arendst/Sonoff-MQTT-OTA-Arduino #39: Problems with Temp/Humidity

mihalski, 2016-10-11:
Hi,

I got my new Sonoff, configured the user_config file including uncommenting #define SEND_TELEMETRY_DHT, and programmed the Sonoff.

I then plugged in the temp/humidity sensor (all the way until you hear the click and it's securely plugged in) and waited for temp/humidity telemetry data.. Nada. All I get is this:

    2:17:53 MQTT: sonoff/UPTIME = 0
    12:17:53 MQTT: sonoff/RSSI = 100
    12:17:53 MQTT: sonoff/LIGHT = On
    12:17:53 MQTT: sonoff/TIME = 2016-10-11T12:17:53

I notice the time is wrong also (I assumed 10 would be correct as I'm in Brisbane/Australia).

Any ideas as to what might be wrong?

Regards,
Michal

arendst, 2016-10-11:
Many could be wrong...

1) If you have a sonoff TH10/16 then forget about using GPIO04 as in my case this is not available on the Sonoff TH10/16 connector; they just did not install the parts needed as I noticed and measured on mine. Only GPIO14 is available on the tip of the connector, the second ring from the tip should have been GPIO04 but is not connected to the ESP8266. The third ring is ground and the last ring, closest to the cable is +3V3.

So in order to use your DHT just swap the defines for GPIO04 and GPIO14 in user_config.h and recompile and upload. That should solve it (unless itead decided to use the second ring of the connector for the DHT sensor in which case you'll have to contact itead for a solution).

BTW I would not be surprised; my received ds18b20 didn't work either as they swapped both power connections of the sensor...

Also remember that itead insists to plug in the sensor first before power on the sonoff TH10/16 as you may most probably short Gnd and +3V3 otherwise.

2) If I change the timezone to 10 it adds ten hours to UTC as expected. What is your result compared to UTC if you execute command ```timezone 10``` and then perform command ```status``` using the serial interface or the console in the webinterface?

Another option is to update both TIME_DST and TIME_STD in user_config.h for your location and execute command ```timezone 99``` in which case it also supports daylight savings.

arendst, 2016-10-11:
Also do not forget to change define DHT_TYPE into DHT22

mihalski, 2016-10-12:
Thank you for the quick reply.

The timezone issue seems to have resolved itself but still no luck with temp/humidity readings.. There is SOME progress though. At least it acknowledges the existence of TEMP.

```
00:00:00 APP: Project sonoff (Topic sonoff1, Fallback DVES_07C2BA, GroupTopic sonoffs) Version 2.0.4
00:00:04 HTTP: Webserver active on sonoff1-0698 with IP address 192.168.1.3
00:00:04 MQTT: Attempting connection...
00:00:04 MQTT: Connected
00:00:04 MQTT: sonoff1/INFO = Sonoff module version 2.0.4, Fallback DVES_07C2BA, GroupTopic sonoffs
00:00:04 MQTT: sonoff1/INFO = Webserver active for Admin on sonoff1-0698 with IP address 192.168.1.3
10:00:40 MQTT: sonoff1/UPTIME = 0
10:00:40 MQTT: sonoff1/RSSI = 88
10:00:40 MQTT: sonoff1/TEMP = 0.0
10:00:40 MQTT: sonoff1/POWER = Off
10:00:40 MQTT: sonoff1/TIME = 2016-10-12T10:00:40
10:02:10 MQTT: sonoff1/STATUS = 2.0.4, 1, sonoff1, 0, POWER, 0, 10, 1, 1, 1, 0
```

I'll keep fiddling with settings while I await any other suggestions.. I'd even settle for a precompiled firmware that you KNOW works (or even a user config) in case I am making some sort of stupid mistake.

Regards,
Michal

arendst, 2016-10-12:
First, after investigating the Sonoff TH10/16 website it seems the DHT sensor is a DHT11 (= AM2301) so forget my suggestion regarding DHT22. The AM2301 should work.

Second, from your first two screen shots I suspect why you only see  sonoff1/TEMP without sonoff1/HUM that you must have selected SEND_TELEMETRY_DS18B20 instead of SEND_TELEMETRY_DHT. The code for SEND_TELEMETRY_DHT will ALWAYS output nothing or both TEMP and HUM and not TEMP only. This confirms your first EDIT.

You might execute command ```seriallog 3``` to see some logging regarding the DHT sensor but I'm afraid it will only show the message "DHT: Timeout waiting for start signal low pulse" in which case it cannot find the sensor.

In my setup both DHT11 and DHT22 perform as expected. You might have a sensor wiring issue as I had with the DS18B20 (swapped power lines) or the signal cable is connected to the wrong ring on the sensor plug.

I only have both DHT11 and DH22 as breadboard items.

When I measure the resistance between Gnd and Uv I measure 300 ohm with meter Gnd connected to sensor Gnd and high resistance between Gnd and Uv with meter Gnd connected to sensor Uv.
When I measure the resistance between Gnd and signal I measure 400 ohm with meter Gnd connected to sensor Gnd and high resistance between Gnd and signal with meter Gnd connected to signal.

This might give some info about possible cable miswire.

arendst, 2016-10-12:
MY MISTAKE. The AM2301 is equal to DHT22. So you could have a DHT11 or an AM2301.

The below works for me:

```
// MQTT - Telemetry
#define TELE_PERIOD            300          // Telemetry (0 = disable, 2 - 3600 seconds)
#define SEND_TELEMETRY_UPTIME               // Enable sending uptime telemetry (if disabled will still send hourly message)
#define SEND_TELEMETRY_RSSI                 // Enable sending wifi RSSI telemetry
//#define SEND_TELEMETRY_DS18B20              // Enable sending DS18B20 temperature telemetry
#define SEND_TELEMETRY_DHT                  // Enable sending DHT11, DHT21, DHT22, AM2301, AM2302 or AM2321 temperature and humidity telemetry
#define SEND_TELEMETRY_POWER                // Enable sending power telemetry

#if MODULE == SONOFF                        // programming header 1:3.3V 2:rx 3:tx 4:gnd
  #define APP_NAME             "Sonoff module"
  #define LED_PIN              13           // GPIO 13 = Green/Blue Led (0 = On, 1 = Off) - Sonoff
  #define LED_INVERTED         1            // 0 = (1 = On, 0 = Off), 1 = (0 = On, 1 = Off)
  #define REL_PIN              12           // GPIO 12 = Red Led and Relay (0 = Off, 1 = On)
  #define KEY_PIN              0            // GPIO 00 = Button
  #define DSB_PIN              14           // GPIO 14 = TEM1 - DS18B20 (Sonoff_TH10A(16A), Sonoff SV)
  #define DHT_PIN              14           // GPIO 14 = TEM1 - DHT22 (Sonoff_TH10A(16A), Sonoff SV)
  #define DHT_TYPE             AM2301       // DHT module type (DHT11, DHT21, DHT22, AM2301, AM2302 or AM2321)

```

mihalski, 2016-10-14:
I don't think I quite grasp the measurements you made. I was able to test contoinuity from the sensor to the socket (but not the ESP as the contacts on that are far too small).

I've attached an image. Not sure if it's of any help at all but I thought I'd try.

http://prnt.sc/ctqub2

I actually purchased 2 basic Sonoff's and 2 Sonoffs with temp/humidity sensor just in case. Going to see if I can use the original firmware to figure anything out. I wish it was easier to trace from the socket to the ESP :/

I also have some queries with regards to the MQTT commands, but I'll start a new thread for that.

Regards,
Michal

arendst, 2016-10-14:
Ah, nice picture. In my case the red and black connections were swapped in the sensor assy. That's what I tried to explain.

Why does the sensor update now? Does it also happen with the sonoff software?

mihalski, 2016-10-14:
Only the unmodified sensor updates the temp/humidity. The one with the Sonoff-MQTT-OTA-Arduino firmware does not.

Physically they are both identical so I don't know what's wrong. Is there a way to dump the Sonoff firmware from the unmodified device?

According to the iOS app it's running firmware 2.0.1 which is the latest version.

Here's an image from the EWeLink app.

http://prnt.sc/ctxl5a

Regards,
Michal

arendst, 2016-10-16:
Michal, I did a rewrite of the DHT sensor code. Could you be so kind to try it out and let me know if this works? You can download it from my Experimental repository as sonoff-dht.

Thnx in advance, Theo.

mihalski, 2016-10-16:
Thank you for making the effort to try and get this working.

Unfortunately no luck:

00:00:00 APP: Project sonoff (Topic sonoff1, Fallback DVES_07C2BA, GroupTopic sonoffs) Version 2.0.4b
00:00:04 HTTP: Webserver active on sonoff1-0698 with IP address 192.168.1.3
00:00:04 MQTT: Attempting connection...
00:00:04 MQTT: Connected
00:00:04 MQTT: sonoff1/INFO = Sonoff module version 2.0.4b, Fallback DVES_07C2BA, GroupTopic sonoffs
00:00:04 MQTT: sonoff1/INFO = Webserver active for Admin on sonoff1-0698 with IP address 192.168.1.3
19:20:03 MQTT: sonoff1/1/POWER = On
19:20:05 MQTT: sonoff1/1/POWER = Off
19:24:38 MQTT: sonoff1/UPTIME = 0
19:24:38 MQTT: sonoff1/POWER = Off
19:24:38 MQTT: sonoff1/TIME = 2016-10-16T19:24:38
19:29:38 MQTT: sonoff1/UPTIME = 0
19:29:38 MQTT: sonoff1/POWER = Off
19:29:38 MQTT: sonoff1/TIME = 2016-10-16T19:29:38

Is there some sort of debugging/probing that could be done? ie send some sort of signal TO the sensor so that I can at least identify if the correct GPIO is being used? Might see if I can capture what's happening on the signal line of the temperature sensor on both the modified firmware and original firmware modules.

arendst, 2016-10-16:
You could enable some more info with ```seriallog 3```. It needs to use GPIO14 anyway as that is the only one available on the sonoff TH

mihalski, 2016-10-16:
Giving it a go now.

```
00:00:00 APP: Project sonoff (Topic sonoff1, Fallback DVES_07C2BA, GroupTopic sonoffs) Version 2.0.4b
00:00:04 HTTP: Webserver active on sonoff1-0698 with IP address 192.168.1.3
00:00:04 MQTT: Attempting connection...
00:00:04 MQTT: Connected
00:00:04 MQTT: sonoff1/INFO = Sonoff module version 2.0.4b, Fallback DVES_07C2BA, GroupTopic sonoffs
00:00:04 MQTT: sonoff1/INFO = Webserver active for Admin on sonoff1-0698 with IP address 192.168.1.3
20:10:43 MQTT: sonoff1/1/POWER = On
20:11:06 MQTT: sonoff1/SERIALLOG = 3
20:14:00 MQTT: sonoff1/UPTIME = 0
20:14:00 MQTT: sonoff1/POWER = On
20:14:00 MQTT: sonoff1/TIME = 2016-10-16T20:14:00
```
Seriallog 3 doesn't appear to add anything to the console.

I also tried to snoop on the signal of the unmodified module with an ancient Saleae Logic but I wasn't able to figure out the data even with the datasheet (http://webmeteobox.ru/docs/AM2301.pdf). It's over my head for now but maybe I'll eventually decipher it.

arendst, 2016-10-16:
For the console you use commando weblog 3

mihalski, 2016-10-16:
I feel a little more stupid than usual.

Here's the output I get now:

```
21:06:34 MQTT: sonoff1/WEBLOG = 3
21:06:35 Config: Saved configuration to flash at F9 and count 90
21:09:50 MQTT: sonoff1/UPTIME = 1
21:09:50 DHT: Timeout waiting for start signal low pulse
21:09:50 MQTT: sonoff1/POWER = On
21:09:50 MQTT: sonoff1/TIME = 2016-10-16T21:09:50
```

I guess that's progress of sorts.

arendst, 2016-10-16:
Michal,

I think I know what's wrong. It seems the AM2301 is not a DHT22 but based on SH1x. My driver won't work with the AM2301. I have to insert a driver for SH1x what will take a few days.

Sorry for the inconvenience.... ;-)

mihalski, 2016-10-18:
Thank you for taking the time to look into this. I'm just glad I wasn't doing something stupid that was preventing things from working :)

If you need me to test anything let me know.

Oh, and sorry for the VERY late response!

arendst, 2016-10-18:
Michal, find in experimental sonoff-dht which hopefully solves your problem. Please try and let me know the result.

mihalski, 2016-10-18:
Hi. I don't think we're quite there yet.

```
00:00:00 Config: Loaded configuration from flash at F9 and count 92
00:00:00 APP: Bootcount 37
00:00:00 Wifi: Patch issue 2186
00:00:00 Wifi: Connecting to XxxXxx(xxxxxxxxxxxxx) in mode 11N as sonoff1-0698
00:00:00 DHT: Max clock cycles 80000
00:00:00 APP: Project sonoff (Topic sonoff1, Fallback DVES_07C2BA, GroupTopic sonoffs) Version 2.0.5b
00:00:00 Config: Saved configuration to flash at FA and count 93
00:00:00 Wifi: STATION_IDLE
00:00:01 Wifi: STATION_IDLE
00:00:03 Wifi: STATION_IDLE
00:00:04 HTTP: Webserver active on sonoff1-0698 with IP address 192.168.1.3
00:00:04 MQTT: Attempting connection...
00:00:04 MQTT: Connected
00:00:04 MQTT: sonoff1/INFO = Sonoff module version 2.0.5b, Fallback DVES_07C2BA, GroupTopic sonoffs
00:00:04 MQTT: sonoff1/INFO = Webserver active for Admin on sonoff1-0698 with IP address 192.168.1.3
00:00:10 RTC: (UTC) Tue Oct 18 21:41:09 2016
00:00:10 RTC: (DST) Sun Mar 27 02:00:00 2016
00:00:10 RTC: (STD) Sun Oct 30 03:00:00 2016
07:42:56 HTTP: Handle root
07:42:59 HTTP: Handle console
07:45:59 MQTT: sonoff1/UPTIME = 0
07:45:59 MQTT: sonoff1/RSSI = 100
07:45:59 DHT: Timeout waiting for start signal low pulse
07:45:59 MQTT: sonoff1/POWER = ON
07:45:59 MQTT: sonoff1/TIME = 2016-10-19T07:45:59
```

Any ideas?

mihalski, 2016-10-21:
Any developments?

Am I the only person who actually has one of these devices? Well 2 actually.

Regards,
Michal

arendst, 2016-10-21:
I've ordered some plain Am2301 from ebay and when they're in start testing and tweeking code. So please be patient a little longer.

AlexTransit, 2016-10-21:
I'm working on the implementation of the PID using the temperature sensor. when completed yet I give code.

mihalski, 2016-10-21:
In the meantime I've ordered a couple of the new Sonoff Pow.

Do you have any idea if their new wifi powerplug can be programmed?

https://www.itead.cc/smart-home/psa-1684.html

arendst, 2016-10-21:
The link points to a stripped down version of an esp12 with itead cloud software called PSA to connect a switch and a relay. You still have to add power supply, button, relay and housing.

As it's an esp8266 my software can be loaded too.

mihalski, 2016-10-21:
Ahh! You're right.

That picture with the socket is really misleading!

mihalski, 2016-11-05:
Hi,

Have you had a chance to look into the AM2301 any further?

Regards,
Michal

arendst, 2016-11-05:
My ordered AM2301 is still in the mail somewhere. I let you know when I have it fixed.

mihalski, 2016-11-09:
For what it's worth I came across this when looking for ways to integrate with Home Assistant:

https://github.com/KmanOz/Sonoff-HomeAssistant/blob/master/arduino/ESPsonoff_TH-v1.0t/ESPsonof_TH-v1.0t.ino

https://github.com/KmanOz/Sonoff-HomeAssistant#9-temperature-sensor-installation-for-th1016-sonoff-switch

It would seem like it supports the AM2301 so maybe it can reveal what the problem is?

Considering flashing it to one of the two units I have to see if it works. Also to see if I can get it to work with Home Assistant as I'm running into some confusion as to how topics/payloads work in this implementation.

arendst, 2016-11-09:
I noticed it uses the same DHT library I implemented in my version. I will investigate where I go wrong. BTW I'm still waiting for my sensor to arrive so I can't test at the moment.

arendst, 2016-11-15:
Received my sensor. Does not work with my code as you noticed. Tried adafruit code. Works fine. Implemented adafruit code in mine. Fails again.

Investigating...

mihalski, 2016-11-15:
That's the only way I've ever tried it. Did you actually change any code or did it just spontaneously work after a recompile?

I'll try it again but it sounds fishy.. I'll take the opportunity to upgrade to the latest version.

mihalski, 2016-11-15:
Just followed the procedure with the latest firmware and this was the result:
```
00:00:00 Config: Loaded configuration from flash at FA and count 213
00:00:00 APP: Bootcount 51
00:00:00 Wifi: Patch issue 2186
00:00:00 Wifi: Connecting to WifiNet(password) in mode 11N as sonoff1-0698
00:00:00 DHT: Max clock cycles 80000
00:00:00 APP: Project sonoff (Topic sonoff1, Fallback DVES_07C2BA, GroupTopic sonoffs) Version 2.0.13
00:00:00 Config: Saved configuration (692 bytes) to flash at F9 and count 214
00:00:00 Wifi: STATION_IDLE
00:00:01 Wifi: STATION_IDLE
00:00:03 Wifi: STATION_IDLE
00:00:04 Wifi: STATION_IDLE
00:00:05 HTTP: Webserver active on sonoff1-0698 with IP address 192.168.1.62
00:00:05 MQTT: Attempting connection...
00:00:05 MQTT: Connected
00:00:05 MQTT: Receive topic cmnd/sonoff1/power, data ON
00:00:05 MQTT: DataCb Topic sonoff1, Group 0, Device 1, Type POWER, Data ON (ON)
00:00:05 MQTT: stat/sonoff1/POWER = ON
00:00:05 MQTT: Receive topic cmnd/sonoff1/power toggle, data ON
00:00:05 MQTT: DataCb Topic sonoff1, Group 0, Device 1, Type POWER TOGGLE, Data ON (ON)
00:00:05 MQTT: stat/sonoff1/SYNTAX = Status, SaveData, SaveSate, Upgrade, Otaurl, Restart, Reset, WifiConfig, Seriallog, Weblog, Syslog, LogHost, LogPort, SSId, Password, Hostname, Webserver
00:00:05 MQTT: stat/sonoff1/SYNTAX = MqttHost, MqttPort, MqttUser, MqttPassword, MqttClient, Topic, ButtonTopic, ButtonRetain, MqttUnits, MessageFormat, GroupTopic, Timezone, Light, Power, Ledstate, TelePeriod
00:00:05 MQTT: stat/sonoff1/SYNTAX = DomoticzInTopic, DomoticzOutTopic, DomoticzIdx, DomoticzKeyIdx, DomoticzUpdateTimer
00:00:05 MQTT: stat/sonoff1/INFO = Sonoff module version 2.0.13, Fallback DVES_07C2BA, GroupTopic sonoffs
00:00:05 MQTT: stat/sonoff1/INFO = Webserver active for Admin on sonoff1-0698 with IP address 192.168.1.62
00:00:05 MQTT: Receive topic cmnd/sonoff1/LIGHT, data ON
00:00:05 MQTT: DataCb Topic sonoff1, Group 0, Device 1, Type LIGHT, Data ON (ON)
00:00:05 MQTT: stat/sonoff1/LIGHT = ON
00:00:05 MQTT: Receive topic cmnd/sonoff1/POWER, data ON
00:00:05 MQTT: DataCb Topic sonoff1, Group 0, Device 1, Type POWER, Data ON (ON)
00:00:05 MQTT: stat/sonoff1/POWER = ON
00:00:05 MQTT: Receive topic cmnd/sonoff1/1/POWER, data OFF
00:00:05 MQTT: DataCb Topic sonoff1, Group 0, Device 1, Type POWER, Data OFF (OFF)
00:00:05 MQTT: stat/sonoff1/1/POWER = OFF
00:00:06 Config: Saved configuration (692 bytes) to flash at FA and count 215
00:00:06 MQTT: stat/sonoff1/1/POWER = OFF
00:00:07 RTC: (UTC) Tue Nov 15 10:58:22 2016
00:00:07 RTC: (DST) Sun Mar 27 02:00:00 2016
00:00:07 RTC: (STD) Sun Oct 30 03:00:00 2016
20:58:33 HTTP: Handle root
20:58:36 HTTP: Handle console
21:01:01 RTC: (UTC) Tue Nov 15 11:01:01 2016
21:01:01 RTC: (DST) Sun Mar 27 02:00:00 2016
21:01:01 RTC: (STD) Sun Oct 30 03:00:00 2016
21:02:30 MQTT: tele/sonoff1/UPTIME = 1
21:03:21 MQTT: tele/sonoff1/UPTIME = 1
21:03:21 MQTT: tele/sonoff1/RSSI = 70
21:03:21 DHT: Timeout waiting for start signal low pulse
21:03:21 MQTT: tele/sonoff1/POWER = OFF
21:03:21 MQTT: tele/sonoff1/TIME = 2016-11-15T21:03:21
```
Still doesn't work :(

mihalski, 2016-11-15:
Could it be something else in your sketch doing timing related things that interfere with reading the sensor?

arendst, 2016-11-15:
That's what I thought was happening yesterday when I couldn't get it to work. Today I can't make it fail! Tried on both wemos AND TH10 and it always works as expected.
```
12:06:51 MQTT: tele/th1/UPTIME = 1 Hr
12:06:51 MQTT: tele/th1/RSSI = 98 %
12:06:51 MQTT: tele/th1/TEMPERATURE = 24.1 C
12:06:51 MQTT: tele/th1/HUMIDITY = 43.5 %
12:06:51 MQTT: tele/th1/LIGHT = OFF
12:06:51 MQTT: tele/th1/TIME = 2016-11-15T12:06:51
```

Are you sure your connector makes good contact with the TH1x? Did you power cycle the TH1x

mihalski, 2016-11-15:
I definitely power cycled it.. Going to take the case off and plug in the cable just in case the click it makes when being plugged in isn't far enough.

mihalski, 2016-11-15:
GAD DAMMIT!!
```
21:24:13 MQTT: tele/sonoff1/UPTIME = 0
21:24:13 MQTT: tele/sonoff1/RSSI = 86
21:24:13 DHT: Received 02, 52, 00, FD, 51 =? 51
21:24:13 MQTT: tele/sonoff1/TEMPERATURE = 25.3
21:24:13 MQTT: tele/sonoff1/HUMIDITY = 59.4
21:24:13 MQTT: tele/sonoff1/POWER = OFF
21:24:13 MQTT: tele/sonoff1/TIME = 2016-11-15T21:24:13
```
I should be happy but I'm just annoyed at myself..

There is literally no perceptible difference in the feel and the click of plugging in the sensor with or without the casing.. But unless I've just experienced the same black magic that you had I'm going to call the casing the issue..

Tomorrow I'll test it repeatedly with and without just to be certain, but if I confirm that this was the problem I will simply enlarge the hole so that there is no risk of this happening again..

I guess it doesn't really explain how the other device was getting temp/humidity even with the case on, but I'll investigate further.

Thank you for the effort you've made to address this.

arendst, 2016-11-15:
You made my day!

mihalski, 2016-11-18:
And now for some reason I can't get it to work to save my life whether the case is off or not. Any chance you could make the temp/humidity telemetry readings at the beginning rather than the end of the 300 seconds so that if it's not working it's obvious from the start?

mihalski, 2016-11-18:
And it's definitely not the temperature module.. I've swapped them over from the one that doesn't have any issues.

mihalski, 2016-11-18:
And it finally worked after retrying it a dozen times.. NO idea what the issue is.. Poor contacts? Something else? Alignment of stars and planets?

mihalski, 2016-11-20:
Thank you for the enhancement!

Makes the Frustration-> Anger -> Power Off -> Power On -> Frustration -> Anger -> Repeat loop MUCH tighter.

I don't understand what the problem is but it seems to work maybe 5% of the time.. And then it can fail out of the blue..

mihalski, 2016-11-20:
I'm tempted to automate my frustration by modifying your firmware to send 0 when it can't get a value and having a 2nd sonoff power it off and restart it if that happens :/ I want to know if it has anything at all to do with the contacts or whether it's some weird software timing issue..

arendst, 2016-11-20:
The next release allows for selecting between the current implementation and the adafruit library. Perhaps that solves your problem.

mihalski, 2016-11-20:
Perhaps. I will also try the other firmware on this Sonoff in case it's some sort of physical difference that is causing the problem. This is why I always buy at least 2 of everything :/ Doesn't guarantee that it will help in troubleshooting but it can certainly help..

BTW, I noticed that the other Sonoff firmware implemented a PIR sensor trigger.. I'm considering ordering some suitable plugs to give that a go. Not necessarily as a trigger for on/off events but as a power source and MQTT delivery system.

arendst, 2016-11-23:
Second DHT driver implemented in version 2.0.17 just released

ecsfang, 2016-11-24:
Hi! I have been using version 2.0.14 since release, and i have no problem with my AM2301. I'm using a TH16 with the sensor. I had to doublecheck the connector, since the cover partially covers the hole. But firmly inserting the sensor with a click, and after that everything has worked as expected. 
Best regards,
Thomas

mihalski, 2016-11-25:
I've switched to the second DHT driver and the problem is definitely with the contacts.. With case on, almost always bad, with case off almost always good. It doesn't matter that it clicks in place just as well with the case on there must me a 0.001mm difference that is enough to throw it off.

More interestingly however, and I don't know if this is just a feature of the second DHT library or your version too, but I can unplug and plug the cable back in or plug it in after powering on and almost straight away the temp/humidity comes up.

It may be due to the fact that you've implemented the main menu readings and also initiated readings as soon as the device starts up that it is now MASSIVELY easier to diagnose problems.

Thank you! :)

arendst, 2016-11-25:
Thank you for your input ecsfang. I'll close this issue and suggest to make the hole in the TH casing just big enough to let the connector make good contact ;-)

closed 2016-11-25 by arendst