arendst/Sonoff-MQTT-OTA-Arduino #40: Itead Sonoff Touch

tam481, 2016-10-11:
Hello, please add support for the newly released Sonoff Touch.

Nayar, 2016-11-13:
Is it based on ESP8266 also? 

+1 for Sonoff Touch

arendst, 2016-11-16:
Send me the hardware and I will try to implement it

danjperron, 2016-11-28:
Works well with the sonoff touch. The only thing I had to do was to set the esp8265 and gnd one pin on power up.
[https://dl.dropboxusercontent.com/u/48891705/touch1.png](url)

Slip7000, 2016-12-11:
Please mail me your Address so i can ship them to you! Treid to load the Touch and not working.

danjperron, 2016-12-11:
I got 4 sonoff touch installed. I'm still waiting for 2 others + 1 sonoff and one S20 smart socket.

Try to,

1 -  Put the esp8265 to flash mode by shorting the pin like I show you on my previous post. Short the pin for 2 seconds on 3.3V power up.

2- Set the cpu to be esp8265.

3 - The switch is like the sonoff one.

B.T.W. I did use a esp12E with to relay and it works fine also. I did a video in french about it

https://youtu.be/T4XjFoNEUm8

This is my gig when I tried your code 

[image]

I'm planing to do a video using a C.H.I.P. + sonoff  + esp12E + mosquitto 1.4 QTT + paho.

[image]

Keep up the good work! I like what you did.

arendst, 2016-12-21:
Great demo! In the meantime I received my Touch today and will also try my software on it in the very near future. According to the above information it should be a breeze...

arendst, 2016-12-23:
Sonoff Touch support is fully implemented in version 3.1.1 just released.

See the wiki for specific compile and upload instructions.

closed 2016-12-23 by arendst