arendst/Sonoff-MQTT-OTA-Arduino #41: Itead Sonoff RF

tam481, 2016-10-12:
Hello,
Please add support for Sonoff RF. I hope it won't be too much work :-)

arendst, 2016-10-12:
What doesn't work with the Sonoff RF?

tam481, 2016-10-13:
I don't have one yet. I wanted to buy one but wanted to know if your code will support it. The idea is that the Sonoff will report its status to MQTT whether switched using RF, button or Wifi. If you can confirm that this is the case then please discard the request.

arendst, 2016-11-16:
Send me the hardware and I will try to implement it

prahjister, 2016-11-27:
Have you received an RF version ?

arendst, 2016-11-27:
Not yet but I might in the near future. Just keep monitoring this issue.

merlokk, 2016-12-14:
Thetre are 2 CPU: one on the module https://www.itead.cc/wiki/433Mhz_RF_receiver and one on the main board https://www.itead.cc/wiki/Sonoff_RF.

So. If you need just implement RF control functions with their remote - there is one additional signal - E_FW. it comes to GPIO0 of ESP8266. 

But. If you need some other functionality....   Receive another remotes, sensors etc...  It needs to reprogram 2 microcontrollers EFM8BB10F2G-A-QFN20.

arendst, 2016-12-21:
I received a sonoff RF today and will verify my software soon.

arendst, 2016-12-24:
I managed to get the Sonoff RF programmed with my code using additional help from Phalox. See my wiki regarding specific actions you may need to take to get programming going.

I failed to get the RF part working with my Kaku switches.

closed 2016-12-24 by arendst