arendst/Sonoff-MQTT-OTA-Arduino #42: Stat

TLaren, 2016-10-12:
I am using Node-Red on a Pi.
I have several Sonoff-MQTT Switches up and working fine.
I would like to read the status of them or get a notification when they switch.
I have tried several combinations using the 'stat' and it does not work for me as expected.
Can you give some examples how it should work?

I have it setup as follows:
cmnd/bedroom/light/power   controls the light
I have tried things like:
stat/bedroom/light/power
stat/bedroom/light
cmnd/bedroom/light/info
cmnd/bedroom/light/power/info

arendst, 2016-10-12:
The wiki might explain it all.

TLaren, 2016-10-13:
Thanks for the quick reply.
Wiki:
Most MQTT commands will result in a status feedback like stat/sonoff/POWER On.

I have a 'mqtt in' (in Node-Red) of stat/sonoff/power (or POWER) and have it's output tied to debug and never get a message.
Do I need to set some flag in the user_config.h or something?

Wiki:
show status information by cmnd/sonoff/status 1.

I try sending a '1' to cmnd/sonoff/status and am not sure where the status message back would go.
Having the above 'mqtt in' in setup as I thought it would go there.

arendst, 2016-10-13:
I guess you are almost there regarding node-red. In the below example you'll see a working flow.

[image]

Keep in mind that:
- The UPPERCASE result name (LIGHT or POWER) needs always be the same as the command name (light or power) as the sonoff software will automatically switch to the one that was entered as a command. So in your case if you use ```cmnd/bedroom/light on``` you must expect an answer like ```stat/bedroom/LIGHT On```.
- In order to catch a MQTT message you'll have to use the exact message or use a wildcard as the last topic part. That's what is missing from yours. See the image above where in your case the receiving mqtt node should be ```stat/bedroom/LIGHT/#```.
- To catch all status messages from bedroom you would need to configure a node with Topic ```stat/bedroom/STATUS/#```.

I suppose this will get you on track.

TLaren, 2016-10-14:
Thanks. More working, I just need to change the way I was doing things.
Because of other projects I had dome with MQTT, I had defined devices with many "/" in the topic and it worked for them. Like this:
    Room = bedroom
    Topics
         bedroom/light
         bedroom/fan
         bedroom/heater
So when I wanted to turn on the light I sent cnmd/bedroom/light/power 1
Now I am sending /cmnd/bedroomlight/power 1
I was trying to retrieve the switch state I used /stat/bedroom/light/POWER and it was not working. 
Now I am using /stat/bedroomlight/POWER and it works

I don't know if I was breaking any rules or just compatibility with your code, but it works now and if I get brave and have time I will look into playing with your code and studying the MQTT specs.

Thanks for the great support.

arendst, 2016-10-14:
You have broken no rules but for my sonoff software I had to decide on some kind of rule to handle commands, payyload and results.

With your latest response you gave me a new insight in using MQTT payloads.

tam481, 2016-10-16:
@TLaren how did you get the topic to publish under specific sub-topics? I would like mine to report to home/bedroom/sonoff. Changing the topic to /home/bedroom/sonoff generates the error message below in the console

MQTT: "/home/bedroom/sonoff"/SYNTAX = MqttHost, MqttPort, MqttUser, MqttPassword, MqttClient, Topic, ButtonTopic, ButtonRetain, GroupTopic, Timezone, Light, Power, Ledstate, TelePeriod

TLaren, 2016-10-16:
I am using Node-Red. With the topic like cnmd/bedroom/light/power I was able to control the light but the status did now work.
After the thread with arendst I switched the topic to cnmd/bedroomlight/power (stat/bedroomlight/POWER/#) and I was able to control and monitor the status of the switch.
I then went back to my original topic, thinking that I was just missing the /# at the end and that did not work.
One thing I did not realize is I control with power and stat with POWER. It seems that is required.
In Node-Red, I want a button that toggles the device (using light/power toggle) and an 'LED' indicator that shows the status on the Dashboard. I have not found an indicator node yet and have not played enough to make one myself. It should be fairly simple, just show one of two icons.
arendst stated that he is re-thinking his code, Maybe multiple sub-topics will be supported.

TLaren, 2016-10-16:
Maybe I should open a new issue or...
I have two units now that I changes the topic from the web interface after initial setup and now when I send a power command I get three blinks of the LED and the switch does not change. Same if I press the button on the unit.
I'm not sure what the syntax is from the console interface but I always get ...SYNTAX = MqttHost, MqttPort, MqttUser, MqttPassword, MqttClient, Topic, ButtonTopic, ButtonRetain, GroupTopic, Timezone, Light, Power, Ledstate, TelePeriod...
One of the units I re-flashed and it did not change the actions. I expect there are settings stored on the switch that flashing does not clear.
Tim

TLaren, 2016-10-16:
Interesting, if I initially configure (just after my first flash from factory firmware) with a topic of, for example, dining-room/ceiling/lamp/power, I can control the switch. If I change it to anything with a subtopic and not just a single topic, the switch does not respond with anything but an error in the console and three blinks of the LED. If I then change it back to a topic (without a subtopic) it then works.

Just letting you know.

tam481, 2016-10-17:
Thanks. Do you have a copy of the original firmware? How did you get that and is it unique to your device, if not, are you able to share it?

I'm going to try and flash another firmware then re-flash Sonoff-MQTT-OTA-Arduino with subtopics and see. I also noticed that -t cmnd/sonoff/POWER -m ON no longer works. The only thing that works now is -t cmnd/sonoff/1/LIGHT -m ON in MQTT whereas before, it did work.

TLaren, 2016-10-17:
I do not have a copy of the firmware, but it may be available on Sonoff's site, they have a support forum. I have no interest in their 'cloud' based firmware.
I just have 20 units so there are still some with the original firmware. Also about 10 of these units have just been flashed but I have not done anything else with them.
I noticed that the firmware here was updated about a week ago, did look yet for what has changed.

arendst, 2016-10-18:
As a workaround for your problem regarding the not supported enlarged MQTT topic you could use node-red to solve it. See following flow:

[image]

Where the function contains the below code:

```
msg.topic="cmnd/sonoff3/" + msg.topic.substr(msg.topic.lastIndexOf("/"),20);
return msg;

```
Now you can execute ```cmnd/bedroom/light/power 2``` and node-red will transfer it to a well understood ```cmnd/sonoff/power 2```.

closed 2016-11-08 by arendst