arendst/Sonoff-MQTT-OTA-Arduino #43: MQTT-commands

Lennyz1988, 2016-10-13:
Hi,

I cannot get the MQTT-commands to work. 

The name of the Sonoff: "DVES_06E890"
The topic I subscribed the Sonoff to: "sonoff"

What is wrong with my command that I enter in the terminal?

` mosquitto_pub -t 'cmnd/sonoff' -h localhost -m 'cmnd/sonoff/power toggle'`

It connects fine to my MQTT server (Mosquitto)

> 
> 1476372515: New client connected from 192.168.1.221 as DVES_06E890 (c1, k120, ulennart).
> 1476372515: Sending CONNACK to DVES_06E890 (0)
> 1476372515: Received SUBSCRIBE from DVES_06E890
> 1476372515: 	cmnd/sonoff/# (QoS 0)
> 1476372515: DVES_06E890 0 cmnd/sonoff/#
> 1476372515: Sending SUBACK to DVES_06E890
> 1476372515: Received SUBSCRIBE from DVES_06E890
> 1476372515: 	cmnd/sonoffs/# (QoS 0)
> 1476372515: DVES_06E890 0 cmnd/sonoffs/#
> 1476372515: Sending SUBACK to DVES_06E890
> 1476372515: Received SUBSCRIBE from DVES_06E890
> 1476372515: 	cmnd/DVES_06E890/# (QoS 0)
> 1476372515: DVES_06E890 0 cmnd/DVES_06E890/#
> 1476372515: Sending SUBACK to DVES_06E890
> 1476372515: Received PUBLISH from DVES_06E890 (d0, q0, r0, m0, 'stat/sonoff/INFO', ... (69 bytes))
> 1476372515: Received PUBLISH from DVES_06E890 (d0, q0, r0, m0, 'stat/sonoff/INFO', ... (71 bytes))

No matter what I try I always get this output:

> 16:35:49 MQTT: Receive topic cmnd/sonoff, data cmnd/sonoff/power toggle
16:35:49 MQTT: DataCb Topic sonoff, Group 0, Device 1, Type <null>, data cmnd/sonoff/power toggle (CMND/SONOFF/POWER TOGGLE)
16:35:49 MQTT: sonoff/SYNTAX = Status, SaveData, SaveSate, Upgrade, Otaurl, Restart, Reset, WifiConfig, Seriallog, Weblog, Syslog, LogHost, LogPort, SSId, Password, Hostname, Webserver
16:35:49 MQTT: sonoff/SYNTAX = MqttHost, MqttPort, MqttUser, MqttPassword, MqttClient, Topic, ButtonTopic, ButtonRetain, GroupTopic, Timezone, Light, Power, Ledstate, TelePeriod
> 

I tried all sort of combinations but I always get the same output. It must be something really simple that I am doing wrong. Can anyone guide my in the right direction?

arendst, 2016-10-13:
See mosquitto_pub syntax.

What should work is ```mosquitto_pub -h localhost -t 'cmnd/sonoff/power' -m 'toggle'```

Lennyz1988, 2016-10-14:
Yes that command works as it should. I must have misinterpreted the page about using the syntax. I've spent hours trying to fix it and I knew it must be something really simple. Thanks for the help!

closed 2016-10-14 by arendst