arendst/Sonoff-MQTT-OTA-Arduino #45: Do you have a 'Donate Button' somewhere?

TLaren, 2016-10-13:
I love your project. I find it useful. Your support is outstanding. And basically it all works.
I'd like to buy you a beer, pizza or whatever.
If you have a Donate Button somewhere, please let me know.

Thanks
Tim

arendst, 2016-10-13:
Thnx for the kudos. I don't but will investigate :-)

Nayar, 2016-10-13:
Great idea. Will donate too :)

eazuka, 2016-10-19:
you are doing an appreciated work so you need to be encourage to do more.

TLaren, 2016-10-19:
If you have a PayPal account: https://www.paypal.com/webapps/mpp/get-started/donate-button

timtimsson, 2016-12-31:
Will donate too :)

gizmochief7, 2017-01-14:
Feature Request: Can we make the monetary press button on the Sonoff the donate button? ;) +1

Webflue, 2017-01-31:
Great idea.  I was just thinking what you guys do for the love of it and some people just don't appreciate the work put into something like this. Got to get mine working yet but would like to buy you a beer or any other beverage of your choice :-)