arendst/Sonoff-MQTT-OTA-Arduino #46: OpenHAB2 Switch and status

tam481, 2016-10-16:
Hi,
How can I get OpenHAB to update the status of the switch if manually triggered?

This is what I currently have. Switching works but status update doesn't :-(

{mqtt=">[broker:cmnd/sonoff/1/LIGHT/state:command:ON:ON],>[broker:cmnd/sonoff/1/LIGHT/state:command:OFF:OFF],<[broker:stat/sonoff/1/LIGHT:state:default"}

tam481, 2016-10-17:
Hello all, I've managed to resolve the issue by using a transform file in OpenHAB2 and changed "<[broker:stat/sonoff/1/LIGHT:state:default]
to
<[broker:stat/sonoff/1/LIGHT:state:MAP(sonoff.map)]

This transformation file changed the text type case from On to ON and Off to OFF and  this worked perfectly.

Could you please update the code so that the Sonoff publishes its status in caps if this doesn't cause issues for other programs?

arendst, 2016-10-18:
Since version 2.0.5 all status messages regarding On or Off now return ON or OFF what should satisfy openHAB needs.

arendst, 2016-11-09:
Since version 2.0.10 you can select what message will be returned by configuring defines MQTT_STATUS_ON and MQTT_STATUS_OFF in user_config.h

closed 2016-11-15 by arendst

andrestrauss, 2017-03-11:
Hi I cannot get the status back updated dynamically on my sitemap
This is my code:

Switch Light1 "Living Room Light" <light> (LR,gLight){ mqtt=">[mosquitto:cmnd/sonoff2/power:command:ON:on],>[mosquitto:cmnd/sonoff2/power:command:OFF:off],<[mosquitto:stat/sonoff2/power:state:map(onoff.map)]" }

All i want to happen is the ui to know when the switch is on or off

Any help please

arendst, 2017-03-11:
Andres, this issue is closed in november last year. do not expect any answer here.

I suggest you start using the superseded version Sonoff-tasmota. There you find an active openhab community and wiki that  may help .

andrestrauss, 2017-03-12:
Hi Theo. Thanks, Please provide link to superseded version