arendst/Sonoff-MQTT-OTA-Arduino #47: Itead 4 Channel WiFi Wireless Switch

pacitzu, 2016-10-19:
Are you planning to support this device?

https://www.itead.cc/4-channel-wifi-wireless-switch.html

Thank

arendst, 2016-10-19:
Looking at the information on the itead website it might work. It seems to me as that they integrated an ESP8266 with the relay control chip used in Sonoff Dual and call it PSB.

Although the documentation contains lots of NC (not connected) from the PSB PinMap I think I see external NC connections to among others TX, Rx and GPIO0 allowing you to upgrade the firmware.

At the moment I will not invest in the hardware as I'm already swamped with sonoffs of all kinds but feel free to provide one and I will have a look into it.

arendst, 2016-10-22:
See Experimental/sonoff-4chnl for initial support. See _releasenotes.ino for additional information.

arendst, 2016-11-09:
Implemented in current version

closed 2016-12-17 by arendst