arendst/Sonoff-MQTT-OTA-Arduino #48: Setting time zone

TLaren, 2016-10-22:
Timezone is set to 99.
Can/how do you set it up for DST/STD and Pacific Time?

When I watch the serial on boot I see:
00:00:10 RTC: (UTC) Sat Oct 22 03:00:34 2016
00:00:10 RTC: (DST) Sun Mar 27 02:00:00 2016
00:00:10 RTC: (STD) Sun Oct 30 03:00:00 2016

I'm a little confused, why so many days between these?
My current time is 8:02 PM Pacific  DST.

Once it get NTP sync it goes to shows:
05:02:30 MQTT: oled/TIME = 2016-10-22T05:02:30

PS: Donate Button?

arendst, 2016-10-22:
If timezone is set to to 99 it uses parameters from user_config.h to switch from/to daylight savings. The parameters are:
```
// Time - Start Daylight Saving Time and timezone offset from UTC in minutes
#define TIME_DST               Last, Sun, Mar, 2, +120  // Last sunday in march at 02:00 +120 minutes

// Time - Start Standard Time and timezone offset from UTC in minutes
#define TIME_STD               Last, Sun, Oct, 3, +60   // Last sunday in october 02:00 +60 minutes
```
Currently they are set for Central European Time where on the last sunday in March at 02:00 the clock advances 60 (=120 from UTC) minutes to start Daylight Savings. On the last sunday in October at 03:00 the clock returns to standard time (=60 from UTC).

In your case you'll have to figure out when daylight savings is initiated and standard time is restored and then change the parameters for both TIME_DST and TIME_STD. The text is syntax picky so you might have to play with it.

Next weekend I'm supposed to go back one hour and keep my fingers crossed ;-).

TLaren, 2016-10-22:
OK thanks.
I thought the 120 and 60 were the change, not the actual offset.
I'll give it a try, and if it does not work have to move to Central European Time Zone.

If that works I'll let you know and if not maybe meet you at the local pub.

TLaren, 2016-10-22:
Works like a champ;
// Time - Start Daylight Saving Time and timezone offset from UTC in minutes
//#define TIME_DST               Last, Sun, Mar, 2, +120  // Last sunday in march at 02:00 +120 minutes
#define TIME_DST               Second, Sun, Mar, 2, -420  // Second Sunday in March at 02:00 -420 minutes UTC-8

// Time - Start Standard Time and timezone offset from UTC in minutes
//#define TIME_STD               Last, Sun, Oct, 3, +60   // Last sunday in october 02:00 +60 minutes
#define TIME_STD               First, Sun, Nov, 2, -480   // First sunday in November 02:00 -480 minutes UTC-7

closed 2016-11-08 by arendst