arendst/Sonoff-MQTT-OTA-Arduino #49: Security implementation

tam481, 2016-10-22:
Hi @arendst 
Do you have any plans to implement TLS communication and authentication in the web configuration portal?

arendst, 2016-10-22:
Other than disabling the webserver no. HTTP was implemented only for configuration. When configured disable webserver and use MQTT.

davidbuzz, 2016-10-26:
with regard to the security of MQTT, the most common recommendation for how to secure MQTT is to use TLS.  Is TLS supported for MQTT?    http://iotdesign.embedded-computing.com/guest-blogs/transport-level-security-tradeoffs-using-mqtt/

arendst, 2016-10-26:
As said above, I'm not planning TLS or any other additional security measure for the simple reason I do NOT WANT TO USE CLOUD FEATURES. That means run local MQTT broker, NTP server and DNS server (preferrably in VirtualBox on my NAS) and stay away from hypes like Instagram, Facebook, Twitter and IoT Cloud.

In my opinion the whole idea behind Home Automation is switching lights when it gets dark and when people enter a room. Monitor power usage and act accordingly.
Not being able to switch my lights in the Netherlands from a location in Singapore is a non-issue. It's marketing hype making you dependent on a paid internet cloud service and vulnerable to intrusion.

Oops, I had to get that out of my system knowing many people live security and think they are safe,

I might be looking in MQTT TLS just for the fun though...

JZacharie, 2016-11-01:
Currenlty TLS 1.2 is'nt working due of ESP Firmware wrong implementation.

I will try to implement TLS 1.1 to the MQTT. for that we need change the MQTT library.
The library compatible TLS is Adafruit_MQTT.h instead pubsubclient. 

To limit mosquitto to TLS 1.1 I have add tls_version tlsv1.1 after the certificate.

# MQTT over TLS/SSL
listener 8883
cafile /etc/mosquitto/certs/chain-ca.pem
certfile /etc/mosquitto/certs/cert.pem
keyfile /etc/mosquitto/certs/privkey.pem
tls_version tlsv1.1

Regards

mertenats, 2016-11-02:
You can use MQTT over a TLS connection by using `WiFiSecureClient` instead of `WiFiClient`. Article: https://io.adafruit.com/blog/security/2016/07/05/adafruit-io-security-esp8266/
Sonoff alternative firmware with TLS: https://github.com/mertenats/Sonoff

JZacharie, 2016-11-07:
It was WiFiClientSecure instead WiFiSecureClient :)

TotallyInformation, 2016-11-19:
+1 for this enhancement - we already have far too many insecure IoT devices around & currently the Sonoff is wide open to abuse.

Though please don't take the above negatively. I've done in an evening what it would have taken me weeks to do without your code! Many thanks for the massive effort you've put into this.

arendst, 2016-11-20:
If we all kept our sonoff connection on premise we would not have a security issue. That's why I host my own local MQTT broker, Home automation tool, NTP and DNS server. No need to direct contact the internet. If your IoT devices DoD just configure your firewall to not let them access internet.

TotallyInformation, 2016-11-20:
Not quite I'm afraid. You are assuming that people have secure Wi-Fi and that is far from guaranteed I'm afraid. As a security specialist, IoT security is a major concern and one of the biggest growth areas of threats right now. Even when you do manage to start with a secure network, it is all too easy to make later changes that have significant impacts.

Indeed, I'm also seeing many blog posts, even about the Sonoff, recommending the use of public cloud based MQTT brokers.

So yes, *I* get the need to keep this safe and secure but I'm a specialist and even for me it isn't always straight-forwards, for people with minimal security knowledge ...

It is up to leaders like us to make sure that things are secure and not to ignore the issues.

tam481, 2016-11-20:
I'm no security specialist but I'm very security concious and my request was to ensure that I can follow security best practice by encrypting communications within my own wifi network. Should my wifi be compromised, at there would be another layer of security before my broker and sensors can be compromised as well.

Personally, I agree with running one's own services locally and not relying on cloud services. In my view, no one knows that I run IoT devices, OpenHAB or MQTT and I have the firewall on my router enabled. I appreciate that does not mean that my setup is therefore secure from hacking attempts but I am considerably less valuable as a target than major cloud service providers because of the value and the number of systems cloud providers would present for potential hackers to compromise.

dagbdagb, 2016-11-20:
I wholeheartedly agree with you all. The value of this software is great, and I am very grateful for it. Thank you, Theo.

Also: IoT-devices generally have no business being on the Internet, and in that perspective one could make the argument of not adding a default gateway even if passed by DHCP-options. Maybe require the owner to add a gateway manually and only after having checked a box where the user acknowledges understanding the implications. Or only accept non-default routes passed by the DHCP-server. Or all of that.

I expect this isn't entirely non-trivial, given that the IPstack is "embedded" in the ESP8266. But I have no clue about the details here, really. I know what I don't know. The point is that the general population have no clue whatsoever about network security at all. They may not even have a firewall. They don't know what they don't know.

That said, adding proper security (in the form of encryption) is largely orthogonal to keeping the device off the Internet. One does not exclude the other. Having this project do both would set a wonderful example to anyone doing IoT-projects.

@arendst : Will you accept contributed code implementing TLS? Not offering my assistance, as I lack the required skills and time to obtain them. But someone else may....

arendst, 2016-11-20:
If someone has a viable TLS implementation for the ESP8266 just let me see and I may incorporate it in this software.

TotallyInformation, 2016-11-24:
I fully understand the difficulties and hopefully a viable stack for ESP8266 gets more reliable.

However, to those who think that having their devices on their own network = security - I'm afraid I must disagree. Your home network cannot be guaranteed to be secure when running consumer kit with limited configuration. It is very common for botnets to hack through consumer routers and from there jump to anything within your network. It is also all to easy to end up with security issues over time as you add and change things on your network.

This is not a theoretical issue, it is a current, real problem and not a problem that stays with you, compromised home networks are massively adding to the insecurity of the Internet and adding to DDOS and other issues.

People using this software are *leaders* in IoT and we have a responsibility not to be screwing things up for everyone else but rather we should be making things better.

arendst, 2016-11-26:
I have implemented MQTT on TLS using WifiClientSecure and current PubSubClient. While keeping most available features (I had to shorten weblog to 10 lines) it's to memory hungry leading to instabilities.

Just to let know...

arendst, 2016-12-10:
MQTT over TLS available since version 2.1.0. Give it a go.

dagbdagb, 2016-12-12:
Will do.

closed 2016-12-27 by arendst

rnollet, 2017-10-17:
Hi, what version of TLS have you implemented?