arendst/Sonoff-MQTT-OTA-Arduino #50: Upgrade error

Beetle-II, 2016-10-24:
Hello.
I'm trying upgrade from version 1.0.32 to version 2.0.5.
Through the Web I get the message:
```
Upload failed
Update error code 2
Update error code 8
```

Through OTA I get the message:
`Failed HTTP error: connection refused`

Beetle-II, 2016-10-24:
I compared the information from the two modules and found the difference:
Flash size 1024kB
Sketch flash size is **4096K**

It seems I had forgotten to change the size of the left and 4096K.

It is possible to flash without connecting FTDI?

arendst, 2016-10-24:
Nope. You''ll need ftdi.

Beetle-II, 2016-10-28:
Hello.
I installed 2 Sonoff relay.
I want to when you two short presses the button instead of switching the relay send mqtt command to switching of the second relay.

Do not figured out where in the code to insert the following command.
If possible please help me ...

arendst, 2016-10-28:
For version 2.0.6 look in sonoff.ino around line 1185

patbeirne, 2016-10-28:
Beetle,
You can also try using the "buttontopic" feature.

For example, if you have two sonoffs called: "kitchen" and "hallway", you can set the buttontopic on the hallway sonoff so that it sends out the string "kitchen". Then you can use the hallway sonoff button to control the **_other_** sonoff.

See also: [https://github.com/arendst/Sonoff-MQTT-OTA-Arduino/wiki/Control-a-Sonoff-using-a-remote-button]

Pat

closed 2016-11-13 by Beetle-II