arendst/Sonoff-MQTT-OTA-Arduino #51: random reboots and relay clicking...  watchdogs and exceptions

davidbuzz, 2016-10-25:
setup:
connects to my local wifi, no problem.
connects to my free MQTT instance at http://www.cloudmqtt.com/ no problem

issue: 
resets / reboots randomly,  sometimes every few seconds, sometimes stays up for 5minutes.

tested with:
Arduino IDE 1.6.12  ( latest )
esp8266 board manager plugin version 2.3.0 ( latest )
Master github code for Sonoff-MQTT-OTA-Arduino from 18th Oct ( practically latest )

notable changes:   I'm using WIFI_CONFIG_TOOL  => WIFI_MANAGER

more details cleaned from the "webserver information" screen over the course of just a few minutes: 
```
Version	2.0.5
Core/SDK version	2_3_0/1.5.3(aec24ac9)
Uptime	0 Hours
Flash write count	114
Boot count	86
Reset reason	Hardware Watchdog
 
SSId (RSSI)	PRETTYFLYFORAWIFI (78%)
Hostname	sonoff-5406
IP address	192.168.192.71
Gateway	192.168.192.1
MAC address	60:01:94:07:B5:1E
 
MQTT Host	m12.cloudmqtt.com
MQTT Port	19498
MQTT Client and
 Fallback Topic	DVES_07B51E
MQTT User	******
MQTT Password	********
MQTT Topic	sonoff
MQTT Group Topic	sonoffs
 
ESP Chip id	505118
Flash Chip id	1327343
Flash size	1024kB
Sketch flash size	1024kB
Sketch size	371kB
Free sketch space	568kB
Free memory	32kB

------------------------------------------------------------
Boot count	87
Reset reason	Software/System restart

------------------------------------------------------------
Boot count	88
Reset reason	Hardware Watchdog
------------------------------------------------------------
Boot count	90
Reset reason	Hardware Watchdog
------------------------------------------------------------
Boot count	91
Reset reason	Hardware Watchdog
------------------------------------------------------------
Boot count	94
Reset reason	Hardware Watchdog
------------------------------------------------------------

Boot count	95
Reset reason	Exception
------------------------------------------------------------
Boot count	97
Reset reason	Hardware Watchdog
------------------------------------------------------------
Boot count	98
Reset reason	Exception
------------------------------------------------------------
Boot count	99
Reset reason	Hardware Watchdog
------------------------------------------------------------
Boot count	100
Reset reason	Hardware Watchdog
------------------------------------------------------------
```

arendst, 2016-10-25:
Looking at the Reset reason it seems somewhere in the code a timeout is occurring initiating the watchdog. Using serial connection and command ```seriallog 3``` might show more information. I would expect to see more information with the Exception Reset reason.

I've signed up to CloudMQTT too and did some testing but it seems to work fine with me.

jwildeboer, 2016-10-25:
Wild guess: How is your ESP8266 powered? I have seen randmom reboots like this quite reliably with cheap USB wall warts that don't deliver stable power. Hook it up to a known good power (USB from laptop/PC?) and see if that helps.

jwildeboer, 2016-10-25:
Wild guess: Maybe the incoming power line isn't stable or catching a lot of inductive load leading to spikes/surges? Try another plug or add a surge filter. That might help.

davidbuzz, 2016-10-26:
further feedback from todays re-testing: 
* unit has not been reflashed since above symptoms, or reconfigured or anything, just opened the case and attached FTDI to programming header so I could 'seriallog 3'
* unit is no longer rebooting / resetting.    I've pressed all the things I did yesterday, and can't make it repeat. 
* only differences are: ( 1 ) case is removed ( 2 ) ftdi is attached ( rx, tx, 3.3v, gnd ) and ( 3 ) airconditioning is running today and was not yesterday. ( room temp yesterday was around 29C, and today is around 22C )

So, at this point, I've got to say that either the unit suffers from heat-related issues, OR, it's somehow related to the addition of 3.3v from the FTDI to the CPU?

Further testing of my theory is ongoing, but at this point unit has been up/online for nearly 40minutes without a reboot, which is a record for me.  :-)

davidbuzz, 2016-10-26:
further testing.....   previously unit continues to function "normally" with only tx,rx,gnd ( no 3.3v) connected.      I plan to re-case it, and see if I can get symptoms to reappear, and see if the moderate heat insrease affects it.

closed 2016-11-15 by arendst