arendst/Sonoff-MQTT-OTA-Arduino #52: WKaku 10k resistor

simoneberchielli, 2016-10-25:
Hello, I was looking at WKaku schematics and I noticed 10k resistor for GPIO15 is missing. What' s the purpose of that resistor?

arendst, 2016-10-25:
The 10k R keeps GPIO15 low during restart so the esp8266 doesn't start in update mode. See esp8266 datasheets.

In wkaku you'll need to use the R between GPIO15 and the PCA. Also see my wiki.

simoneberchielli, 2016-10-27:
Thank you for reply, I checked both documents. I built WKaku on breadboard but I'm having some issues. I'm using an ESP-12F module with all memory erased, v2.0.6 of your project and Arduino IDE 1.6.12. I managed to upload FW by pressing button and browsing web server for configuration. Led working also. But board is recognized as Sonoff Dual, button and relay don't work. I checked all connections, I cannot figure out what i'm doing wrong

arendst, 2016-10-27:
During initial power up I'm testing for Sonoff or Sonoff dual by checking pin GPIO12 where the sonoff has  a relay connected with resistors to Gnd. The Sonoff Dual does not.

In your case this test fails and it selects Sonoff Dual.

I was expecting this and implemented a command to easily overcome this. Try command ```model 1```. This should solve your problem.

simoneberchielli, 2016-10-27:
`model 1` in command console makes it working, thanks! I tryied also `model 2` to get back to Dual. In case of Dual, what's GPIO configuration to use 2 relays? In previous release (2.0.5) there were some specific parameters in user_config.h as I remember. Do you remove from 2.0.6?

arendst, 2016-10-27:
Dual does not use GPIO for relays. It has a dedicated chip connected to the serial interface using serial commands to switch relais. See Sonoff Dual schematics.

simoneberchielli, 2016-10-31:
I saw the PSB schematic and I found piece of code where serial commands are sent. Sorry for my noobish questions but I'm new to your project and I would have more time to read all code. Thank you!

closed 2016-11-08 by arendst