arendst/Sonoff-MQTT-OTA-Arduino #53: 1 topic for command and status

StaliXx, 2016-10-27:
Good day. I use majordomo system and in it to display the status of the switch is used MQTT topic which can be 0 or 1. I changed it in the code. When you toogle the switch system sends 0 or 1 in topic which use for status. But when I try to use the same topic for commands and status it doesn't work. I would like to ask if there is way to manage sonoff same MQTT topic which is published status.
Thank you.

arendst, 2016-10-27:
I deliberately made a difference between sending (cmnd/) and receiving (stat/ or tele/) MQTT topics as otherwise the unit would become in limbo switching the relay on/off forever with command ```light toggle```.

If you want them to be the same just change it in user_config.h and see what happens.

Advant2, 2016-11-08:
Good afternoon. I use Majordomo system and in it to display the switch state is used MQTT theme, which can be 0 or 1. Can it be in the settings (user_config.h)  It is possible that the system sends a command  ( stat/sonoff/POWER - (ON - OFF) or (1,0)). Thank you very much.

arendst, 2016-11-08:
Sounds good. I will evaluate.

arendst, 2016-11-09:
Implemented in version 2.0.10

Advant2, 2016-11-09:
Everything works well. Many thanks for the quick change of the program. Many thanks.

closed 2016-11-09 by arendst