arendst/Sonoff-MQTT-OTA-Arduino #55: Problems compiling latest 2.0.6

swapper82, 2016-10-30:
Getting compiler errors, no idea if this is only on my side.

```
sonoff:1370: error: 'class HardwareSerial' has no member named 'baudRate'

  if (Serial.baudRate() != Baudrate) {

             ^

exit status 1
'class HardwareSerial' has no member named 'baudRate'
```

```
 if (Serial.baudRate() != Baudrate) {
    snprintf_P(log, sizeof(log), PSTR("APP: Need to change baudrate to %d"), Baudrate);
    addLog(LOG_LEVEL_INFO, log);
    delay(100);
    Serial.flush();
    Serial.begin(Baudrate);
    delay(10);
    Serial.println();
  }
```

arendst, 2016-10-30:
class HardwareSerial is part of the ESP8266 core for arduino package and introduced in version 2.3.0.

swapper82, 2016-10-30:
Ah ok! 
That's why i did not find it in the normal reference for serial.
http://esp8266.github.io/Arduino/versions/2.3.0/doc/reference.html#serial
Its a esp8266 only thing.

Think i where running the staging version 2.3.0-rc2 but that did not work.
Managed after some manual removal of ESP boards config to install the stable 2.3.0 and it now compiles.
Thanx for a very quick reply!

closed 2016-10-30 by arendst