arendst/Sonoff-MQTT-OTA-Arduino #56: Turn on and sendStatus on boot

palitu, 2016-10-31:
Firstly - Kudos on the project

secondly - i have a request, can you please have the unit turn on, and send the stat/sonoff/LIGHT topic on boot.

I am integrating with Home Assistant, but my units are also switched by the mains (due to WAF).  so they are regularly turned off.  Ultimately, whenever the unit gets power, i want it to automatically turn on and send it's status out.

I have been playing with setting it at the end of setup(), just after the relay pin is assigned.  However i am not 100% sure on how to implement.  I have looked at trying to emulate a button press:

```
char scmnd[20];
snprintf_P(scmnd, sizeof(scmnd), commands[1]);
strtok(scmnd, " ");
send_button(scmnd);
do_cmnd(scmnd);
```

but to be honest - i don't really understand C that well.

any help would be super!

cheers,

arendst, 2016-10-31:
Request implemented in version 2.0.7a. You can download it from repository Experimental.

Give it a try and let me know.

palitu, 2016-11-01:
Thanks - from what i can see, you simply create a variable that gets is tested against every second.  If the variable <> 0, it will call do_cmnd()

palitu, 2016-11-02:
I have found an issue with it - i will finish my investigations and post up the changes that i made - see if you want to include in the master.

cheers!

palitu, 2016-11-03:
I created a new definition
```
#define POWERON_STATE          1            // Immediately turn power on on power up. (0 = disable, 1 = enable)
```

I needed it to turn on ASAP, so i added the following at the end of setup()
```
#ifdef POWERON_STATE
    setRelay(1);  
#endif // Power On immediately on power up
```

Then i modified the code you added in every_second()
```
snprintf_P(svalue, sizeof(svalue), PSTR("%d/%s %d"), i +1, sysCfg.mqtt_subtopic, **1**);
```
I think the original logic actually toggled the state of the relay and reported it's value.
```
snprintf_P(svalue, sizeof(svalue), PSTR("%d/%s %d"),  i +1, sysCfg.mqtt_subtopic, (power & (0x01 << i)) ? 1 : 0);
```

Thanks so much for your input - and you need to get your donate button up and running!

cheers

closed 2016-11-10 by arendst