arendst/Sonoff-MQTT-OTA-Arduino #58: Wifi password in repo

mikaelsundin, 2016-11-05:
Great project, i am looking into make some similar. 
I will go for sonoff RF + a attiny85 to decode nexa(i thing nexa is swedish brand...).

It may be some security issues to have your ssid and password checked in the repo.

arendst, 2016-11-05:
Ah there I have it. I was wondering when this would come about ;-)

My educated guess is that you won't be able to connect to my wifi router from Sweden. So is my neighbour living 200 meters away.

Nayar, 2016-11-08:
Is it possible to have a setting like this: `user_config.h` and `user_config.default.h.`?

In the git repo, only `user_config.default.h` is commited. Then if `user_config.h` is present, the latter overrides the values in `user_config.default.h`. Code would be separated from user data.

drhelmer, 2016-12-14:
Separating the users user_config.h from the default would be great, but I had quite a problem today updating from git to get the latest version, and then scroll through the changed user_config.h and diff it with my customized user_config.h. And as the format of user_config.h changes from release to release, it would be even better if we could just have a separate file with overrides for the values in user_config.h, like user_config_override.h. In that file you could put all the values that you would like to override in user_config.h when compiling new firmware. Obviously the way it is now you need to edit user_config.h each time you get a new version of this software, as you would like to be able to reset to your own config values, not the default ones supplied!

arendst, 2016-12-15:
There is no urgent need to copy your changes everytime a new version is released because your settings are being saved in flash and I put a lot of effort in keeping your settings available with every release. So without changing the define CFG_HOLDER in user_config.h your changes don't do anything anyway...

drhelmer, 2016-12-15:
OK, but what about new defines in user_config.h that you want to apply, I guess if you change contents of the new user_config you must do a reset to get them into flash, or change the CONFIG_HOLDER define?

thomasf, 2016-12-15:
It should also be possible to rearrange user_config a bit so that gcc build define flags can be used modifying the source code at all.

drhelmer, 2016-12-15:
@arendst Right, so if you want updated defines, you need to do a reset/change CFG_HOLDER, which reads in all default values that you once changed to your own, like wifi SSID etc. and overwrites them with the ones in the new user_config.h, so to avoid that you must painstakingly go through a diff of user_config.h and change defaults to the ones you need, I dont like that, really.

arendst, 2016-12-15:
gcc build define overrides won't work in Arduino IDE. I've tried it, tested it and failed it.

arendst, 2016-12-17:
I've introduced a kind of user override config file (user_config_override.h - what's in a name) with version 3.0.7 just released.

I think it won't make you happy but play with it until you notice that updating user_config.h is still needed to enable some features.

closed 2016-12-27 by arendst