arendst/Sonoff-MQTT-OTA-Arduino #60: Web button Toggle does not toggle relay on sonoff

hjgode, 2016-11-06:
When using the Toggle button on the main web screen, the relay does not toggle but the URL of the main page changes to "http://192.168.0.121/?o=1" and the web page shows ON or OFF correctly. When I change this to "http://192.168.0.121/?o=0" the relay is switched off and then a change back to http://192.168.0.121/?o=1 switches the relay to on.

Can you help? I tried to figure out hoe the toggle works but could not fix it.

patbeirne, 2016-11-08:
If you use the URL ""http://192.168.0.121/?o=1" that will toggle the relay. Please check the relay operation. If the web pages shows ON and OFF, then the relay is probably going "on" and "off".

If you are using MQTT commands, you can use '0' to turn the relay off, '1' to turn it on, and '2' to toggle it.

hjgode, 2016-11-08:
Hi

yes I know how it works but I assume the relay has to toggle if I
click on the Main Menu web site on the [Toggle] button. If I do so,
only the URL changes but neither the status text ON/OFF changes nor
toggles the relay.

~Josef

hjgode, 2016-11-08:
Ther is an issue with the html action code if only one relay is there (standard sonoff). I took a look at the html code in browser and it says:

		<form action='/?o=1' method='post'>
		<div style='text-align:center;
		font-weight:bold;
		font-size:60px'>
		ON</div>
		<br/>
		<button>
		Toggle</button>
		</form>

But to toggle the relay it should use /?o=2.

In webserver.ino it is:

      page += F("<table style='width:100%'><tr>");
      for (byte idx = 1; idx <= Maxdevice; idx++) {
        page += F("<td style='width:");
        page += String(100 / Maxdevice);
        page += F("%'><form action='/?o=");
        page += String(idx);
        page += F("' method='post'><div style='text-align:center;font-weight:bold;font-size:");
        page += String(70 - (Maxdevice * 10));
        page += F("px'>");
        page += (power & (0x01 << (idx -1))) ? "ON" : "OFF";
        page += F("</div><br/><button>Toggle");
        if (Maxdevice > 1) {
          page += F(" ");
          page += String(idx);
        }
        page += F("</button></form></td>");

which may work for multiple relays but not for standard sonoff.

hjgode, 2016-11-08:
The executing code in webserver.ino says:

      if (strlen(webServer->arg("o").c_str())) {
        snprintf_P(log, sizeof(log), PSTR("HTTP: handleRoot arg='%s'"), webServer->arg("o").c_str());
        addLog(LOG_LEVEL_DEBUG, log);
	#ifdef MQTT_SUBTOPIC
        snprintf_P(svalue, sizeof(svalue), PSTR("%s/%s 2"), webServer->arg("o").c_str(), sysCfg.mqtt_subtopic);

        snprintf_P(log, sizeof(log), PSTR("HTTP: handleRoot: %s/%s 2"), webServer->arg("o").c_str(), sysCfg.mqtt_subtopic);
        addLog(LOG_LEVEL_DEBUG, log);
	#else
        if (atoi(webServer->arg("o").c_str()) == 1) {
          snprintf_P(svalue, sizeof(svalue), PSTR("light 2"));
        } else {
          snprintf_P(svalue, sizeof(svalue), PSTR("%s/light 2"), webServer->arg("o").c_str());
        }
	#endif
        snprintf_P(log, sizeof(log), PSTR("HTTP: handleRoot: doCmd '%s'"), svalue);
        addLog(LOG_LEVEL_DEBUG, log);

        do_cmnd(svalue);

As MQTT_SUBTOPIC is defined as "POWER" by default the command svalue will be:

       1/POWER 2

Maybe this should be only 
       POWER 2

arendst, 2016-11-08:
As Pat said above, You are wrong in the assumption that the number after ?o= should control the relay; The number after ?o= addresses the relay, not the required state. It will always toggle depending on the current state the relay is in.

It works for both Pat and me so the software is fine.

If your relay doesn't switch it might be that the software has not recognized your hardware correctly. For a sonoff with one relay the second field in the plain ```status``` message should be 1. For a Sonoff Dual this should be 2. Do not expect to switch the first relay on a Sonoff when the software has detected a sonoff Dual as addressing relays works totally different in a Sonoff Dual.

Easy check: How many toggle switches do you see on your webpage? If it's two on a simple Sonoff then you'll have to convince the software once that it is a simple sonoff. That's when undocumented command ```model 1``` comes in play.

hjgode, 2016-11-08:
Hello and Thanks for the clarification.

I only see one Toggle button on the web server Main Page.

Doing some more tests and looking at the code I wonder...

I can use 

    mosquitto_pub -d -h atom2 -t cmnd/sonoff2/POWER -q 1 -m "2"

to toggle the relay (I renamed the topic sonoff to sonoff2 as I have several sonoff devices). With "1" and "0" I can switch ON and OFF.

I can also use 

    mosquitto_pub -d -h atom2 -t cmnd/sonoff2/1/light -q 1 -m "2"

(or "0" or "1") to control the relay. But the web server code does not work for me.

hjgode, 2016-11-08:
I  now cloned the project again and tried with my Arduino IDE 1.6.11 compiled firmware and web toggle does not work. Before I tried to add some debug lines but they were never been used...

Now I flashed the firmware coming with the git clone of the project and web toggle does work as expected. So, the arduino code or whatever is NOT Arduino 2.x compatible. Seems that I have to install arduino ide 2.0.3 :-((

thanks for your patience and help.

~Josef

closed 2016-11-08 by hjgode

hjgode, 2016-11-10:
I removed my arduino prefs dir and installed (unpacked the arduino files, Linux Debian 8) a new arduino IDE (1.6.12) and added esp8266 v2.3.0 via boards manager.

Now I can compile a correct working firmware without the Toggle issue. Very weird, I do not know what was wrong with my install and there was no compile warning or error.