arendst/Sonoff-MQTT-OTA-Arduino #62: Sonoff Pow Voltage reading

Toshibass, 2016-11-08:
Hi arendst I noticed this post...

_With my implementation (2.0.5a in experimental repository) it has Voltage (V), Current (A), Current power (W), Power Factor, Period power (Wh) and daily power (kWh). I do not think frequency detection is possible.

17:50:18 MQTT: sonoff/TODAY_POWER = 0.256 kWh
17:50:18 MQTT: sonoff/PERIOD_POWER = 18 Wh
17:50:18 MQTT: sonoff/CURRENT_POWER = 999 W
17:50:18 MQTT: sonoff/POWER_FACTOR = 0.99
**17:50:18 MQTT: sonoff/VOLTAGE = 221 V**
17:50:18 MQTT: sonoff/CURRENT = 4.558 A_

However I do not get this result, with relay open no load I get 0v, when relay is closed with no load I get 0v to 1v, when relay is closed with load I get max 5v

arendst, 2016-11-08:
I noticed too that the readings are way off. I'm working on a solution.

Toshibass, 2016-11-08:
ok great, so I assume your sonoff_pow arrived then.

The 5v reading was with my 50w soldering iron plugged in 

sonoff_pow            clip on ammeter
0.122A                    not enough decimal places to get a reading
14W
5V

I Just did a further test on a fan heater and compared readings with my clip on type ammeter (not calibrated) house volts = 246v

sonoff_pow            clip on ammeter
3.983A                    7.8A
816W                      7.8Ax246V=1918W  i guess thats the 2kw setting
189V

2.067A                    4.2A
437W                      4.2Ax246V=1033W  i guess thats the 1kw setting
102V

Still getting resets, fan heater occasional + when relay is deactivated + and when no load, anything with a transformer is worse and some led lighting crazy every 2 seconds, are you seeing this problem.

I just tried modifying the user_config.h to **sonoff** instead of **sonoff_pow** and re-flashed, result its rock solid no resets with anything, doesn't matter what I plug in.

Toshibass, 2016-11-08:
ok great, so I assume your sonoff_pow arrived then.

The 5v reading was with my 50w soldering iron plugged in

sonoff_pow________ clip on ammeter
0.122A _____________not enough decimal places to get a reading
14W
5V

I Just did a further test on a fan heater and compared readings with my clip on type ammeter (not calibrated) house volts = 246v

sonoff_pow ________clip on ammeter
3.983A _____________7.8A
816W ______________7.8Ax246V=1918W i guess thats the 2kw setting
189V

2.067A _____________4.2A
437W ______________4.2Ax246V=1033W i guess thats the 1kw setting
102V

Still getting resets, fan heater occasional + when relay is deactivated + and when no load, anything with a transformer is worse and some led lighting crazy every 2 seconds, are you seeing this problem.

I just tried modifying the user_config.h to sonoff instead of sonoff_pow and re-flashed, result its rock solid no resets with anything, doesn't matter what I plug in.

arendst, 2016-11-08:
Updated to 2.0.8 providing major Pow code changes and calibration.

Toshibass, 2016-11-08:
wow 2.0.8 is a vast improvement, its completely solved my reset problem and the readings appear to be much closer ... superb, I was beginning to think this sonoff_pow was duff, 

One point perhaps because I disabled  DOMOTICZ and WEBSERVER I encountered a small problem compiling ... issue was in Sonoff.ino  line 545    **domoticz_update_timer = 2;**  domoticz_update_timer was not declared in this scope .. I just // the line and it works fine for me.

I'm having a bit of an issue with TelePeriod at the moment I'm seeing intermittent readings like constant load but readings are fine for a while then all values are zero for a while then it corrects its self, in reality I need to work through my node-red code to ensure that's not the problem, so I will report back sometime tomorrow.    

I may stop using TelePeriod in favour of calling status 8 that way I can define how often I want to poll for readings when the relay is energised and stop polling when the relay is de-energised as its a bit pointless asking for the power readings when I know its going to be zero, also I can start polling as soon as the relay energises instead of waiting for the time out on TelePeriod what do you think ?

Anyway I appreciate your prompt action on this issue thanks for your help with this supa software which incidentally is a bit easier to understand now you split up the code abit...

arendst, 2016-11-08:
Both issues solved in 2.0.9

jumpjack, 2016-11-09:
What should I write into a sketch to read HLW8012  directly without any additional library, firmware, MQTT server or whatelse?
I know that ESP communicates with HLW8012 through GPIO5, GPIO13 (MTCK) and GPIO14 (MTMS), but which is the protocol?

[Image](http://win98.altervista.org/SonoffPow-ann.png)

arendst, 2016-11-09:
I rest my case RIP

closed 2016-11-10 by arendst

Toshibass, 2016-11-10:
Just wanted to add ... finally had some time to test 2.0.10 and it appears to be working fine which is brilliant I just need to re-write some of my node-red code and add in BLYNK buttons and give it a long period soak test then I want to look at implementing your webserver and your ota stuff (currently using arduino ota for uploads) then implement amazon echo.... thanks for your time and the effort you have put in to make this software, it is very good.