arendst/Sonoff-MQTT-OTA-Arduino #64: Configuring Home Assistant

mihalski, 2016-11-09:
Has anyone successfully configured Home Assistant?

I've tried the following:
```
switch:
  - platform: mqtt
    name: "Bedroom Fan"
    state_topic: "stat/sonoff1/power"
    command_topic: "cmnd/sonoff1/power"
    payload_on: "ON"
    payload_off: "OFF"
```
But it seems that this cannot tell the state of the switch and is only capable of powering it on, and not off. Anyone know what I might be doing wrong here?

TLaren, 2016-11-09:
Try:
   state_topic: "stat/sonoff1/power/#"
   payload_on: "On"
   payload_off: "Off"

mihalski, 2016-11-09:
I think you meant for command_topic rather than state_topic right?

Here's what that yields:
```
(homeassistant_venv) homeassistant@homepi:~/.homeassistant $ hass --script check_config
Testing configuration at /home/homeassistant/.homeassistant
16-11-09 17:44:13 ERROR (MainThread) [homeassistant.bootstrap] Invalid config for [switch.mqtt]: Invalid MQTT topic name for dictionary value @ data['command_topic']. Got 'cmnd/sonoff1/power/#'. (See ?:?). Please check the docs at https://home-assistant.io/components/switch.mqtt/
Failed config
  switch.mqtt:
    platform: mqtt
    command_topic: cmnd/sonoff1/power/#
    name: Office
    optimistic: False
    payload_off: Off
    payload_on: On
    qos: 0
    retain: True
    state_topic: stat/sonoff1/power
```
For what it's worth, an alternative Sonoff firmware config looks like this and works perfectly:
```
  - platform: mqtt
    name: "Master Bedroom Fan"
    state_topic: "home/sonoff/master_bedroom/1/stat"
    command_topic: "home/sonoff/master_bedroom/1"
    qos: 1
    payload_on: "on"
    payload_off: "off"
    retain: true
```
That's available at https://github.com/KmanOz/Sonoff-HomeAssistant

I've never been able to get my head around how this firmware utilises MQTT topics and payloads. The documentation doesn't seem to mention payloads at all. Payload is the same as method? I don't know :/

mihalski, 2016-11-09:
I think I finally have it going. You were right about the upper case payload..
```
  - platform: mqtt
    name: "Office"
    state_topic: "stat/sonoff1/POWER"
    command_topic: "cmnd/sonoff1/power"
    payload_on: "ON"
    payload_off: "OFF"
    optimistic: false
    qos: 0
    retain: true
```
So far so good.

mihalski, 2016-11-09:
OK.. I don't know why but it seems like the status updates do not appear on the same topic when the button is used.. so it's stat/sonoff1/1/LIGHT when you press the button (that by itself works) but then you can't switch it remotely because that shows up on stat/sonoff1/POWER..

Is this a bug or intentional?

arendst, 2016-11-09:
Solved in version 2.0.10 #65

closed 2016-11-15 by arendst

cahbtexhuk, 2016-11-27:
I experiencing something completely unexplainable
I've been fiddling with Home Assistant and your firmware and I had hard times trying to make the switch work and reflect it in HA. Copied the switch config as per example, replacing mqtt topic with my values - does not work.
I've subscribed with mosquitto_sub to listen for all the topics - cmnd being sent, stat being sent back from sonoff. Does not work, HA is still confused. Button sends stat/sonoff-2425/1/POWER. I've copied the response into config - visually, none of the characters were changed - and it automagically it started working.