arendst/Sonoff-MQTT-OTA-Arduino #65: Status updates not on consistent topic

mihalski, 2016-11-09:
Hi,

I ran across this while trying to configure Home Assistant in another thread. Basically it conculded with the following:

> It seems like the status updates do not appear on the same topic when the button is used.. so it's stat/sonoff1/1/LIGHT when you press the button (that by itself works) but then you can't switch it remotely because that shows up on stat/sonoff1/POWER.

Is this a bug or intentional? Or something that I can configure? Or perhaps another topic altogether that I should be monitoring?

arendst, 2016-11-09:
It's intentionally! We (Me, myself and I) had to decide how to handle legacy commands while preparing for Sonoff Dual and the 4 channel hardware.

I therefore introduced this number just before the command part of the topic.

If you change your defined MQTT topic strings in your home automation application from ```cmnd/sonoff/power``` to ```cmnd/sonoff/1/power``` you'll be prepared for your first Sonoff Dual

arendst, 2016-11-09:
Should be resolved in released version 2.0.10

mihalski, 2016-11-10:
Thank you!
As far as I can tell at this stage it seems to be working perfectly! :)

closed 2016-11-10 by arendst