arendst/Sonoff-MQTT-OTA-Arduino #66: Handling topic for multiple devices

guglielmino, 2016-11-09:
Hello,

looking at the source I saw topic is made using PROJECT as device identifier. So, leaving default configuration and publishing to "cmnd/sonoff1/power" the relay turn on or off.
If I want to use multiple devices in my network I think I need to change the name of PROJECT for every device, is that right? Would it not be better if every device generate a unique name to be used in instead of PROJECT?  

Thanks

mihalski, 2016-11-09:
No you can simply change the MQTT topic to sonoff1, sonoff2, etc.. Or anything else you would like.

guglielmino, 2016-11-09:
Yes this is clear, but It is not the most comfortable solution.
Immagine using it with 10 devices: I have to modify the source and deploy the firmware for every device. And, worst, I can use OTA because it deploys a single firmware for all devices.
A simple solution could be to use something like MQTT_CLIENTID method, something like "sonoff_%06X" using the last 6 chars of MAC address. What do You think about?

mihalski, 2016-11-09:
Like I stated previously you DO NOT need to modify the source. The MQTT topic in the compiled code is just a default that can be changed via the web interface and will persist after firmware updates.

arendst, 2016-11-09:
mihalski is right. You just change the topic once for each sonoff like sonoff1, sonoff2 etc. You can still update all sonoffs by their grouptopic being default sonoffs.

BTW you can also address the sonoff with their unique fallback topic which happens to be the MQTTClient define

So all your concerns have been handled gracefully since the initial software release.

It still seems an big hurdle to RTFW...

closed 2016-11-10 by arendst