arendst/Sonoff-MQTT-OTA-Arduino #67: Homebridge

DJ-Q, 2016-11-09:
Has anyone managed to get this to work via homebridge?  I am trying to use the homebridge-mqttswitch plugin but i'm not sure of the correct syntax to use in the config.json, with the following setup, all I ever get is the OFF command sent, never the ON command

{
        "bridge": {
        "name": "Homebridge-02",
        "username": "CC:22:3D:E3:CE:30",
        "port": 51826,
        "pin": "031-45-154"
},

        "platforms": [
        ],
        "accessories": [
        {
                "accessory": "mqttswitch",
                "name": "BigLamp",
                "url": "mqtt://127.0.0.1",
                        "username": "",
                        "password": "",
                        "caption": "Big Lamp",
                        "topics": {
                                "statusGet": "stat/sonoff/power",
                                "statusSet": "cmnd/sonoff/power",
                                "onValue": "ON",
                                "offValue": "OFF"

                        }
        }
]
}

patbeirne, 2016-11-09:
I don't know homebridge, but if it's case sensitive, your line "statusGet" should end with an upper-case "POWER"

DJ-Q, 2016-11-09:
I keep getting the following in the console, regardless of if im selecting On or Off.  So near yet so far 

stat/sonoff/POWER = OFF

DJ-Q, 2016-11-10:
I have increased the logging level and from the output I am getting the following

DataCb Topic sonoff, Group 0, Device 1, Type POWER, Data false (FALSE)
DataCb Topic sonoff, Group 0, Device 1, Type POWER, Data true (TRUE)

Does this mean the plugin is sending "TRUE" and "FALSE" rather than "ON" and "OFF" that we are expecting?

arendst, 2016-11-10:
Correct

closed 2016-11-11 by DJ-Q

Slip7000, 2016-12-08:
Hi,  what code did you load on the Sonoff ? I'm struggling to get the right code to load on my sonoff

DJ-Q, 2016-12-08:
Hi Slip7000

What are you trying to achieve?  To get this working I flashed the sonoff with arendst's excellent firmware.  I then used homebridge to send the correct MQTT commands to switch On/Off

Slip7000, 2016-12-08:
Hi DJ-Q

Thank you for the reply, I bought a couple of Sonoff switches and going to install Homebridge on my mac. But was not sure what firmware will work with the Sonoff and the Homekit and Homebridge so that I can control the switches with Siri. Will you mind to share the code you added to you Sonoff here as an example, please. Also when you added Homebridge can you only control them with Siri or is there a manual way as well of even from the switches button ?

Slip7000, 2016-12-20:
@DJ-Q ok got 2 Sonoff's loaded with @arendst  software that is the best ever!!!!!!!! Thank you @arendst for the great work!!