arendst/Sonoff-MQTT-OTA-Arduino #70: Sonoff POW TODAY_POWER is wrong

Reinhartd, 2016-11-09:
the **TODAY_POWER** reseted every hour to "0", Version 2.0.9!

arendst, 2016-11-09:
Does UPTIME increment every hour? If not, your pow reboots. Every reboot resets TODAYPOWER.

Also, does your pow time sync with NTP?

Reinhartd, 2016-11-09:
i have 2 pow, the UPTIME is max 1 hour. 
one pow is sync with NTP and one not, the Year is here 1970.

Here is a Plot for this situation: [https://forum.fhem.de/index.php/topic,60336.msg518440.html#msg518440](url)
the red Line is TODAY_POWER.

what can i do?

arendst, 2016-11-09:
Try latest version 2.0.10 that solves two exceptions as that causes your reboots.

Reinhartd, 2016-11-09:
ok, i will test 2.0.10!

Thank you, for your great Work!

Reinhartd, 2016-11-09:
i have flashed 2.0.10 on all 2 pow. I can see the reboot broblem in a few hours.

`00:02:11 MQTT: tele/sonoffpow/TIME = 1970-01-01T00:02:11`
what can i do to sync pow with NTP?  I have an older sonoff switch, this device sync with NTP.

closed 2016-11-15 by arendst