arendst/Sonoff-MQTT-OTA-Arduino #71: sonoff pow rebooting every hour

Reinhartd, 2016-11-10:
Hello!

 yesterday i have posted the problem with TODAY_POWER, and this problem is the reason.
I have 2 sonoff pow and 2 sonoff Switches.
- sonoff Switch with V 2.0.10 NTP is OK, and no rebooting
- sonoff pow with 2.0.10 **NTP is NOT OK**, and **rebooting every hour**.

when i disable SONOFF_POW and activate SONOFF ( on the sonoff_pow module in user_config.h ), then sync with NTP is OK and no reboot (but i have no data from energy).
`//#define MODULE                 SONOFF_POW `

here is the syslog with the reboot situation
- ********************* Syslog Log sonoffpow2 ************************
Nov 10 09:25:58 sonoffpow-6772 ESP-MQTT: tele/sonoffpow/TIME = 1970-01-01T01:00:01
*** here is a change off hour on the module sonoffpow and make a reboot ***
Nov 10 09:26:08 sonoffpow-6772 ESP-RTC: (UTC) Thu Nov 10 08:26:07 2016
Nov 10 09:26:08 sonoffpow-6772 ESP-RTC: (DST) Sun Mar 27 02:00:00 2016
Nov 10 09:26:08 sonoffpow-6772 ESP-RTC: (STD) Sun Oct 30 03:00:00 2016
Nov 10 09:26:10 sonoffpow-6772 ESP-Config: Saved configuration (692 bytes) to flash at FA and count 187
Nov 10 09:26:10 sonoffpow-6772 ESP-Wifi: Checking connection...
Nov 10 09:26:10 sonoffpow-6772 ESP-Wifi: Connected
Nov 10 09:26:11 sonoffpow2-6566 ESP-MQTT: tele/sonoffpow2/UPTIME = 1
Nov 10 09:26:11 sonoffpow2-6566 ESP-MQTT: tele/sonoffpow2/RSSI = 60
Nov 10 09:26:11 sonoffpow2-6566 ESP-MQTT: tele/sonoffpow2/TODAY_POWER = 0.045
Nov 10 09:26:11 sonoffpow2-6566 ESP-MQTT: tele/sonoffpow2/PERIOD_POWER = 0
Nov 10 09:26:11 sonoffpow2-6566 ESP-MQTT: tele/sonoffpow2/CURRENT_POWER = 0
Nov 10 09:26:11 sonoffpow2-6566 ESP-MQTT: tele/sonoffpow2/POWER_FACTOR = 0.00
Nov 10 09:26:11 sonoffpow2-6566 ESP-MQTT: tele/sonoffpow2/VOLTAGE = 0
Nov 10 09:26:11 sonoffpow2-6566 ESP-MQTT: tele/sonoffpow2/CURRENT = 0.000
Nov 10 09:26:11 sonoffpow2-6566 ESP-MQTT: tele/sonoffpow2/LIGHT = ON
Nov 10 09:26:11 sonoffpow2-6566 ESP-MQTT: tele/sonoffpow2/TIME = 1970-01-01T01:00:01
*** here is a change off hour on the module sonoffpow2 and make a reboot ***
Nov 10 09:26:16 sonoffpow2-6566 ESP-Wifi: Checking connection...
Nov 10 09:26:16 sonoffpow2-6566 ESP-Wifi: Connected
Nov 10 09:26:20 sonoffpow2-6566 ESP-RTC: (UTC) Thu Nov 10 08:26:19 2016
Nov 10 09:26:20 sonoffpow2-6566 ESP-RTC: (DST) Sun Mar 27 02:00:00 2016
Nov 10 09:26:20 sonoffpow2-6566 ESP-RTC: (STD) Sun Oct 30 03:00:00 2016
Nov 10 09:26:22 sonoffpow2-6566 ESP-Config: Saved configuration (692 bytes) to flash at F9 and count 104
Nov 10 09:26:30 sonoffpow-6772 ESP-Wifi: Checking connection...

LG

arendst, 2016-11-10:
What happens when all logging is set to 0? (seriallog, weblog and syslog).

Reinhartd, 2016-11-10:
i have found a solution!
the init time of first WIFI connection is very slow, if 9 seconds is too low then can the NTP init not runing on reboot! 
i have changed this line and NTP is now running on my POWs. 

>  if ((rtcTime.Second > 57) && !ntpsync) {

here is the log, NTP is now running on my POW after rebooting!
> 00:00:46 Wifi: Checking connection...
00:00:46 Wifi: Connected
00:00:58 RTC: (UTC) Thu Nov 10 19:01:54 2016
00:00:58 RTC: (DST) Sun Mar 27 02:00:00 2016
00:00:58 RTC: (STD) Sun Oct 30 03:00:00 2016
20:01:56 Config: Saved configuration (692 bytes) to flash at FA and count 221
20:02:02 Wifi: Checking connection...

LG

Reinhartd, 2016-11-10:
... and no reboot when change the hour, TODAY_POWER is now ok!

Do you think this method is ok?

arendst, 2016-11-10:
It's a nice workaround.

arendst, 2016-11-11:
In version 2.0.11 I've updated NTP sync to wait for a wifi connection. You might give it a try.

Reinhartd, 2016-11-12:
Thank you for your good work an for the new Version, i have tested it in the Night on 2 devices!
in 2.0.11 the reboot Situation is very better, reboot ist only 10 - 12 hours. I have all logs disabled.
When the device is rebooting, the TODAY_POWER is saved, it's very good.
When the power consume is 0, then the PERIOD_POWER is always 0, is this correct? When power consume is coming, then the PERIOD_POWER is increase.

The calibration of HLW working fine and **NTP sync is 100% ok**!

LG

Reinhartd, 2016-11-12:
hm, my WIFI Router is rebooting ervery day at 07:00, is this a Problem when the pow lost the Connection?
I disable the automatic Wlan reboot and test it again!

arendst, 2016-11-12:
PERIOD_POWER is the power consumption between two teleperiod results. As you noticed it's 0 when no power has been consumed. Note that it's resolution is Wh so with a small teleperiod you won't see any value either.

You router reboot might well be the cause of sonoff reboot. I will look into it and see if a more gracefull recovery is possible.

Reinhartd, 2016-11-13:
i have diasbled the automatic router reboot and my sonoff pow ist now running 27 hours an never reboot with 2.0.11! All functions now working correct!
Thank you for your good Support!

LG

closed 2016-11-15 by arendst