arendst/Sonoff-MQTT-OTA-Arduino #74: compiling issue for a newbie.

maccer28, 2016-11-11:
im having a problem compiling on Arduino IDE, it keeps  getting an incorrect conversion error.

```
Arduino: 1.6.12 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 1M (64K SPIFFS), ck, Disabled, None"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Rory\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Rory\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Rory\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=1M64,ResetMethod=ck,Debug=Disabled,DebugLevel=None____ -ide-version=10612 -build-path C:\Users\Rory\AppData\Local\Temp\arduino_build_40102 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.esptool.path=C:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.9 -prefs=runtime.tools.mkspiffs.path=C:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -verbose C:\Users\Rory\Documents\Arduino\portable\sonoff\sonoff.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Rory\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Rory\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Rory\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=1M64,ResetMethod=ck,Debug=Disabled,DebugLevel=None____ -ide-version=10612 -build-path C:\Users\Rory\AppData\Local\Temp\arduino_build_40102 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.esptool.path=C:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.9 -prefs=runtime.tools.mkspiffs.path=C:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -verbose C:\Users\Rory\Documents\Arduino\portable\sonoff\sonoff.ino
Using board 'generic' from platform in folder: C:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0
Using core 'esp8266' from platform in folder: C:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0
Detecting libraries used...
"C:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\Rory\AppData\Local\Temp\arduino_build_40102/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10612 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\generic" "C:\Users\Rory\AppData\Local\Temp\arduino_build_40102\sketch\sonoff.ino.cpp" -o "nul"
"C:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\Rory\AppData\Local\Temp\arduino_build_40102/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10612 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\generic" "-IC:\Users\Rory\Documents\Arduino\libraries\pubsubclient-master\src" "C:\Users\Rory\AppData\Local\Temp\arduino_build_40102\sketch\sonoff.ino.cpp" -o "nul"
"C:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\Rory\AppData\Local\Temp\arduino_build_40102/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10612 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\generic" "-IC:\Users\Rory\Documents\Arduino\libraries\pubsubclient-master\src" "-IC:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\Ticker" "C:\Users\Rory\AppData\Local\Temp\arduino_build_40102\sketch\sonoff.ino.cpp" -o "nul"
"C:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\Rory\AppData\Local\Temp\arduino_build_40102/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10612 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\generic" "-IC:\Users\Rory\Documents\Arduino\libraries\pubsubclient-master\src" "-IC:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\Ticker" "-IC:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src" "C:\Users\Rory\AppData\Local\Temp\arduino_build_40102\sketch\sonoff.ino.cpp" -o "nul"
"C:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\Rory\AppData\Local\Temp\arduino_build_40102/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10612 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\generic" "-IC:\Users\Rory\Documents\Arduino\libraries\pubsubclient-master\src" "-I [... cut ...]
Using library ESP8266httpUpdate at version 1.1 in folder: C:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266httpUpdate 
Using library ESP8266WebServer at version 1.0 in folder: C:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer 
Using library DNSServer at version 1.1.0 in folder: C:\Users\Rory\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\DNSServer 
exit status 1
invalid conversion from 'const char*' to 'const uint8_t* {aka const unsigned char*}' [-fpermissive] 
```
whats am i doing wrong????????
Thanks

arendst, 2016-11-11:
Install the correct pubsubclient. See link in wiki.

maccer28, 2016-11-12:
cheers all sorted now.

closed 2016-11-12 by maccer28